open62541/open62541 #2544: Getting BadSecurityChecksFailed failure very often (every 5-10 mins) on the secure channel in the server

bvinodkumar2008, 2019-03-06:
## Description
Upon trying the encryption samples for server and client, i've found that in the server we see the BadSecurityChecksFailed error quite often which will result in the closure of the secure channel and the session. This ideally shouldn't happen as the secure channel is established with the same pair of server and client certificates with trusted certificate of CA. Due to this reason, the client is connecting back and getting the last published data.

## Background Information / Reproduction Steps

### Run server
$gcc -std=c99 -I../../include/ -g server.c ../open62541.c -L. -lmbedtls -lmbedx509 -lmbedcrypto -pthread -o server
$./server /etc/ssl/opcua/opcua_server_certificate.der /etc/ssl/opcua/opcua_server_key.der /etc/ssl/ca/ca_certificate.der &> server.txt &

### Run client
$gcc -std=c99 -I../../include/ -g client.c ../open62541.c -L. -lmbedtls -lmbedx509 -lmbedcrypto -pthread -o client
$./client /etc/ssl/opcua/opcua_client_certificate.der /etc/ssl/opcua/opcua_client_key.der /etc/ssl/ca/ca_certificate.der &> client.txt &

**Note**: I'm not able to attach the .der certificates here. The opcua_server_certificate.der and opcua_client_certificate.der are signed by ca_certificate.der. Let me know if you need these certificates.

Created a gist for server.c, client.c and their respective logs at https://gist.github.com/bvinodkumar2008/49151adddcd0636a062f5c53f78a1b66

## Checklist

Please provide the following information:

 - [46979d3b87ecc8c13cac66c0019fd31024b55b6a] open62541 Version (release number or git tag):
 - [No ] Other OPC UA SDKs used (client or server): 
 - [Ubuntu 16.04] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

is-ffl, 2019-03-06:
very similar situation here.
- server: open62541/examples/encryption/server_basic256sha256.c
- client: uaxpert 1.5.0  (1.5.1 as well)
- certificates:
     server certificate generated with open62541/tools/certs/create_self-signed.py
     client certificate (public key) taken from uaexpert PKI (der-format)
- OS: Xubuntu 18.04
- Error message: Connection n | Processing the message failed with error BadSecurityChecksFailed
  session will be re-established automatically.
  occurence: approx. every 7:30 min (with very low server activity)
- codebase: open62541 dev, master

Infinity95, 2019-03-08:
Can you check if #2557 fixes the issue for you?

closed 2019-03-10 by jpfr

bvinodkumar2008, 2019-03-11:
Hi @Infinity95, I see that the PR has got merged and issue has been closed. I will check on master and revert.

is-ffl, 2019-03-11:
@Infinity95:
First test with described testcase  (test duration 40 mins) does not show "BadSecurityChecksFailed" or automatic reconnections.
I will do further tests and come back as soon as an issue occurs.
Thanks a lot for the fix!

commit b4d0141e (2019-03-10): Fix #2544