open62541/open62541 #2555: Getting segmentation fault issue in asym_getLocalSignatureSize_sp_basic256sha256 API in client code on long run (>12 hrs)

bvinodkumar2008, 2019-03-08:
## Description
Upon trying the encryption samples for server and client, i've found that the client runs into the segmentation fault in `asym_getLocalSignatureSize_sp_basic256sha256` API (more details in the backtrace log of gdb attached below). 

## Background Information / Reproduction Steps

**Run server**:
```
$gcc -std=c99 -I../../include/ -g server.c ../open62541.c -L. -lmbedtls -lmbedx509 -lmbedcrypto -pthread -o server
$./server /etc/ssl/opcua/opcua_server_certificate.der /etc/ssl/opcua/opcua_server_key.der /etc/ssl/ca/ca_certificate.der &> server.txt &
```

**Run client**:
```
$gcc -std=c99 -I../../include/ -g client.c ../open62541.c -L. -lmbedtls -lmbedx509 -lmbedcrypto -pthread -o client
$./client /etc/ssl/opcua/opcua_client_certificate.der /etc/ssl/opcua/opcua_client_key.der /etc/ssl/ca/ca_certificate.der &> client.txt &
```

Note: I'm not able to attach the .der certificates here. The opcua_server_certificate.der and opcua_client_certificate.der are signed by ca_certificate.der. The certs are available at https://github.com/bvinodkumar2008/opcua_certs

Created a gist for server.c, client.c and their respective logs at https://gist.github.com/bvinodkumar2008/49151adddcd0636a062f5c53f78a1b66

Backtrace in gdb on the core-dumped file after loading the symbol-file (open62541_test):

```
(gdb) bt

#0  0x0000000000470247 in asym_getLocalSignatureSize_sp_basic256sha256 (securityPolicy=0xc40ae8, cc=0xc4cd10) at ../open62541.c:51139

#1  0x0000000000411659 in hideBytesAsym (channel=0xc40d00, buf_start=0x7ff1174dbbf0, buf_end=0x7ff1174dbbf8) at ../open62541.c:19816

#2  0x0000000000411c54 in UA_SecureChannel_sendAsymmetricOPNMessage (channel=0xc40d00, requestId=593976, content=0x7ff1174dbc90,

    contentType=0x47a540 <UA_TYPES+7776>) at ../open62541.c:19961

#3  0x000000000045a382 in openSecureChannel (client=0xc409a0, renew=false) at ../open62541.c:42262

#4  0x000000000045b422 in UA_Client_connectInternal (client=0xc409a0, endpointUrl=0x697720 <endpoint> "opc.tcp://localhost:65003",

    endpointsHandshake=true, createNewSession=true) at ../open62541.c:42657

#5  0x000000000045b560 in UA_Client_connect (client=0xc409a0, endpointUrl=0x697720 <endpoint> "opc.tcp://localhost:65003") at ../open62541.c:42728

#6  0x0000000000405781 in runClient (ptr=0x0) at ../open62541_wrappers.c:686

#7  0x00007ff117ac06ba in ?? ()

#8  0x0000000000000000 in ?? ()

(gdb)
```

## Checklist

Please provide the following information:

 - [ https://github.com/open62541/open62541/commit/46979d3b87ecc8c13cac66c0019fd31024b55b6a] open62541 Version (release number or git tag):
 - [No] Other OPC UA SDKs used (client or server): 
 - [ Ubuntu 16.04] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ChHaa, 2019-03-11:
Could you share the core-dump file and your binaries pls ?

bvinodkumar2008, 2019-03-11:
Sure, will send it by early tomorrow?

bvinodkumar2008, 2019-03-13:
Hi @ChHaa ,
Unfortunately i have lost both those files and have started the long run yesterday, will be sharing it as soon as i hit the issue.

Regards,
Vinod

bvinodkumar2008, 2019-03-21:
I'm closing this issue as i'm not able to repro this. I will try upgrading to the latest master and if i run into the same issue, will raise it back.

closed 2019-03-21 by bvinodkumar2008