open62541/open62541 #2562: receiveServiceResponse returns BADCONNECTIONCLOSED

davy7125, 2019-03-11:
## Description

I am trying to connect a machine with a certificate and a login / password but the error BADCONNECTIONCLOSED (0x80AE0000) is returned by the function "receiveServiceResponse" in "openSecureChannel".

Some logs are attached below so that you will see the different steps involved in the connection. How can I debug further? Is it possible that it comes from a configuration problem around the certificate?

What is interesting is that our program, using the development version of open62541, is already working for other machines and the machine we are trying to connect is successfully connected by the software "UA Expert".

## Logs
```
07:46:05,587 INFO: successfully configured a default connection with UA_ClientConfig_setDefault
07:46:05,587 INFO: configuring a secure connection with the certificate
07:46:05,603 INFO: successfully configured with UA_SecurityPolicy_None
07:46:05,603 INFO: successfully configured with UA_SecurityPolicy_Basic256Sha256
07:46:05,603 INFO: successfully configured with UA_SecurityPolicy_Basic128Rsa15
07:46:05,603 INFO: Success in creating and configuring a client
07:46:05,618 INFO: browsing endpoints
07:46:05,618 INFO: [Open62541, client] SecurityPolicy not specified -> use default #None
07:46:05,618 WARN: [Open62541, security policy] No PKI plugin set. Accepting all certificates
07:46:05,618 INFO: [Open62541, client] TCP connection established
07:46:05,634 INFO: [Open62541, client] Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
07:46:05,634 INFO: found 7 endpoint(s)
07:46:05,634 INFO: URL of endpoint 0 is opc.tcp://10.131.56.251:4840, security mode = 'none (1)', security policy = 'http://opcfoundation.org/UA/SecurityPolicy#None', security level = '0'
07:46:05,634 INFO: URL of endpoint 1 is opc.tcp://10.131.56.251:4840, security mode = 'sign (2)', security policy = 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15', security level = '55'
07:46:05,634 INFO: URL of endpoint 2 is opc.tcp://10.131.56.251:4840, security mode = 'sign and encrypt (3)', security policy = 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15', security level = '105'
07:46:05,634 INFO: URL of endpoint 3 is opc.tcp://10.131.56.251:4840, security mode = 'sign (2)', security policy = 'http://opcfoundation.org/UA/SecurityPolicy#Basic256', security level = '60'
07:46:05,634 INFO: URL of endpoint 4 is opc.tcp://10.131.56.251:4840, security mode = 'sign and encrypt (3)', security policy = 'http://opcfoundation.org/UA/SecurityPolicy#Basic256', security level = '110'
07:46:05,634 INFO: URL of endpoint 5 is opc.tcp://10.131.56.251:4840, security mode = 'sign (2)', security policy = 'http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256', security level = '65'
07:46:05,634 INFO: URL of endpoint 6 is opc.tcp://10.131.56.251:4840, security mode = 'sign and encrypt (3)', security policy = 'http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256', security level = '115'
07:46:05,634 INFO: try to connect to 'opc.tcp://10.131.56.251:4840' with username = 'XXX' and password = 'XXX'
07:46:05,650 INFO: [Open62541, client] Connecting to endpoint opc.tcp://10.131.56.251:4840
07:46:05,650 INFO: [Open62541, client] SecurityPolicy not specified -> use default #None
07:46:05,650 WARN: [Open62541, security policy] No PKI plugin set. Accepting all certificates
07:46:05,650 INFO: [Open62541, client] TCP connection established
07:46:05,650 INFO: [Open62541, client] Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
07:46:05,650 INFO: [Open62541, client] Endpoint and UserTokenPolicy unconfigured, perform GetEndpoints
07:46:05,665 INFO: [Open62541, client] Found 7 endpoints
07:46:05,665 INFO: [Open62541, client] Rejecting endpoint 0: security mode doesn't match
07:46:05,665 INFO: [Open62541, client] Rejecting endpoint 1: security mode doesn't match
07:46:05,665 INFO: [Open62541, client] Endpoint 2 has 5 user token policies
07:46:05,665 INFO: [Open62541, client] Rejecting UserTokenPolicy 0 in endpoint 2: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic256' not available
07:46:05,665 INFO: [Open62541, client] Selected Endpoint opc.tcp://10.131.56.251:4840 with SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
07:46:05,665 INFO: [Open62541, client] Selected UserTokenPolicy UserName_256Sha256_Token with UserTokenType UserName and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
07:46:05,665 INFO: [Open62541, client] Disconnect to switch to a different SecurityPolicy
07:46:05,665 INFO: [Open62541, client] Connecting to endpoint opc.tcp://10.131.56.251:4840
07:46:05,681 INFO: [Open62541, client] TCP connection established
07:46:05,759 ERR : [Open62541, secure channel] Receiving service response failed with error 0x80AE0000
07:46:05,759 ERR : [Open62541, client] Opening a secure channel failed
07:46:05,759 ERR : [Open62541, client] Couldn't connect the client to a TCP secure channel
07:46:05,759 ERR : connection to opc.tcp://10.131.56.251:4840 returned error 0x80AE0000
```

Infinity95, 2019-03-11:
Which Server and Client are you using?
Cann you supply a Wireshark dump?

davy7125, 2019-03-13:
I am using the open62541 library in the current development state (+ some logs) and the machine we are trying to connect is a Siemens SINUMERIK 840D SL. You will find attached the wireshark dump.
[opcua_filtered.zip](https://github.com/open62541/open62541/files/2962275/opcua_filtered.zip)

Infinity95, 2019-03-18:
The server returns an error message with BADINVALIDARGUMENT.
Do you have access to the server logs? Without more information it is hard to tell what is going wrong here, since the securechannel open requests are encrypted.
In order to further debug, we would need to check either the servers logs, or the content of the open messages, for which we would need the private key of the server.
If you cannot provide the private key, you could decrypt the message and post the decrypted version as a hex string.

To decrypt the packages, you could try setting the key in wireshark if you have access to it.

davy7125, 2019-03-19:
I followed this link to decrypt https://support.citrix.com/article/CTX116557
The result is "Continuation Data", I guess something is wrong or I didn't get the right server key.

[image]

We couldn't find the logs on the server either... Really hard to debug I hope we will find something.

Infinity95, 2019-03-19:
Ah, my mistake. I thought wireshark had a feature to decrypt OPCUA traffic. Apparently not.
OPC UA doesn't use TLS/SSL for encryption.

You can try using the [scapy plugin](https://github.com/Infinity95/scapy/tree/feature/opcua/scapy/contrib/opcua) i wrote for opcua to decrypt the packet.

davy7125, 2019-03-20:
Unfortunately I'm not comfortable at all with this tool. I have 3 files .pcapng, .der and .pem (actually a forth one which is also a .pem) so I should have everything but in the example the method reading .pcapng (read_pcap) returns the variable "pc" that is not used.

So I'm enclosing the files here if by chance you can try something fast but on our side we will focus more on the server logs if we can finally find / enable some.

[data.zip](https://github.com/open62541/open62541/files/2987248/data.zip)

heppth, 2019-03-27:
I do not have a solution, but my issue #2574 seems to be the same problem. Perhaps I can help spending more information. An other way would be, that one of the specialists making a short test againt the unified automation cpp server (download [here](https://www.unified-automation.com/downloads/opc-ua-servers.html)).

fanucwj, 2020-10-21:
I have the same problem as you。the server is Siemens SINUMERIK 840D SL which is  is successfully connected by the software "UA Expert"..   have you solved the problem???? Thank you in advance

heppth, 2020-10-21:
> I have the same problem as you。the server is Siemens SINUMERIK 840D SL which is is successfully connected by the software "UA Expert".. have you solved the problem???? Thank you in advance

I do not have any problems with the newest version. 
I have a tip for you: Use the certificate generated by UaExpert to establish the connection. This way you can rule out that there is a problem with the certificate you generated. And don't make the same mistake as I did [here](https://github.com/open62541/open62541/issues/2574#issuecomment-686546739). If the UaExpert certificate works, you can try to generate one yourself that meets the requirements.

fanucwj, 2020-10-21:
thank you very much for you replying . I use this library in arm, may i use your version to connect my server?  thank you in advance!!

fanucwj, 2020-10-21:
> I am using the open62541 library in the current development state (+ some logs) and the machine we are trying to connect is a Siemens SINUMERIK 840D SL. You will find attached the wireshark dump.
> [opcua_filtered.zip](https://github.com/open62541/open62541/files/2962275/opcua_filtered.zip)
> 
> We can read the error 0x80AB0000 (UA_STATUSCODE_BADINVALIDARGUMENT) after the "OpenSecureChannel" request. I guess this is due to the certificate?

I have the same problem as you。the server i try to connect is Siemens SINUMERIK 840D SL which is is successfully connected by the software "UA Expert".. have you solved the problem???? Thank you in advance

teccheck, 2025-03-18:
As this issue hasn't seen any recent activity, and I'm unable to test the problem because I don't have the hardware, I'll close this

closed 2025-03-18 by teccheck