open62541/open62541 #2564: LifeCycle, how to know in constructor if instance added to Objects or Types folder?

juangburgos, 2019-03-11:
## Description

Using the Server API.

[image]

I register multiple custom VariableTypes which then I use in a custom ObjectType called `MyNewObjectType`.  Then I create an instanc eof this type called `Test`.

I register the same constructor callback for all types, which takes the new NodeId and browses *Inverse* (upwards) to find the parent NodeId. This is done **recursivelly** until the obtained parent NodeId is either the ObjectsFolder, TypesFolder or UA_NODEID_NULL.

The issue I find is that for some cases, I find impossible to determine if the node being instantiated belongs to the ObjectsFolder or TypesFolder (see attached image).

When calling the constructor for all four cases marked with **red** arrows, it is only possible to get the NodeId of the direct parent, but when getting the NodeId of the grand parent is returned as UA_NODEID_NULL. So is impossible to follow the path upwards to find the ObjectsFolder or TypesFolder inside the constructor callback.

The constructor of the *parent* of the cases marked with **red** arrows are actually called *afterwards* and at that point is possible to browse completely upwards, but then is too late, because the children were left undetermined to know if they belong to the ObjectsFolder or TypesFolder.

Is there something I can use to find out directly in the constructor callback if a Node belongs to the ObjectsFolder or TypesFolder for every possible case?

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: any
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-03-13:
Can you post the code you use to find the parent?
Note that the references inside the ObjectType are `hasComponent` or `hasProperty`.
But between ObjectTypes the references are `hasSubtype`.

juangburgos, 2019-03-15:
Hi,

Here is the (C++) code. I will try to create a standalone example to replicate my problem. But if you se any issue in the code below please let me know:

```c++
UA_NodeId QOpcUaServerNode::getParentNodeId(const UA_NodeId & childNodeId, UA_Server * server)
{
	UA_BrowseDescription * bDesc = UA_BrowseDescription_new();
	bDesc->nodeId          = childNodeId; // from child
	bDesc->browseDirection = UA_BROWSEDIRECTION_INVERSE; //  look upwards
	bDesc->includeSubtypes = true;
	bDesc->resultMask      = UA_BROWSERESULTMASK_BROWSENAME | UA_BROWSERESULTMASK_DISPLAYNAME;
	// browse
	UA_BrowseResult bRes = UA_Server_browse(server, 0, bDesc);
	Q_ASSERT(bRes.statusCode == UA_STATUSCODE_GOOD);
	QList<UA_NodeId> listParents;
	while (bRes.referencesSize > 0)
	{
		for (size_t i = 0; i < bRes.referencesSize; i++)
		{
			UA_ReferenceDescription rDesc = bRes.references[i];
			UA_NodeId nodeId = rDesc.nodeId.nodeId;
			listParents.append(nodeId);
		}
		bRes = UA_Server_browseNext(server, true, &bRes.continuationPoint);
	}
	// cleanup
	UA_BrowseDescription_deleteMembers(bDesc);
	UA_BrowseDescription_delete(bDesc);
	UA_BrowseResult_deleteMembers(&bRes);
	// return
	Q_ASSERT_X(listParents.count() <= 1, "QOpcUaServer::getParentNodeId", "Child code it not supposed to have more than one parent.");
	return listParents.count() > 0 ? listParents.at(0) : UA_NODEID_NULL;
}
```

juangburgos, 2019-03-15:
Ok, it took me a while, but at the end of this post there is a standalone example. On the client it produces something like this:

[image]

On the left we see the **order in which the constructors are called**, which I think is correct. On the right we see the **assigned NodeIds**.

Marked with a **star** we see the nodes that cannot reach the root folder during the constructor call. The output of the example program prints the parents-chain reachible from each node when the constructor is called:

```
1) [ns=0;i=316] BaseDataVariableType Constructor
>> Parents -> [ns=0;i=119] -> [UA_NS0ID_BASEDATAVARIABLETYPE]

2) [ns=0;i=319] BaseDataVariableType Constructor
>> Parents -> [ns=0;i=318] -> [ns=0;i=317] -> [UA_NS0ID_BASEDATAVARIABLETYPE]

3) [ns=0;i=318] MyVariableType Constructor
>> Parents -> [ns=0;i=317] -> [UA_NS0ID_BASEDATAVARIABLETYPE]

4) [ns=0;i=323] BaseDataVariableType Constructor
>> Parents -> [ns=0;i=322] -> [UA_NODEID_NULL]

5) [ns=0;i=322] MyVariableType Constructor
>> Parents -> [ns=0;i=321] -> [ns=0;i=320] -> [UA_NS0ID_BASEOBJECTTYPE]

6) [ns=0;i=321] MyVariableOtherType Constructor
>> Parents -> [ns=0;i=320] -> [UA_NS0ID_BASEOBJECTTYPE]

7) [ns=0;i=327] BaseDataVariableType Constructor
>> Parents -> [ns=0;i=326] -> [UA_NODEID_NULL]

8) [ns=0;i=326] MyVariableType Constructor
>> Parents -> [ns=0;i=325] -> [UA_NODEID_NULL]

9) [ns=0;i=325] MyVariableOtherType Constructor
>> Parents -> [ns=0;i=324] -> [UA_NS0ID_OBJECTSFOLDER]

10) [ns=0;i=324] MyObjectType Constructor
>> Parents -> [UA_NS0ID_OBJECTSFOLDER]
```

As it can be seen, nodes `ns=0;i=323`, `ns=0;i=327` and `ns=0;i=326` cannot reach a parent further than its direct parent. So it seems impossible to determine in the constructor whether the new node is going to belong to *Objects*, or *Types* folder, and this is my issue.

The output is produced by this code:

```c++
#include <assert.h>
#include <signal.h>
#include "open62541.h"

static void printNodeId(const UA_NodeId nodeId)
{
	printf("ns=%i;", nodeId.namespaceIndex);
	switch (nodeId.identifierType)
	{
	case UA_NODEIDTYPE_NUMERIC:
		printf("i=%i", nodeId.identifier.numeric);
		break;
	case UA_NODEIDTYPE_STRING:
		printf("s=%s", nodeId.identifier.string.data);
		break;
	case UA_NODEIDTYPE_GUID:
		printf("g=%i%i%i&i", nodeId.identifier.guid.data1, nodeId.identifier.guid.data2, nodeId.identifier.guid.data3, nodeId.identifier.guid.data4);
		break;
	case UA_NODEIDTYPE_BYTESTRING:
		printf("b=%s", nodeId.identifier.byteString.data);
		break;
	}
}

static UA_NodeId getParentNodeId(const UA_NodeId childNodeId, UA_Server * server)
[... 265 lines cut ...]
	running = false;
}

int main(void) 
{

	UA_ServerConfig *config = UA_ServerConfig_new_default();
	UA_Server       *server = UA_Server_new(config);

	UA_NodeId objectsNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);

	setBaseDataVariableTypeConstructor(server);

	addMyVariableType(server);
	addMyVariableOtherType(server);

	addMyObjectType(server);
	addMyObjectInstance(objectsNodeId, server);

	UA_StatusCode retval = UA_Server_run(server, &running);
	UA_Server_delete(server);
	UA_ServerConfig_delete(config);
	return (int)retval;
}
```

jpfr, 2019-03-17:
Thanks for the example code. That helped a lot!

The source of the issue was that some node constructors were called once the TypeDefintion references were added but before the parent reference was added.

The behavior is now more consistent in PR #2576.
First, we create all mandatory children and add the references.
Then, in a second pass, we call the constructors of the children from the tree leaves upwards.

jpfr, 2019-03-18:
Merged on master.
Can you verify and close?

juangburgos, 2019-03-18:
Yes, verified and closing. You guys are doing an **awesome job**. Thank you for this library.

Another question out of curiosity. I have been using `v0.3-rc4` so far, and obtained the NodeIds that you saw in my example above. But now in master I get very different NodeIds, are they now pseudo-random?

```
1) [ns=0;i=1479928322] BaseDataVariableType Constructor
>> Parents -> [ns=0;i=3312717455] -> [UA_NS0ID_BASEDATAVARIABLETYPE]

2) [ns=0;i=1915794170] BaseDataVariableType Constructor
>> Parents -> [ns=0;i=2426563266] -> [ns=0;i=1853330170] -> [UA_NS0ID_BASEDATAVARIABLETYPE]

3) [ns=0;i=2426563266] MyVariableType Constructor
>> Parents -> [ns=0;i=1853330170] -> [UA_NS0ID_BASEDATAVARIABLETYPE]

4) [ns=0;i=3876777541] BaseDataVariableType Constructor
>> Parents -> [ns=0;i=4120066047] -> [ns=0;i=2964493476] -> [ns=0;i=891152822] -> [UA_NS0ID_BASEOBJECTTYPE]

5) [ns=0;i=4120066047] MyVariableType Constructor
>> Parents -> [ns=0;i=2964493476] -> [ns=0;i=891152822] -> [UA_NS0ID_BASEOBJECTTYPE]

6) [ns=0;i=2964493476] MyVariableOtherType Constructor
>> Parents -> [ns=0;i=891152822] -> [UA_NS0ID_BASEOBJECTTYPE]

7) [ns=0;i=4239222959] BaseDataVariableType Constructor
>> Parents -> [ns=0;i=2523914624] -> [ns=0;i=690960032] -> [ns=0;i=3777284079] -> [UA_NS0ID_OBJECTSFOLDER]

8) [ns=0;i=2523914624] MyVariableType Constructor
>> Parents -> [ns=0;i=690960032] -> [ns=0;i=3777284079] -> [UA_NS0ID_OBJECTSFOLDER]

9) [ns=0;i=690960032] MyVariableOtherType Constructor
>> Parents -> [ns=0;i=3777284079] -> [UA_NS0ID_OBJECTSFOLDER]

10) [ns=0;i=3777284079] MyObjectType Constructor
>> Parents -> [UA_NS0ID_OBJECTSFOLDER]
```

closed 2019-03-18 by juangburgos