open62541/open62541 #2567: installation on debian 4.9.110

Sun-Dordje, 2019-03-12:
hello, im a student from the university of Bordeaux (France) which try to implement a client and server OPC UA on my Debian 4.9.110. 
I followed the tutorial from https://open62541.org/doc/current/index.html, but when 
i want to build a simple Server, its does not work, i think i have problem with the library installation.

i did these steps:

1)-git clone https://github.com/open62541/open62541.git
-cd open62541
-git submodule update --init --recursive
mkdir build && cd build

2)-make -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_AMALGAMATION=OFF -DUA_PACK_DEBIAN=ON ..
	out===>-- The selected architecture is: posix
	-- Performing Test CC_HAS_-D_FORTIFY_SOURCE=2
	-- Performing Test CC_HAS_-D_FORTIFY_SOURCE=2 - Success
	-- Configuring done
	-- Generating done
	-- Build files have been written to: /root/open62541/build

3)-make
	out===> [  0%] Built target open62541-generator-transport
	[  3%] Built target open62541-generator-statuscode
	[  5%] Built target open62541-generator-types
	[  5%] Built target open62541-generator-namespace
	[ 78%] Built target open62541-object
	[ 98%] Built target open62541-plugins
	[100%] Built target open62541

4)make install
	out===> Install configuration: "RelWithDebInfo"
	-- Up-to-date: /usr/include/lib/libopen62541.so.0.4.0
	[...]
	-- Up-to-date: /usr/include/lib/cmake/open62541/open62541Targets.cmake
	-- Installing: /usr/include/lib/cmake/open62541/open62541Targets-relwithdebinfo.cmake
	[...]
	-- Up-to-date: /usr/include/share/open62541/tools/certs
	-- Up-to-date: /usr/include/share/open62541/tools/certs/localhost.cnf
	[...]
	-- Up-to-date: /usr/include/include/open62541/ua_types_encoding_binary.h
	-- Up-to-date: /usr/include/include/open62541/ua_types_generated_encoding_binary.h
	-- Up-to-date: /usr/include/include/open62541/ua_transport_generated.h

(this directory seems not normal : /usr/include/include/open62541 ?)

5) i have download open62541.c/.h and deposit it in open62541/examples/

6)i go in /open62541/examples and make : gcc -std=c99  open62541.c server.c -o server

	=> fatal error: ua_server.h: No such file or directory

	So i tried to add open62541/ front of headers in server.c but :

	=> indefinite reference to "UA_ServerConfig_new_customBuffer"

I tried to include libraries with gcc with the option -I/usr/include/open62541  but no success.

I think it's a problem with a wrong path installation but i dont know what i need to change.
Im lost so if someone can help me its very cool.
Sorry for my english.

thanks
Dordjé

## Description

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-03-12:
You should not use the `-DUA_PACK_DEBIAN=ON` option, if you just want to do a normal installation.
Also, why do you additionally download the amalgamated files in step 5 if you already install them?

Did you have a look at:
https://open62541.org/doc/current/installing.html

We also provide some pre-build .deb packages:
https://launchpad.net/~open62541-team/+archive/ubuntu/daily

After installation you can just copy the examples folder (from master branch) to somewhere in your file system and execute (don't build it inside the open62541 folder):

```bash
cd examples
mkdir build && cd build
cmake ..
make -j
```

Sun-Dordje, 2019-03-12:
thanks to answer.

>You should not use the -DUA_PACK_DEBIAN=ON option, if you just want to do a normal installation.

first i tried without this, but i had same errors and the file debian/rules mention it so i tried with this option.

>Also, why do you additionally download the amalgamated files in step 5 if you already install them?

i followed the tutorial : "To get started, downdload the open62541 single-file release from http://open62541.org or generate it according to the build instructions with the “amalgamation” option enabled." and i put the option  -DUA_ENABLE_AMALGAMATION=OFF so i download the files...

>https://open62541.org/doc/current/installing.html

i followed exactly these steps...

i always have this error:
=> fatal error: ua_server.h: No such file or directory
when i try to compile a simple server from https://open62541.org/doc/current/tutorial_server_firststeps.html

headers and dependencies are not in the good path..?

Dordjé

Sun-Dordje, 2019-03-13:
Hi, after some tests im always in the dark, i repeat what i did:

1)su -
2)cd ~
3)git clone https://github.com/open62541/open62541.git

	>Clonage dans 'open62541'...
	remote: Enumerating objects: 39693, done.
	remote: Total 39693 (delta 0), reused 0 (delta 0), pack-reused 39693
	Réception d'objets: 100% (39693/39693), 21.08 MiB | 435.00 KiB/s, fait.
	Résolution des deltas: 100% (30628/30628), fait.

4)cd open62541/

5)git submodule update --init --recursive

	>Sous-module 'deps/mdnsd' (https://github.com/Pro/mdnsd.git) enregistré pour le chemin 'deps/mdnsd'
	Sous-module 'deps/ua-nodeset' (https://github.com/OPCFoundation/UA-Nodeset.git) enregistré pour le chemin 'deps/ua-nodeset'
	Clonage dans '/root/open62541/deps/mdnsd'...
	Clonage dans '/root/open62541/deps/ua-nodeset'...
	Chemin de sous-module 'deps/mdnsd' : '9e953b8e4c54d50ba0e174f1e98cfca18f933126' extrait
	Chemin de sous-module 'deps/ua-nodeset' : '5bbf784e9376f7230098149dc0218f318a48d630' extrait

6)cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL ..

	>-- The C compiler identification is GNU 6.3.0
	-- The CXX compiler identification is GNU 6.3.0
	-- Check for working C compiler: /usr/bin/cc
	-- Check for working C compiler: /usr/bin/cc -- works
	-- Detecting C compiler ABI info
	-- Detecting C compiler ABI info - done
	-- Detecting C compile features
	-- Detecting C compile features - done
	-- Check for working CXX compiler: /usr/bin/c++
	-- Check for working CXX compiler: /usr/bin/c++ -- works
	-- Detecting CXX compiler ABI info
	-- Detecting CXX compiler ABI info - done
	-- Detecting CXX compile features
	-- Detecting CXX compile features - done
	-- Found PythonInterp: /usr/bin/python (found version "2.7.13") 
	-- Found Git: /usr/bin/git (found version "2.11.0") 
	-- The selected architecture is: posix
	-- Performing Test CC_HAS_-fstack-protector-strong
	-- Performing Test CC_HAS_-fstack-protector-strong - Success
	-- Performing Test CC_HAS_-fstack-clash-protection
	-- Performing Test CC_HAS_-fstack-clash-protection - Failed
	-- Performing Test CC_HAS_-mcet -fcf-protection
	-- Performing Test CC_HAS_-mcet -fcf-protection - Failed
	-- Performing Test CC_HAS_-D_FORTIFY_SOURCE=2
	-- Performing Test CC_HAS_-D_FORTIFY_SOURCE=2 - Success
	-- Found Sphinx: /usr/bin/sphinx-build  
	-- Found LATEX: /usr/bin/latex   
	-- Configuring done
	-- Generating done
	-- Build files have been written to: /root/open62541/build

7)make

	>[  1%] Generating src_generated/ua_transport_generated.c, src_generated/ua_transport_generated.h, src_generated/ua_transport_generated_handling.h, src_generated/ua_transport_generated_encoding_binary.h
	[  1%] Built target open62541-generator-transport
	Scanning dependencies of target open62541-generator-statuscode
	[  3%] Generating src_generated/ua_nodeids.h
	[  3%] Generating src_generated/ua_statuscodes.h, src_generated/ua_statuscodes.c
	[  3%] Built target open62541-generator-statuscode
	Scanning dependencies of target open62541-generator-types
	[  5%] Generating src_generated/ua_types_generated.c, src_generated/ua_types_generated.h, src_generated/ua_types_generated_handling.h, src_generated/ua_types_generated_encoding_binary.h
	[  5%] Built target open62541-generator-types
	Scanning dependencies of target open62541-generator-namespace
	[  7%] Generating src_generated/ua_namespace0.c, src_generated/ua_namespace0.h
	INFO:__main__:Preprocessing /root/open62541/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
[... 303 lines cut ...]
	-- Installing: /usr/local/include/open62541/ua_securitypolicies.h
	-- Installing: /usr/local/include/open62541/ua_network_tcp.h
	-- Installing: /usr/local/include/open62541/ua_architecture_functions.h

8)cd ../.. (im in /root)
9)cp -r open62541/example .
10)cd example/
11)wget https://github.com/open62541/open62541/releases/download/v0.3.0/open62541.c
12)wget https://github.com/open62541/open62541/releases/download/v0.3.0/open62541.h
13)touch server.c  (here i put the code from https://open62541.org/doc/current/tutorial_server_firststeps.html)
14) gcc -std=c99 open62541.c serv.c -o serv

	>server.c:1:23: fatal error: ua_server.h: No such file or directory
	 #include <ua_server.h>
		               ^
	compilation terminated.
15) and with the option gcc -I: gcc -std=c99 -I /usr/local/include/open62541/  open62541.c serv.c -o serv

	>/tmp/ccdXCKCZ.o : in function « UA_ServerConfig_new_minimal » :
	serv.c:(.text+0xfd) : undefined reference to « UA_ServerConfig_new_customBuffer »
	collect2: error: ld returned 1 exit status

hope somebody can help me, thanks.

Dorjdé

Pro, 2019-03-13:
You are mixing different versions...

We changed the includes between 0.3 and master.

In step 3 you are cloning, which automatically checks out master

Then, why do you download the files in step 11 and step 12?? You already install the stack in step 7.

Therefore try the following:

- Remove step 11 and step 12.
- After step 10, before trying to create your own file, first try with cmake if it works:
```
cd examples/
mkdir build && cd build
cmake ..
make -j
```

Then we recommend creating a new CMake project for your own code and include open62541 as a library.

The `ua_server.h` file is installed in your system. Check the output above:
> -- Installing: /usr/local/include/open62541/ua_server.h

Therefore you need to set the additional include directory if you want to call gcc directly. And also use the precompiled library with the `-l` switch.

```bash
gcc -std=c99 -lopen62541 -I/usr/local/include/open62541/ serv.c -o serv
```

The wrong include path will be fixed with #2448

Sun-Dordje, 2019-03-13:
oh my god it's work and i was block for a long time just for this... thank you so much Stefan Profanter!

closed 2019-03-14 by Pro