open62541/open62541 #2568: Generating certificate and encryption keys

smokingaces88, 2019-03-12:
## Description
A simple question: I am having trouble generating certificate and keys from the "create_self-signed.py". I think I need to pass some argument to this script because the message I get is where it fails at:

if len(sys.argv) < 2:
    sys.exit('Usage: %s directory to output certificates' % sys.argv[0])

Can anyone suggest what parameter needs to be included? I have searched around but can't seem to find an answer.

## Checklist

Please provide the following information:

 - [0.3 ] open62541 Version (release number or git tag):
 - [client ] Other OPC UA SDKs used (client or server): 
 - [windows 10 ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

is-ffl, 2019-03-12:
When using this script with (x)ubuntu, I have to pass a path as argument, where the certificate/key will be written to.
Example:
./create_self-signed.py .
(note the dot at the end of the cmdline) will place certificate/key to the current directory.
I assume, it will be similar with Win10

smokingaces88, 2019-03-12:
Thanks for the reply. Having followed your suggestion I am able to get a bit further but now I have the following error:

'openssl' is not recognized as an internal or external command, operable program or batch file.
'openssl' is not recognized as an internal or external command, operable program or batch file.
'openssl' is not recognized as an internal or external command, operable program or batch file.
Traceback (most recent call last): 
    File "tools\certs\create_self-signed.py", line 41, in <module> os.remove("localhost.key")
FilenotFoundError: [WinError 2] The system cannot find the file specified: 'localhost.key'

any ideas?

is-ffl, 2019-03-12:
the script generates the certificate/key by using the openssl package.
The error message indicates openssl is not installed on your pc or not reachable by the script.
https://stackoverflow.com/questions/50625283/how-to-install-openssl-in-windows-10

juangburgos, 2019-04-11:
Hi,
I added to my project's documentation some notes on how to generate certificates for open62541 server that work with UA Expert client, maybe they help anybody who has struggled to create certificates:

https://github.com/juangburgos/QUaServer#create-certificates

jpfr, 2019-04-13:
@juanburgos , nice description.
Can we reuse some of that in our docs as well?
The core principles are the same after all.

Your C++ API looks really elegant to use.
We should refer to that more prominently.
As a first step, you could add a link to the wiki.

Then we can make a section in the docs on language bindings that use open62541.

juangburgos, 2019-04-15:
@jpfr , thank you, of course you can reuse anything you like. If you feel I could expand on something in my docs or make it better somehow, please let me know.

I would be honored if you mention my project on your wiki, but please bear in mind the project is very young and I am quite a rookie programmer (I am actually not a software engineer).

Pro, 2019-04-30:
I added your project to our wiki:
https://github.com/open62541/open62541/wiki/References-to-open62541

Now we should improve our certificates doc using your content. Maybe someone could create a PR? :)

teccheck, 2025-03-18:
The script was reworked to function without any arguments. In that case it assumes default values

closed 2025-03-18 by teccheck