open62541/open62541 #2574: Unable to connect to UaCPPServer via username and password

heppth, 2019-03-14:
- [X] open62541 Version (release number or git tag): latest master
 - [X] Other OPC UA SDKs used (client or server): Unified Automation test server (cpp)
 - [X] Operating system: Ubuntu Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached

Since an encrypted connection should now be possible, I have been trying to connect to the Unified Automation test server for a day. Without success.

If have debugged the code. Im always getting Error 0x80ae0000 from method `openSecureChannel(UA_Client *client, UA_Boolean renew) `.

What I have done:
- Created valid certificate and private key
- Added certificate to trusted list
- Disabled all kind of certificate validation in server
- Tried all combinations of policy urls (None, Basic128Rsa15, Basic256Sha256)
- Tried all message security modes (NONE, SIGNN, SIGNANDENCRYPT)

An connection without username and password is working as exacted, but as soon as encryption comes into play, it's not longer possible to establish a connection. What could I do wrong? Are there any known restrictions?

jpfr, 2019-03-15:
Which version of open62541?
The log output should be much improvement on the current master branch.

heppth, 2019-03-15:
The header file says Git-Revision: 0.3-rc2-830-g58a3423e. This must be pretty new

heppth, 2019-03-25:
Hi,

today I testet the connection with the latest version `0.3-rc2-858-gd070a2ff`. Still no success. Here is the log:

> [2019-03-25 09:07:17.740 (UTC+0000)] [32minfo/client[0m	Connecting to endpoint opc.tcp://192.168.178.54:48010
[2019-03-25 09:07:17.740 (UTC+0000)] debug/client[0m	Initialize the SecurityPolicy context
[2019-03-25 09:07:17.740 (UTC+0000)] [32minfo/client[0m	SecurityPolicy not specified -> use default #None
[2019-03-25 09:07:17.741 (UTC+0000)] [33mwarn/securitypolicy[0m	No PKI plugin set. Accepting all certificates
[2019-03-25 09:07:17.742 (UTC+0000)] [32minfo/client[0m	TCP connection established
[2019-03-25 09:07:17.742 (UTC+0000)] debug/network[0m	Sent HEL message
[2019-03-25 09:07:17.744 (UTC+0000)] debug/network[0m	Received ACK message
[2019-03-25 09:07:17.744 (UTC+0000)] debug/channel[0m	Requesting to open a SecureChannel
[2019-03-25 09:07:17.744 (UTC+0000)] debug/channel[0m	OPN message sent
[2019-03-25 09:07:17.745 (UTC+0000)] debug/client[0m	Decode a message of type OpenSecureChannelResponse
[2019-03-25 09:07:17.745 (UTC+0000)] [32minfo/client[0m	Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2019-03-25 09:07:17.745 (UTC+0000)] [32minfo/client[0m	Endpoint and UserTokenPolicy unconfigured, perform GetEndpoints
[2019-03-25 09:07:17.746 (UTC+0000)] debug/client[0m	Sending a request of type 426
[2019-03-25 09:07:17.747 (UTC+0000)] debug/client[0m	Decode a message of type GetEndpointsResponse
[2019-03-25 09:07:17.748 (UTC+0000)] [32minfo/client[0m	Found 7 endpoints
[2019-03-25 09:07:17.748 (UTC+0000)] [32minfo/client[0m	Endpoint 0 has 3 user token policies
[2019-03-25 09:07:17.748 (UTC+0000)] [32minfo/client[0m	Rejecting UserTokenPolicy 0 (anonymous) in endpoint 0: configuration doesn't match
[2019-03-25 09:07:17.748 (UTC+0000)] [32minfo/client[0m	Selected Endpoint opc.tcp://THINKPAD:48010 with SecurityMode Sign and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2019-03-25 09:07:17.748 (UTC+0000)] [32minfo/client[0m	Selected UserTokenPolicy UserName with UserTokenType UserName and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2019-03-25 09:07:17.749 (UTC+0000)] [32minfo/client[0m	Disconnect to switch to a different SecurityPolicy
[2019-03-25 09:07:17.749 (UTC+0000)] [32minfo/client[0m	Connecting to endpoint opc.tcp://192.168.178.54:48010
[2019-03-25 09:07:17.749 (UTC+0000)] debug/client[0m	Initialize the SecurityPolicy context
[2019-03-25 09:07:17.750 (UTC+0000)] [32minfo/client[0m	TCP connection established
[2019-03-25 09:07:17.750 (UTC+0000)] debug/network[0m	Sent HEL message
[2019-03-25 09:07:17.751 (UTC+0000)] debug/network[0m	Received ACK message
[2019-03-25 09:07:17.751 (UTC+0000)] debug/channel[0m	Requesting to open a SecureChannel
[2019-03-25 09:07:18.115 (UTC+0000)] debug/channel[0m	OPN message sent
[2019-03-25 09:07:18.117 (UTC+0000)] [31merror/channel[0m	Receiving service response failed with error BadConnectionClosed
[2019-03-25 09:07:18.117 (UTC+0000)] [31merror/client[0m	Opening a secure channel failed
[2019-03-25 09:07:18.117 (UTC+0000)] [31merror/client[0m	Couldn't connect the client to a TCP secure channel

heppth, 2019-06-17:
Hi, 
we are hoping since many months that the library can be used to connect to OPC-UA servers using encryption. We are testing continuously the newest versions, but we always get the same error. The[ OPC-UA server](https://www.unified-automation.com/de/downloads/opc-ua-servers/file/download/details/opc-ua-c-demo-server-v161-windows.html) from unified automation is not the main target but the easiest test case. Today I have created an easy test program by using the latest sample code. Please, can anyone give me a hint, what to do to connect to a server using encrypted connection and username/password authentication?

```
int main()
{
	// Load certificate generated by UA-Expert
	UA_ByteString* certificate = loadFile("uaexpert.der");
	UA_ByteString* privateKey = loadFile("uaexpert_key.pem");

	// Trust list;
	size_t trustListSize = 0;
	UA_STACKARRAY(UA_ByteString, trustList, trustListSize);

	// Revocation list
	UA_ByteString* revocationList = NULL;
	size_t revocationListSize = 0;

	// Config
	UA_Client* client = UA_Client_new();
	UA_ClientConfig* config = UA_Client_getConfig(client);
	config->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
	UA_ClientConfig_setDefaultEncryption(config, *certificate, *privateKey,
		trustList, trustListSize,
		revocationList, revocationListSize);

	// Set description as specified in certificate
	config->clientDescription.applicationName = UA_LOCALIZEDTEXT_ALLOC("en", "UaExpert");
	config->clientDescription.applicationType = UA_APPLICATIONTYPE_CLIENT;
	config->clientDescription.applicationUri = UA_STRING_ALLOC("urn:THINKPAD:UnifiedAutomation:UaExpert");

	// Connect
	UA_StatusCode retval = UA_Client_connect_username(client, "opc.tcp://localhost:48010", "root", "secret");
	if (retval != UA_STATUSCODE_GOOD) {
		UA_Client_delete(client);
		return EXIT_FAILURE;
	}
}

UA_ByteString* loadFile(const char* path)
{
	FILE* f = fopen(path, "r");
	if (f == NULL)
		return NULL;

	fseek(f, 0, SEEK_END);
	long fsize = ftell(f);
	fseek(f, 0, SEEK_SET);  /* same as rewind(f); */

	UA_ByteString* result = UA_ByteString_new();
	UA_ByteString_allocBuffer(result, (size_t)fsize + 1);
	memset(result->data, 0, result->length);
	fread(result->data, result->length, 1, f);
	fclose(f);

	return result;
}
```

Here is the output:
```
[2019-06-17 21:11:06.688 (UTC+0200)] info/client        Connecting to endpoint opc.tcp://DEV:48010
[2019-06-17 21:11:06.690 (UTC+0200)] debug/client       Initialize the SecurityPolicy context
[2019-06-17 21:11:06.691 (UTC+0200)] info/client        SecurityPolicy not specified -> use default #None
[2019-06-17 21:11:06.692 (UTC+0200)] warn/securitypolicy        No PKI plugin set. Accepting all certificates
[2019-06-17 21:11:06.768 (UTC+0200)] info/client        TCP connection established
[2019-06-17 21:11:06.770 (UTC+0200)] debug/network      Sent HEL message
[2019-06-17 21:11:06.773 (UTC+0200)] debug/network      Received ACK message
[2019-06-17 21:11:06.773 (UTC+0200)] debug/channel      Requesting to open a SecureChannel
[2019-06-17 21:11:06.775 (UTC+0200)] debug/channel      OPN message sent
[... 38 lines cut ...]
[opcuadumps.zip](https://github.com/open62541/open62541/files/3298496/opcuadumps.zip)

```

In wireshark I can the the following error response. 

```
Frame 18: 90 bytes on wire (720 bits), 90 bytes captured (720 bits) on interface 0
Ethernet II, Src: Microsof_b2:36:00 (00:15:5d:b2:36:00), Dst: QuantaCo_8e:1b:4e (e8:9a:8f:8e:1b:4e)
Internet Protocol Version 6, Src: 2003:ee:5bcf:c400:89ef:d370:e69f:2b3d, Dst: 2003:ee:5bcf:c400:b5ab:ee33:c597:5ef0
Transmission Control Protocol, Src Port: 48010, Dst Port: 58863, Seq: 29, Ack: 1951, Len: 16
OpcUa Binary Protocol
    Message Type: ERR
    Chunk Type: F
    Message Size: 16
    Error: 0x80ab0000 [BadInvalidArgument]
    Reason: [OpcUa Null String]
```

I have attached two wireshark dumps:

- One with the error response (generated by open62541 lib)
- One with a working connection (generated by UaExpert)

Running the tests on Ubuntu or Windows gives the same result. The above output is from the newest version of master branch.

heppth, 2020-09-03:
Sometimes you have to let a year go by and look at the problem again.

The bug is clearly in my test code:
```
UA_ByteString* loadFile(const char* path)
{
	FILE* f = fopen(path, "r");
	...
```
Because the file is opened via "r", it is opened in text mode. Certificates saved in DER format may have `\0` in the middle. Thus it is possible that only a part of the certificate is read in.

The fix is very simple:
`	FILE* f = fopen(path, "rb");`

Sorry for the trouble.

closed 2020-09-03 by heppth