open62541/open62541 #2577: Client can hang on blocking send calls

cookiejarlid, 2019-03-17:
## Description

UA_send loop in ua_network_tcp.c:connection_write is being called in blocking mode without a timeout, causing the client to hang arbitrarily and indefinitely.

## Background Information / Reproduction Steps

The issue can't be reproduced consistently as it is time sensitive.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master branch
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

For the non-async API, I would suggest setting the SO_SNDTIMEO socket option (as well as the SO_RCVTIMEO since any UA_recv blocking call will be affected by the same issue) and checking for E_AGAIN/E_WOULDBLOCK.

For the async API, which is currently and incorrectly making use of blocking socket function calls, I would suggest that the socket be set to non-blocking mode and a timeout check be added.

I would like to emphasize the gravity of calling blocking socket functions on async API calls. These calls should be required to return control to the user code immediately, and not be dependent on any possible network delays.

jpfr, 2019-03-18:
We are moving to full-on event-based networking in the WIP PR #2408.
Until then, feel free to propose code changes.

It seems these changes would all fit into /arch/ua_network_tcp.c with minimal API changes.

andranik-s, 2020-01-29:
@jpfr can you please elaborate on "full-on event-based networking"? Does that mean non blocking IO or something else? I'm asking because after taking a glance at https://github.com/open62541/open62541/pull/2408 it seems to me it still uses blocking IO: [UA_TCP_DataSocket_send, for example](https://github.com/mlgiraud/open62541/blob/9f9ef8fe43fbf446d37f0adb245bfe0e3c17902e/plugins/networking/ua_tcp_data_socket.c#L214-L227).

andranik-s, 2020-01-29:
Also in the context this issue what do you (@jpfr, @Pro) think about libev/libevent? I was thinking of proposing a PR that will add optional non blocking IO support using libev, but wanted to check with you first.

Just to clarify that PR will not change the current code (or at least I'll try to minimize the changes). All the changes will be in conditional inclusion blocks like UA_USE_LIBEV_EXPERIMENTAL.

Will you consider such PR?

jpfr, 2020-07-07:
@andranik-s , yes, the idea is to move to an event-based control flow.
The work is happening in the background. See #3409 for the upcoming network layer.
Libev would be easy to add once this has landed.

Closing this issue. Since v1.1 the client respects timeouts everywhere.

closed 2020-07-07 by jpfr

mlgiraud, 2020-07-17:
I would like to add that the receive logic should be fully asynchronous in the new networking code, but the send logic is not, as @andranik-s pointed out.
But with the coming changes it should be a lot easier to implement fully asynchronous send as well.