open62541/open62541 #2578: unknown type name 'UA_PubSubConnection' ESP32 PubSub

Marioam, 2019-03-18:
Hi everybody,
I am trying to port open62541 for ESP32 and using eclipse. I´m trying to establish a PubSub connection and there was no problem to publish from the ESP32. But when I try to implement a subscriber I get the following error messages:

`In file included from C:/Users/M/eclipse-workspace/PubSub/src/iot/bridge/opcua_sub.c:8:0:
C:/Users/M/eclipse-workspace/PubSub/src/iot/bridge/opcua_sub.h:49:54: error: unknown type name 'UA_PubSubConnection'
  void subscriptionPollingCallback(UA_Server *server, UA_PubSubConnection *connection);

C:/Users/M/eclipse-workspace/PubSub/src/iot/bridge/opcua_sub.c: In function 'opcua_sub_task':
C:/Users/M/eclipse-workspace/PubSub/src/iot/bridge/opcua_sub.c:43:2: error: unknown type name 'UA_PubSubConnection'
  UA_PubSubConnection *connection = UA_PubSubConnection_findConnectionbyId(server, connectionIdent);

C:/Users/M/eclipse-workspace/PubSub/src/iot/bridge/opcua_sub.c:43:36: error: implicit declaration of function 'UA_PubSubConnection_findConnectionbyId' [-Werror=implicit-function-declaration]
  UA_PubSubConnection *connection = UA_PubSubConnection_findConnectionbyId(server, connectionIdent);

C:/Users/M/eclipse-workspace/PubSub/src/iot/bridge/opcua_sub.c:43:36: warning: initialization makes pointer from integer without a cast [-Wint-conversion]
C:/Users/M/eclipse-workspace/PubSub/src/iot/bridge/opcua_sub.c:45:32: error: request for member 'channel' in something not a structure or union
   UA_StatusCode rv = connection->channel->regist(connection->channel, NULL, NULL);

C:/Users/M/eclipse-workspace/PubSub/src/iot/bridge/opcua_sub.c:45:60: error: request for member 'channel' in something not a structure or union
   UA_StatusCode rv = connection->channel->regist(connection->channel, NULL, NULL);

C:/Users/M/eclipse-workspace/PubSub/src/iot/bridge/opcua_sub.c:48:62: error: 'subscriptionPollingCallback' undeclared (first use in this function)
    UA_Server_addRepeatedCallback(server, (UA_ServerCallback) subscriptionPollingCallback, connection, 100, &subscriptionCallbackId);

C:/Users/M/eclipse-workspace/PubSub/src/iot/bridge/opcua_sub.c:48:62: note: each undeclared identifier is reported only once for each function it appears in
C:/Users/M/eclipse-workspace/PubSub/src/iot/bridge/opcua_sub.c: At top level:
C:/Users/M/eclipse-workspace/PubSub/src/iot/bridge/opcua_sub.c:59:53: error: unknown type name 'UA_PubSubConnection'
 void subscriptionPollingCallback(UA_Server *server, UA_PubSubConnection *connection)`

Thanks for the Help!

 - [ ] open62541 Version (release number or git tag): 0.3-rc2-375-g9bdb868
 - [ ] Operating system: FreeRTOS/LwIP

jpfr, 2019-03-26:
PubSub is not implemented on the 0.3 branch.
Please use a current master.

closed 2019-03-26 by jpfr

Marioam, 2019-03-27:
Thanks for the response.
I tried it with the current master branche but the same errors occure when trying to implement a Subscriber. Also the Publisher I have implemented, wich was working previosly, does not work anymore using the current master branch.  

`[1970-01-01 00:00:04.586 (UTC+0000)] info/session       Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: No TypeDefinition for i=15303; Use the default TypeDefinition
for the Variable/Object
[1970-01-01 00:00:04.866 (UTC+0000)] info/userland      PubSub channel requested
[1970-01-01 00:00:04.908 (UTC+0000)] info/session       Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: Node type i=21105 not found
[1970-01-01 00:00:04.919 (UTC+0000)] info/network       TCP network layer listening on opc.tcp://ESP32:4840/
`
After these messages it just stops without publishing anything.
Hope sombody has an suggestion what i´m doing wrong.
Thanks!