open62541/open62541 #2579: Can i exclude References from browsing by type?

tobiaslocker, 2019-03-18:
- [x] open62541 Version v0.3.0:
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

I can't find any Information on excluding references from browsing by type. Background is, that an unfiltered browse results in an Infinite recursion on References of type `HasNotifier`. Tested with the following public server: `opc.tcp://opcua.demo-this.com:51210/UA/SampleServer` with the following implementation for browsing which is mainly taken from example code.

```
UA_BrowseRequest Node::make_browse_request(const BrowseResultMask &mask) {
  UA_BrowseRequest browse_req;
  UA_BrowseRequest_init(&browse_req);
  browse_req.requestedMaxReferencesPerNode = 0;
  browse_req.nodesToBrowse = UA_BrowseDescription_new();
  browse_req.nodesToBrowseSize = 1;
  browse_req.nodesToBrowse[0].nodeId = m_node_id;
  browse_req.nodesToBrowse[0].resultMask = static_cast<u_int32_t>(mask);
  return browse_req;
}

std::vector<ReferenceDescription>
Node::browse_children(const BrowseResultMask &mask) {
  std::vector<ReferenceDescription> children;
  auto browse_request = make_browse_request(mask);
  UA_BrowseResponse bResp =
      UA_Client_Service_browse(m_client.get(), browse_request);
  for (size_t i = 0; i < bResp.resultsSize; ++i) {
    for (size_t j = 0; j < bResp.results[i].referencesSize; ++j) {
      UA_ReferenceDescription ref = bResp.results[i].references[j];
      children.push_back(ReferenceDescription(ref));
    }
  }
  UA_BrowseRequest_deleteMembers(&browse_request);
  UA_BrowseResponse_deleteMembers(&bResp);
  return children;
}
```

jpfr, 2019-03-18:
browse_req.nodesToBrowse[0].referenceTypeId = myRefType;
browse_req.nodesToBrowse[0].includeSubtypes = true;

tobiaslocker, 2019-03-18:
So there can only one type of reference (and its subtypes) be browsed for a given start node? How could i specify more that one type?

jpfr, 2019-03-18:
In that case you have to browse several times.
However, since browse results are only one level deep, you can filter before recursing into the next browse request.

tobiaslocker, 2019-03-18:
That seems to be convenient. Thanks for the quick reply

closed 2019-03-18 by tobiaslocker