open62541/open62541 #2581: Problem in OPCUA client c++

ghazivakili, 2019-03-18:
Dear All,

I am creating middleware with opc-ua and I have some problem in debugging of my problem,

I developed the opc-ua server with 1000 node and 1000 client also is writing data on the server (I know this is not very common to write from client to server in opc-ua but this is a test)

and 1000 client is creating in one program from another pc. 

when I am connecting with 40 clients it is working but when I add more than 40 it is showing this error : 

client error error code-> -2141847552
also in the server, i have this message always 
/** server log **/
[03/18/2019 13:07:23.630] info/channel  Connection 18 | SecureChannel 361 | OpenSecureChannel: Opened SecureChannel
[03/18/2019 13:07:23.631] info/session  Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
[03/18/2019 13:07:23.634] info/channel  Connection 18 | SecureChannel 361 | CloseSecureChannel
**when the connection has rejected in server writen :**
[03/18/2019 16:50:58.414] info/network  Connection 57 | Connection closed from remote
//*** end **/

server and client is open62541 v0.2
the operating system is ubuntu 18.04

## Description

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [x] Other OPC UA SDKs used (client or server): 
 - [x] Operating system:
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ghazivakili, 2019-03-19:
I guess I found the problem, the problem is 
         config.maxSessions = 2000;
        config.maxSecureChannels= 2000;

in the server, but i found anotehr problem, when i am running the server i create a thread for each node separately, and i tried to add mutex luck when i am reading data from the share value. 
but some time i get the "Segmentation fault (core dumped)" and when i check with debugger shows i have a problem in this function "UA_StatusCode check = UA_Server_readValue(server, args->NodeId, &value);" but I lock the thread before and after ??
do you have any idea why it happens?

ccvca, 2019-03-19:
When you need to debug SegFaults the Tools valgrind (Linux) or DrMemory (Linux and Windows) very good, as they show, who has freed the memory in case of illegal accesses.

fanucwj, 2019-05-25:
hello ,may i ask you a question?
expect for you response
thank you advance

Pro, 2019-05-27:
Your config limits the number of sessions to 2000. You probably reach that limit.

Also, the 0.2 branch is very old and there have been many bugfixes in 0.3. Even better use the current master branch.

If you are using multiple threads, you have to make sure that every call, where you are using the `server` variable, is mutexed. Otherwise you will have quite some issues.

To further debug your segfault, either use valgrind or gdb or clang Address Sanitizer to get a full stacktrace of the issue.

In your title you mention `C++`. We do not have any officially supported C++ stack. Please try to contact the developer of your used stack to fix the issue with you together.

closed 2019-05-27 by Pro