open62541/open62541 #2583: VS2005 compile fails with Open62541 v0.3.0

JesperVilladsen, 2019-03-19:
Hi
We currently have Open62541 v.0.2.0 release candidate running on a Windows CE 6.0 X86 platform and VS2005 with the most basic mirroring functionality, however I am to work on a cross country project where pub/sub is needed, thus I want to compile the 0.3.0 version. The compiler gives me the following errors:

[image]

I am new to gitHub so the solution might already be available? Please help

Best regards
Jesper

jpfr, 2019-03-19:
VS2005 does not support C99 constructs.
But you can compile the library as C++98.

This can be forced in cmake or via the VS2005 project settings.

JesperVilladsen, 2019-03-19:
Thank you,
seems to work

closed 2019-03-19 by JesperVilladsen