open62541/open62541 #2584: Weird behaviour displaying enum strings in Server

juangburgos, 2019-03-19:
## Description

I have been trying to create my own **enum** as discussed in #2032. I copied and cleaned the code and noticed a weird behaviour in `UA_Server_addDataTypeNode`. 

Under my understanding, if the agrument `const UA_NodeId requestedNewNodeId` is `UA_NODEID_NULL`, then the server assigns a node id through the `UA_NodeId *outNewNodeId` argument. Else it assigns the node id that we pass in. In principle, either of the two options should **not** change the runtime behaviour of the server.

Nevertheless when creating an enum, if using `UA_NODEID_NULL`, the client **does not display** the enum string in the value:

[image]

But then using something else, (e.g. `UA_NODEID_STRING (1, "TestEnumDataType")`), then the client **does display** the enum string correctly:

[image]

Just by changing one single argument we get a different runtime behaviour? Why would this be?

The full code is attached below.

## Code

 ```c
#include <signal.h>
#include <stdio.h>
#include "open62541.h"

UA_Boolean running = true;

typedef enum
{
	TestEnum_NONE = 0,
	TestEnum_ONE  = 1,
	TestEnum_TWO  = 2,
	__TestEnum_FORCE32BIT = 0x7fffffff
} TestEnum;
UA_STATIC_ASSERT(sizeof(TestEnum) == sizeof(UA_Int32), enum_must_be_32bit);

static void stopHandler(int sig) {
	UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
	running = false;
}

static UA_NodeId newNodeId;
static void addTestEnumType(UA_Server* server) {
	UA_DataTypeAttributes ddaatt = UA_DataTypeAttributes_default;
	ddaatt.description = UA_LOCALIZEDTEXT((char*)(""), (char*)("TestEnumDataType"));
	ddaatt.displayName = UA_LOCALIZEDTEXT((char*)(""), (char*)("TestEnumDataType"));

	UA_StatusCode st = UA_Server_addDataTypeNode(
		server,
		UA_NODEID_NULL, // NOTE : with UA_NODEID_STRING (1, "TestEnumDataType") it works
		UA_NODEID_NUMERIC(0, UA_NS0ID_ENUMERATION),
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
		UA_QUALIFIEDNAME (1, (char*)("TestEnumDataType")),
		ddaatt,
		NULL,
		&newNodeId);
	assert(st == UA_STATUSCODE_GOOD);

	UA_VariableAttributes pattr = UA_VariableAttributes_default;
	pattr.description = UA_LOCALIZEDTEXT((char*)(""), (char*)("EnumStrings"));
	pattr.displayName = UA_LOCALIZEDTEXT((char*)(""), (char*)("EnumStrings"));
	pattr.dataType    = UA_TYPES[UA_TYPES_LOCALIZEDTEXT].typeId;

	UA_UInt32 axesDeminsion[1] = { 0 };
	pattr.valueRank            = 1; 
	pattr.arrayDimensionsSize  = 1;
	pattr.arrayDimensions      = axesDeminsion;

	UA_LocalizedText test2[3];
	test2[0] = UA_LOCALIZEDTEXT((char*)(""), (char*)("NONE"));
	test2[1] = UA_LOCALIZEDTEXT((char*)(""), (char*)("ONE"));
	test2[2] = UA_LOCALIZEDTEXT((char*)(""), (char*)("TWO"));

	UA_Variant_setArray(&pattr.value, test2, 3, &UA_TYPES[UA_TYPES_LOCALIZEDTEXT]);

	UA_Server_addVariableNode(
		server,
		UA_NODEID_NULL, 
		newNodeId, 
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY),
		UA_QUALIFIEDNAME (0, (char*)("EnumStrings")),
		UA_NODEID_NUMERIC(0, UA_NS0ID_PROPERTYTYPE), 
		pattr, 
		NULL,
		NULL);
}

static void addVariable(UA_Server* server) {
	TestEnum p;
	p = TestEnum_TWO;
	UA_VariableAttributes vattr = UA_VariableAttributes_default;

	vattr.description = UA_LOCALIZEDTEXT((char*)(""), (char*)("TestEnum"));
	vattr.displayName = UA_LOCALIZEDTEXT((char*)(""), (char*)("TestEnum"));
	vattr.dataType    = newNodeId;
	vattr.valueRank   = -1;
	vattr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
	UA_Variant_setScalar(&vattr.value, &p, &UA_TYPES[UA_TYPES_INT32]);

	UA_Server_addVariableNode(server,
		UA_NODEID_STRING (1, "TestEnum"), 
		UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
		UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
		UA_QUALIFIEDNAME (1, "TestEnum"), 
		UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
		vattr, 
		NULL,
		NULL);
}

int main(void) {
	signal(SIGINT, stopHandler);
	signal(SIGTERM, stopHandler);

	UA_ServerConfig* config = UA_ServerConfig_new_default();
	UA_Server* server = UA_Server_new(config);
	addTestEnumType(server);
	addVariable(server);

	UA_StatusCode retval = UA_Server_run(server, &running);
	UA_Server_delete(server);
	UA_ServerConfig_delete(config);
	return (int)retval;
}
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [x] Other OPC UA SDKs used (client or server): UA Expert Client
 - [x] Operating system: Windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

juangburgos, 2020-03-20:
Its been a year, maybe close it due to inactivity

closed 2020-03-20 by juangburgos

JensMertelmeyer, 2021-09-09:
The difference is that `UA_NODEID_NULL` will have a numeric NodeId from **namespace zero** returned, whereas `UA_NODEID_STRING (1, "TestEnumDataType")` explicitly asks for something from **namespace 1**.

I have defined
`#define CREATE_NUMERIC_NODEID UA_NODEID_NUMERIC(myNamespaceIndex, 0)`
and use `CREATE_NUMERIC_NODEID` instead of `UA_NODEID_NULL` everywhere.

juangburgos, 2021-09-09:
@JensMertelmeyer , thanks that is clear, but why would this define whether the enum string is displayed or not?

JensMertelmeyer, 2021-09-13:
Whether it pretty-prints enums is up to the client.
I tested the following clients:

1. [Traeger OPC Watch](https://docs.traeger.de/de/software/sdk/opc-ua/net#download)
2. [Softing dataFEED OPC Client](https://industrial.softing.com/products/opc-ua-and-opc-classic-sdks/opc-ua-demo-client.html)
3. [UnifiedAutomation UaExpert](https://www.unified-automation.com/products/development-tools/uaexpert.html)
4. [Prosys OPC UA Browser](https://prosysopc.com/products/opc-ua-browser/)
5. [Integration Objects OPC UA Client](https://integrationobjects.com/sioth-opc/sioth-opc-unified-architecture/opc-ua-client/)

UaExpert and Softing dataFEED were the only ones that even seem attempt to lookup the enum definition. Both require that `EnumStrings` have their name in *namespace zero*. I have been unable to find the rule for this, but this ought to be the way.

The whole namespace thing is still confusing to me. I bought two books about OPC UA and I'm still none the wiser 🤔 

Maybe it's best to just accept it. A unique name is made up of namespace *and* the name itself, and where should those clients look, if not in the Namespace zero?

juangburgos, 2021-09-13:
Funny that you mention it, I also find it rather confusing. 

My initial undestanding was that *namespace 0* is for nodes that "are named" by the local server (including the ones defined by the core specification). And any other namespace is for nodes named by other "naming authorities". 

This is a little vague and there are no practical examples provided. But then I saw a server that used the [NamespaceArray node of the Server Object](https://reference.opcfoundation.org/v104/Core/docs/Part5/6.3.1/) to list a bunch of URIs. This array is a list of URIs that has indexes from `0` to `N`, so apparently the *namespace index* of a node id is the index in this array.

So if we take the node id of any node in the server, and find that the namespace index is, for example, `5` then can go to the *NamespaceArray node of the Server Object*, look at index `5` of that array, and find the URI (most commonly a website) of the authority that named (defined) that node. Then maybe from that URI we can autmatically download more information (meta-information) regaring that node.

Now, maybe clients only resolve enums of namespace 0 because they are sure those enums definitions can be found in the same server, and they do not try to resolve enums from different namesapces because they might not be found in the server, and they would have to resolve them using the URI. I don't know.

* <https://reference.opcfoundation.org/v104/Core/docs/Part3/8.2.2/>

JensMertelmeyer, 2021-09-13:
> But then I saw a server that used the [NamespaceArray node of the Server Object](https://reference.opcfoundation.org/v104/Core/docs/Part5/6.3.1/) to list a bunch of URIs. This array is a list of URIs that has indexes from `0` to `N`

Yes, exactly. I recently asked a related question about namespaces here: https://github.com/open62541/open62541/issues/4622
The gist of it is that [UA_Server_addNamespace(..)](https://open62541.org/doc/1.2/server.html?highlight=ua_server_addnamespace#utility-functions) to add your own namespace, and but all *your* stuff in there. 

> Now, maybe clients only resolve enums of namespace 0 because they are sure those enums definitions can be found in the same server

The `EnumStrings` property is not *our* stuff, it's part of the standard UA model and therefor goes into namespace zero. That's how I understand it. The instance of the enumeration itself (not the type definition) goes into your namespace, just like your example code already does it by using `UA_Server_addVariableNode(..)`.

juangburgos, 2021-09-13:
Mmm... seems I am going to have to add an API for `UA_Server_addNamespace` in [QUaServer](https://github.com/juangburgos/QUaServer). Also move nodes in [QUaModbusClient](https://github.com/juangburgos/QUaModbusClient) to another namespace. Thanks for the info.

LeeFRC2016, 2023-06-23:
I updated the code above to run with version 1.3.5 (installed on Ubuntu 20.04 with apt-get, gcc 9.4.0). Now, 'UA_Server_addVariableNode' fails with 'BadTypeMismatch'. It seems like this was working in 2020.....

Can this use case add added to 'examples/custom_datatype/server_types_custom.c' to keep it current with release updates?

```
#include <signal.h>
#include <stdio.h>
#include <open62541/server.h>
#include <open62541/plugin/log_stdout.h>
#include <open62541/server_config_default.h>

UA_Boolean running = true;

typedef enum
{
    TestEnum_NONE = 0,
    TestEnum_ONE = 1,
    TestEnum_TWO = 2,
    __TestEnum_FORCE32BIT = 0x7fffffff
} TestEnum;
UA_STATIC_ASSERT(sizeof(TestEnum) == sizeof(UA_Int32), enum_must_be_32bit);

static void stopHandler(int sig)
{
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
    running = false;
}

static UA_NodeId newNodeId;
static void addTestEnumType(UA_Server *server)
{
    UA_DataTypeAttributes ddaatt = UA_DataTypeAttributes_default;
    ddaatt.description = UA_LOCALIZEDTEXT((char *)(""), (char *)("TestEnumDataType"));
    ddaatt.displayName = UA_LOCALIZEDTEXT((char *)(""), (char *)("TestEnumDataType"));

    UA_StatusCode st = UA_Server_addDataTypeNode(
        server,
        UA_NODEID_STRING(1, (char *)"TestEnumDataType"), // NOTE : with UA_NODEID_STRING (1, "TestEnumDataType") it works
        UA_NODEID_NUMERIC(0, UA_NS0ID_ENUMERATION),
        UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
        UA_QUALIFIEDNAME(1, (char *)("TestEnumDataType")),
        ddaatt,
        NULL,
        &newNodeId);
    if (st != UA_STATUSCODE_GOOD)
    {
        printf("UA_Server_addDataTypeNode failed %s\n", UA_StatusCode_name(st));
    }
    assert(st == UA_STATUSCODE_GOOD);

    UA_VariableAttributes pattr = UA_VariableAttributes_default;
    pattr.description = UA_LOCALIZEDTEXT((char *)(""), (char *)("EnumStrings"));
    pattr.displayName = UA_LOCALIZEDTEXT((char *)(""), (char *)("EnumStrings"));
    pattr.dataType = UA_TYPES[UA_TYPES_LOCALIZEDTEXT].typeId;

    UA_UInt32 axesDeminsion[1] = {0};
    pattr.valueRank = 1;
    pattr.arrayDimensionsSize = 1;
    pattr.arrayDimensions = axesDeminsion;

    UA_LocalizedText test2[3];
    test2[0] = UA_LOCALIZEDTEXT((char *)(""), (char *)("NONE"));
    test2[1] = UA_LOCALIZEDTEXT((char *)(""), (char *)("ONE"));
    test2[2] = UA_LOCALIZEDTEXT((char *)(""), (char *)("TWO"));

    UA_Variant_setArray(&pattr.value, test2, 3, &UA_TYPES[UA_TYPES_LOCALIZEDTEXT]);

    st = UA_Server_addVariableNode(
        server,
        UA_NODEID_NULL,
        newNodeId,
        UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY),
        UA_QUALIFIEDNAME(0, (char *)("EnumStrings")),
        UA_NODEID_NUMERIC(0, UA_NS0ID_PROPERTYTYPE),
        pattr,
        NULL,
        NULL);

    if (st != UA_STATUSCODE_GOOD)
    {
        printf("UA_Server_addDataTypeNode failed %s\n", UA_StatusCode_name(st));
    }
    assert(st == UA_STATUSCODE_GOOD);
}

static void addVariable(UA_Server *server)
{
    TestEnum p;
    p = TestEnum_TWO;
    UA_VariableAttributes vattr = UA_VariableAttributes_default;

    vattr.description = UA_LOCALIZEDTEXT((char *)(""), (char *)("TestEnum"));
    vattr.displayName = UA_LOCALIZEDTEXT((char *)(""), (char *)("TestEnum"));
    vattr.dataType = newNodeId;
    vattr.valueRank = -1;
    vattr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
    UA_Variant_setScalar(&vattr.value, &p, &UA_TYPES[UA_TYPES_INT32]);

    UA_StatusCode st = UA_Server_addVariableNode(server,
                                                 UA_NODEID_STRING(1, (char *)("TestEnum")),
                                                 UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                                                 UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                                                 UA_QUALIFIEDNAME(1, (char *)("TestEnum")),
                                                 UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
                                                 vattr,
                                                 NULL,
                                                 NULL);

    if (st != UA_STATUSCODE_GOOD)
    {
        printf("UA_Server_addVariableNode failed %s\n", UA_StatusCode_name(st));
    }
    assert(st == UA_STATUSCODE_GOOD);
}

int main(void)
{
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));
    addTestEnumType(server);
    addVariable(server);

    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    return (int)retval;
}

```