open62541/open62541 #2586: Fail to compile the autoid nodeset

basyskom-meerkoetter, 2019-03-19:
I am trying to create a server with the nodeset defined by the autoid spec.

As a first step I've downloaded the xml description from here.
https://opcfoundation.org/UA/schemas/AutoID/1.0/

I had the issue that I've not found a bsd file. See here: https://github.com/open62541/open62541/issues/2582

I've extracted the embedded bsd information and stored it in AutoID/Opc.Ua.AutoID.Types.bsd.

I've used the server_nodeset_plcopen example as a testing ground.

`diff --git a/examples/nodeset/CMakeLists.txt b/examples/nodeset/CMakeLists.txt
index 3eb417f..c706dc5 100644
--- a/examples/nodeset/CMakeLists.txt
+++ b/examples/nodeset/CMakeLists.txt
@@ -98,10 +98,21 @@ if(UA_NAMESPACE_ZERO STREQUAL "FULL")
         INTERNAL
     )
 
+    ua_generate_nodeset_and_datatypes(
+        NAME "autoid"
+        FILE_CSV "${FILE_CSV_DIRPREFIX}/AutoID/Opc.Ua.AutoID.NodeIds.csv"
+        FILE_BSD "${FILE_BSD_PLCOPEN_DIRPREFIX}/AutoID/Opc.Ua.AutoID.Types.bsd"
+        NAMESPACE_IDX 5
+        FILE_NS "${FILE_NS_DIRPREFIX}/AutoID/Opc.Ua.AutoID.NodeSet2.xml"
+        DEPENDS "di"
+        INTERNAL
+    )
+
     add_example(server_nodeset_plcopen server_nodeset_plcopen.c
                 ${PROJECT_BINARY_DIR}/src_generated/ua_types_di_generated.c
                 ${PROJECT_BINARY_DIR}/src_generated/ua_namespace_di.c
-                ${PROJECT_BINARY_DIR}/src_generated/ua_namespace_plc.c)
+                ${PROJECT_BINARY_DIR}/src_generated/ua_namespace_plc.c
+                ${PROJECT_BINARY_DIR}/src_generated/ua_namespace_autoid.c)
     add_dependencies(server_nodeset_plcopen open62541-generator-ns-plc)`

The build now fails with the following error:

`[70/251] Generating ../../src_generated/ua_namespace_autoid.c, ../../src_generated/ua_namespace_autoid.h
FAILED: cd /home/fmee/projects/foss/open62541/examples/nodeset && /usr/bin/python /home/fmee/projects/foss/open62541/tools/nodeset_compiler/nodeset_compiler.py --internal-headers --types-array=UA_TYPES --types-array=UA_TYPES_DI --types-array=UA_TYPES_AUTOID --existing=/home/fmee/projects/foss/open62541/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml --existing=/home/fmee/projects/foss/open62541/deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml --xml=/home/fmee/projects/foss/open62541/deps/ua-nodeset/AutoID/Opc.Ua.AutoID.NodeSet2.xml /home/fmee/projects/foss/open62541/src_generated/ua_namespace_autoid
INFO:__main__:Preprocessing (existing) /home/fmee/projects/foss/open62541/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing (existing) /home/fmee/projects/foss/open62541/deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml
INFO:__main__:Preprocessing /home/fmee/projects/foss/open62541/deps/ua-nodeset/AutoID/Opc.Ua.AutoID.NodeSet2.xml
ERROR:datatypes:ns=2;i=6147: Expected XML element with tag Grade but found EncodingMask instead
ERROR:datatypes:ns=2;i=6147: Expected XML tag <ExtensionObject> for multipart type, but found CodeType instead.
ERROR:datatypes:ns=2;i=6147: Expected XML element with tag Symbology but found ScanData instead
ERROR:datatypes:ns=2;i=6147: Expected XML element with tag ImageId but found Timestamp instead
ERROR:datatypes:ns=2;i=6128: Expected XML element with tag NMEA but found SwitchField instead
ERROR:datatypes:ns=2;i=6128: Expected encoding ['Local', [['X', [u'Double'], None], ['Y', [u'Double'], None], ['Z', [u'Double'], None], ['Timestamp', [u'UtcTime', [u'DateTime'], None], None], ['DilutionOfPrecision', [u'Double'], None], ['UsefulPrecicision', [u'Int32'], None]], None] but found none in body.
ERROR:datatypes:ns=2;i=6128: Expected encoding ['WGS84', [['N/S Hemisphere', [u'String'], None], ['Latitude', [u'Double'], None], ['E/W Hemisphere', [u'String'], None], ['Longitude', [u'Double'], None], ['Altitude', [u'Double'], None], ['Timestamp', [u'UtcTime', [u'DateTime'], None], None], ['DilutionOfPrecision', [u'Double'], None], ['UsefulPrecisionLatLon', [u'Int32'], None], ['UsefulPrecisionAlt', [u'Int32'], None]], None] but found none in body.
ERROR:datatypes:ns=2;i=6128: Expected encoding ['Name', [u'LocationName', [u'String'], None], None] but found none in body.
INFO:__main__:Generating Code for Backend: open62541
Traceback (most recent call last):
  File "/home/fmee/projects/foss/open62541/tools/nodeset_compiler/nodeset_compiler.py", line 205, in <module>
    generateOpen62541Code(ns, args.outputFile, args.generate_ns0, args.internal_headers, args.typesArray, args.encode_binary_size)
  File "/home/fmee/projects/foss/open62541/tools/nodeset_compiler/backend_open62541.py", line 238, in generateOpen62541Code
    code = generateNodeCode_begin(node, nodeset, generate_ns0, parentref, encode_binary_size, code_global)
  File "/home/fmee/projects/foss/open62541/tools/nodeset_compiler/backend_open62541_nodes.py", line 416, in generateNodeCode_begin
    [code1, codeCleanup1, codeGlobal1] = generateVariableNodeCode(node, nodeset, encode_binary_size)
  File "/home/fmee/projects/foss/open62541/tools/nodeset_compiler/backend_open62541_nodes.py", line 113, in generateVariableNodeCode
    [code1, codeCleanup1, codeGlobal1] = generateValueCode(node.value, nodeset.nodes[node.id], nodeset)
  File "/home/fmee/projects/foss/open62541/tools/nodeset_compiler/backend_open62541_nodes.py", line 316, in generateValueCode
    isArrayElement=True)
  File "/home/fmee/projects/foss/open62541/tools/nodeset_compiler/backend_open62541_nodes.py", line 168, in generateExtensionObjectSubtypeCode
    logger.debug("Value    " + str(node.value))
UnicodeEncodeError: 'ascii' codec can't encode character u'\u2019' in position 14: ordinal not in range(128)
[70/251] Building C object CMakeFiles/open62541-object.dir/src_generated/ua_namespace0.c.o`

Any ideas?

https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports
https://help.github.com/articles/basic-writing-and-formatting-syntax/

## Description

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master, 88980e971ff7d2a
 - [NA] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: server_nodeset_plcopen
 - [NA] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [NA] Wireshark network dump attached
 - [-] Self-contained code example attached
 - [-] Critical issue

Stasik0, 2019-03-19:
Seems to be a unicode problem \u2019 [is this quotation mark](https://www.fileformat.info/info/unicode/char/2019/index.htm)... maybe within the nodeset file e.g. here "The code doesn’t match the given target value"

You could try to run your files through https://pypi.org/project/Unidecode/ to get rid of unicode characters

Pro, 2019-03-20:
Your changes themselves look good. Which python version are you using?

basyskom-meerkoetter, 2019-03-20:
> Your changes themselves look good. Which python version are you using?

python --version says: Python 2.7.12

teccheck, 2025-03-18:
The AutoID nodeset should compile now. For nodesets without a bsd we currently extract the bsd from a base64 encoded blob from the NodeSet2.xml