open62541/open62541 #2587: Use NodeSet2.xml Definition fields instead of .bsd files for datatype generator

Pro, 2019-03-20:
Currently we require Types.bsd files for the datatype_generator.py to generate the corresponding custom datatypes.

In newer NodeSets, the DataType Definition is stored inside the NodeSet2.xml. (e.g. see #2582)

We should use this Definition instead of the .bsd files.
This can be achieved by changing the nodeset_compiler in a way, that it also creates the types files, since it already parses all the information. For this the current datatype_generator.py code should be merged into the nodeset compiler.

jpfr, 2019-03-20:
We could move generate_datatypes.py as a method into the nodeset compiler.
So that it takes the raw string as input and not a file.
Then, generate_datatypes.py can remain as a standalone wrapper for those with existing .bsd files.

Can we keep the .bsd files for the builtin types baked into the library?
I find it quite educational to show the xml.

Pro, 2019-03-20:
> We could move generate_datatypes.py as a method into the nodeset compiler.
> So that it takes the raw string as input and not a file.

Maybe we can have the following solution:

Provide a command line switch `datatypes-source` which allows to select options how the datatypes should be generated:
- `bsd` Use the .bsd file which is passed as another parameter
- `encoded` Use the base64 encoded string inside the NodeSet2.xml
- `definition` Use the definition tags of each DataType

Like this we support old Nodesets, and also newer ones which do not come with a .bsd file and also without a base64 encoded type definition.
Also we can keep our builtin types definition as .bsd.

Lo-TW, 2021-09-08:
> you can use UaModeler: Open the NodeSet2.xml, confirm that it should be converted, and then just export it again. With this, UaModeler will also export a .bsd file

Hi @Pro,
I download UaModeler Version 1.6.5 472, and I try this way. 
1. The UaModeler can't open xml file directly.
2. If I open a project, UaModeler can export xml and export NodeIds(csv) only.

How to export the bsd file in the UaModeler？
Thanks！