open62541/open62541 #2589: Compile fail with UNICODE

JesperVilladsen, 2019-03-20:
Hi,
I am using v0.3.0 and have it compiled for VS2005 for WinCE, however when I added it to my mainproject i received LNK2019 errors on __imp_FormatMessage as my mainproject includes UNICODE. 
I have tried compiling Open62541 with UNICODE but without any success. I receive the following error:  error C2664: 'FormatMessageW' : cannot convert parameter 5 from 'LPSTR' to 'LPWSTR' which is %error_str.
Did I forget to add some preprocessor definitions or can how do I solve this?

jpfr, 2019-03-20:
Are you missing a library to link against?
In which line does the error occur if UNICODE is enabled? (whatever that means...)

JesperVilladsen, 2019-03-21:
I will try to explain :)
In my application Unicode character set is requied, hence I want to compile the Open62541 with "Character Set" set to "Use Unicode Character Set" in my configuration properties.
I am using vs2005 and my application mainly runs WinCE, however also use Win32 for debug purposes. 
When i set Unicode as character set Open62541 does not build and I get the following error:
[image]
As far as I understand, FormatMessage (line 15570 in open62541.h) should choose to use FormatMessageW when unicode character set is chosen and otherwise FormatMessageA. However, FormatMessageW uses a type LPWSTR as fifth input parameter instead of LPSTR, this I cannot find a solution for. I have tried converting the types using altbase.h functionality.
I am no code-wizard, so all help is very much appreciated :)

jpfr, 2019-03-26:
Lpwstr is utf16.
Can you use utf8 for Unicode? We assume utf8 for all strings. This is also defined in the opc ua specification.

https://stackoverflow.com/questions/38316563/how-to-convert-lpwstr-to-char-with-utf-8-encoding

teccheck, 2025-03-18:
The standard defines utf-8 as the default for chars it can be used to encode Unicode symbols. @jpfr provided a way to convert between utf-16 and utf-8, so this should be done

closed 2025-03-18 by teccheck