open62541/open62541 #2590: Add warn_unused_result to function declarations

Pro, 2019-03-20:
While writing the commit https://github.com/open62541/open62541/pull/2588/commits/fd9c61ba0e9e81007c3fb390792effe644280fba I thought that we should introduce a warning, if one uses server functions, e.g., `UA_Server_addVariableNode` and ignores the return value.

The `-Wunused-result` warning is enabled in our stack by using `-Wall`. During development we should also get an error, if one does not check the return value.

To achieve this we have to annotate our methods with the `__attribute__((warn_unused_result))` attribute.

E.g.:

```c
#define UA_UNUSED_RESULT __attribute__((warn_unused_result))

static UA_INLINE UA_UNUSED_RESULT UA_StatusCode
UA_Server_addVariableNode(UA_Server *server, const UA_NodeId requestedNewNodeId,
```

Which results in compile errors like:
```
/repos/open62541/examples/server_ctt.c:122:5: error: ignoring return value of ‘UA_Server_addVariableNode’, declared with attribute warn_unused_result [-Werror=unused-result]
     UA_Server_addVariableNode(server, myIntegerNodeId, parentNodeId, parentReferenceNodeId,
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                               myIntegerName, baseDataVariableType, myVar, NULL, NULL);
                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/repos/open62541/examples/server_ctt.c:135:5: error: ignoring return value of ‘UA_Server_addVariableNode’, declared with attribute warn_unused_result [-Werror=unused-result]
     UA_Server_addVariableNode(server, myInteger2NodeId, parentNodeId, parentReferenceNodeId,
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                               myInteger2Name, baseDataVariableType, myVar, NULL, NULL);
                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

@open62541/open62541-core any opinion on this?

Infinity95, 2019-03-20:
This is a good idea.
Isn't it possible to enable this as default without using the __attribute__?

Pro, 2019-03-20:
I did some tests, but somehow I was not able to get the warning without explicitly adding the attribute...

Here's a simple example where it does not work:
https://godbolt.org/z/OTOfZD

Compared to (with the attribute):
https://godbolt.org/z/saHXQv

See also:
https://stackoverflow.com/q/13334324/869402

Maybe in combination with this:
https://github.com/sinelaw/elfs-clang-plugins#warn_unused_result

jpfr, 2019-03-21:
We have something similar already.
https://github.com/open62541/open62541/blob/ecee9f241e361bb40c952ddc197e9787f321d675/arch/ua_architecture_definitions.h#L139

If `UA_INTERNAL` is defined before the first #include, additional methods are tested for unchecked return values. But that is just a hack atm.

We should not force these tests on all users.
Maybe we can make the *global* definition of UA_UNUSED_RESULT conditional to a configuration-macro definition. So that users can enable the tests. And they are always enabled in internal code by setting the macro before the first include.

closed 2025-03-25 by teccheck