open62541/open62541 #2591: Reading values of tags in Kepware

abdellah96, 2019-03-20:
Hello,
I'm using open62541 to read values from industrial tags. The nodeId string of the tag i'm willing to read has the format : "ns=2;s=$tag_name". I used the function : 
UA_Client_readValueAttribute(client, UA_NODEID_STRING(2,  $tag_name), val); 
But it returns nothing when i print the value.
An exemple of a tag_name : Rectification U - Fanuc.UGV8_7073.CN_JOG_MODE
Is there any mistake ?

jpfr, 2019-03-20:
Which status code is returned by UA_Client_readValueAttribute?

abdellah96, 2019-03-21:
Thank you for your reply. I managed to solve the issue. It was just a problem in the variable type UA_Float not UA_Int32. But still, how can i know the type of the values when receiving them from the server ?

jpfr, 2019-03-21:
Read returns an UA_Variant.
The variant contains information about the type.

closed 2019-03-21 by jpfr

abdellah96, 2019-03-21:
Yeah, thank you, i should read more the doc. Otherwise, what's the best way to retrieve values from all tags in a fast way. I've done it only for one tag using the tutorial example and i configured the publishing and sampling interval on 10 ms but the performances seems to be poor. I'm using the following code.
 ` UA_CreateSubscriptionRequest request = UA_CreateSubscriptionRequest_default();
    request.requestedPublishingInterval = 10.0;
    request.priority = 10;

    UA_CreateSubscriptionResponse response = UA_Client_Subscriptions_create(client, request,
                                                                            NULL, NULL, NULL);

    UA_UInt32 subId = response.subscriptionId;
    if(response.responseHeader.serviceResult == UA_STATUSCODE_GOOD)
        printf("Create subscription succeeded, id %u\n", subId);

    UA_MonitoredItemCreateRequest monRequest =
        UA_MonitoredItemCreateRequest_default(UA_NODEID_STRING(2, "Rectification U - Fanuc.UGV8_7073.CN_JOG_MODE"));
    monRequest.requestedParameters.samplingInterval = 10.0;
    monRequest.requestedParameters.discardOldest = true;
    monRequest.requestedParameters.queueSize = 100;

    UA_MonitoredItemCreateResult monResponse =
    UA_Client_MonitoredItems_createDataChange(client, response.subscriptionId,
                                              UA_TIMESTAMPSTORETURN_BOTH,
                                              monRequest, NULL, handler_TheAnswerChanged, NULL);
    if(monResponse.statusCode == UA_STATUSCODE_GOOD)
        printf("Monitoring 'Rectification U - Fanuc.UGV8_7073.CN_JOG_MODE', id %u\n", monResponse.monitoredItemId);

    /* The first publish request should return the initial value of the variable */
    UA_Client_run_iterate(client, 10000);`

Sorry, i'm new in opc-ua protocol.