open62541/open62541 #2594: Q: does anyone have examples implementing the UA_EUInformation and UA_EURange on server?

GeorgeSellner, 2019-03-21:
## Description

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mlang-de, 2019-03-21:
Maybe PR #2238 contains the solution. See file examples/tutorial_server_dataaccess.c.

HenningNT, 2020-03-25:
I can't find the file, where should I look?

GeorgeSellner, 2020-10-05:
@Pro @mlang-de 

So I am able to construct the arrays containing built-in data types, enum etc.
[image]
Unfortunately Euromap 82.2 creates structured types, then uses them in arrays. The current code generator produces the following error, which I suspect is part of the issues I'm experiencing.
`WARNING:nodes:Unknown Field Attribute ArrayDimensions`
In the end, the methods that I was using to write the array of enum does not actually change the Value property.
This is my result:
[image]
While this is what I am told it should look like:
[image]
link to Euromap 83
[https://www.euromap.org/en/euromap83/](https://www.euromap.org/en/euromap83/)
attached is latest 82.2 file
[Opc.Ua.PlasticsRubber.HotRunner.NodeSet2.xml.zip](https://github.com/open62541/open62541/files/5330008/Opc.Ua.PlasticsRubber.HotRunner.NodeSet2.xml.zip)
[image]
[image]

`[ 78%] Generating ../../src_generated/open62541/euromap83_nodeids.h
[ 78%] Built target open62541-generator-ids-euromap83
[ 78%] Generating ../../src_generated/open62541/types_euromap83_generated.c, ../../src_generated/open62541/types_euromap83_generated.h, ../../src_generated/open62541/types_euromap83_generated_handling.h, ../../src_generated/open62541/types_euromap83_generated_encoding_binary.h
[ 78%] Built target open62541-generator-types-euromap83
[ 79%] Generating ../../src_generated/open62541/namespace_euromap83_generated.c, ../../src_generated/open62541/namespace_euromap83_generated.h
INFO:__main__:Preprocessing (existing) /home/mm-prog/workspace/open62541/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing (existing) /home/mm-prog/workspace/open62541/deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml
INFO:__main__:Preprocessing /home/mm-prog/workspace/open62541/examples/nodeset/euromap/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
INFO:__main__:Generating Code for Backend: open62541
INFO:__main__:NodeSet generation code successfully printed
[ 79%] Built target open62541-generator-ns-euromap83
[ 79%] Generating ../../src_generated/open62541/euromap82_2_nodeids.h
[ 79%] Built target open62541-generator-ids-euromap82_2
[ 80%] Generating ../../src_generated/open62541/types_euromap82_2_generated.c, ../../src_generated/open62541/types_euromap82_2_generated.h, ../../src_generated/open62541/types_euromap82_2_generated_handling.h, ../../src_generated/open62541/types_euromap82_2_generated_encoding_binary.h
[ 80%] Built target open62541-generator-types-euromap82_2
[ 80%] Generating ../../src_generated/open62541/namespace_euromap82_2_generated.c, ../../src_generated/open62541/namespace_euromap82_2_generated.h
INFO:__main__:Preprocessing (existing) /home/mm-prog/workspace/open62541/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing (existing) /home/mm-prog/workspace/open62541/deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml
INFO:__main__:Preprocessing (existing) /home/mm-prog/workspace/open62541/examples/nodeset/euromap/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
INFO:__main__:Preprocessing /home/mm-prog/workspace/open62541/examples/nodeset/euromap/Opc.Ua.PlasticsRubber.HotRunner.NodeSet2.xml
WARNING:nodes:Unknown Field Attribute ArrayDimensions
ERROR:datatypes:ns=3;i=6215: Expected XML element with tag Zones but found ExtensionObject instead
INFO:__main__:Generating Code for Backend: open62541
INFO:__main__:NodeSet generation code successfully printed
[ 80%] Built target open62541-generator-ns-euromap82_2
Scanning dependencies of target server_nodeset_euromap
[ 80%] Building C object examples/nodeset/CMakeFiles/server_nodeset_euromap.dir/euromap/server_nodeset_euromap.c.o
[ 82%] Building C object examples/nodeset/CMakeFiles/server_nodeset_euromap.dir/__/__/src_generated/open62541/namespace_di_generated.c.o
[ 82%] Building C object examples/nodeset/CMakeFiles/server_nodeset_euromap.dir/__/__/src_generated/open62541/namespace_euromap83_generated.c.o
[ 82%] Building C object examples/nodeset/CMakeFiles/server_nodeset_euromap.dir/__/__/src_generated/open62541/namespace_euromap82_2_generated.c.o
[ 83%] Building C object examples/nodeset/CMakeFiles/server_nodeset_euromap.dir/__/__/src_generated/open62541/types_di_generated.c.o
[ 83%] Building C object examples/nodeset/CMakeFiles/server_nodeset_euromap.dir/__/__/src_generated/open62541/types_euromap83_generated.c.o
[ 83%] Building C object examples/nodeset/CMakeFiles/server_nodeset_euromap.dir/__/__/src_generated/open62541/types_euromap82_2_generated.c.o
[ 84%] Linking C executable ../../bin/examples/server_nodeset_euromap
[ 87%] Built target server_nodeset_euromap
`

GeorgeSellner, 2020-10-07:
Update-
I resolved my issue. Will update this thread and hopefully an example.

--Cheers

vvpnet, 2021-09-07:
> Update-
> I resolved my issue. Will update this thread and hopefully an example.
> 
> --Cheers

Hello. Show me an example. I'm not like it doesn't work out.

teccheck, 2025-03-18:
Seems like this is fixed. Since it's been four years, I'll close this issue

closed 2025-03-18 by teccheck