open62541/open62541 #2595: Add Support: OPC UA Pub/Sub over AMQP

ketiiot, 2019-03-22:
## Description
Currently, open62541 does not support OPC UA Pub/Sub over AMQP, which is referred to Part 14 of the OPC UA specification. 
I have a phased proposal to implement OPC UA Pub/Sub over AMQP.

- Phase 1: Based on the open62541 publish
                - AMQP implementation with RabbitMQ(Broker)
                - JSON implementation
- Phase 2: Development of OPC UA subscriber module with
                 - AMQP functionality implementation 
                - JSON implementation

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [ 0.3.0 version] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-03-30:
Dear @ketiiot ,

great to see this announcement!
It will be great to support AMQP in our PubSub implementation.

Looking forward to your contributions. Just one question:

You write that you want to use the RabbitMQ library.
Since open62541 is a C library, is this the dependency?

https://github.com/alanxz/rabbitmq-c

I am a bit worried because this RabbitMQ library does not mention AMQP 1.0. OPC UA PubSub **absolutely requires** AMQP 1.0. There is a huge incompatible difference to earlier versions.

The following library is also in C. They explicitly state compatibility with AMQP 1.0.
And the community is very active. There might be more alternative. I only searched for a few minutes.

https://github.com/apache/qpid-proton/tree/master/c

Can we decide on the library dependencies early on?
We should avoid duplicate work due to technical or license issues late in the development process.

Looking forward to hearing back from you!

ketiiot, 2019-04-03:
Dear @jpfr 

I will answer your questions step by step.

1. Do you choose the RabbitMQ library because open61541 is a C library?
: At the beginning stage, I chose the RabbitMQ library because the library is very common and widely used.

2. RabbitMQ library does not mention AMQP 1.0, but OPC UA PubSub absolutely requires AMQP 1.0
: You’re right. Currently, RabbitMQ provides the Plug-In for AMQP 1.0, but it does not provide
full-compatibility with AMQP 1.0.

3. The Qpid-Proton library is also in C. They explicitly state compatibility with AMQP 1.0. And the community is very active. There might be more alternative.
: Yes, Qpid-Proton officially announced the compatibility with AMQP 1.0. 
According to your suggestion, I will be able to implement AMQP/OPC UA based on the Qpid-Proton. 
(I will also search more suitable libraries for this issue) 

Honestly, I almost finalized the implementation of AMQP/OPC UA with Rabbit MQ broker. 
After I finish it ASAP, I will implement AMQP over OPC UA based on the Qpid-Proton. 

Although the library is changed, the transmission mechanism is almost the same. Thus, what I need to do is to modify the APIs for the new library.

Thank you for the good advice!

Pro, 2019-04-03:
> Honestly, I almost finalized the implementation of AMQP/OPC UA with Rabbit MQ broker.
> After I finish it ASAP, I will implement AMQP over OPC UA based on the Qpid-Proton.

If you would like to get some feedback on your implementation, feel free to create a pull request early on. We may be able to give you some hints and feedback. Early on code review may also reduce the work for the final PR.

sohailmushtaq, 2019-04-10:
Dear @ketiiot and @Pro,
any news about OPC UA/AMQP implementation?
Thank you!

ketiiot, 2019-04-13:
Dear @sohailmushtaq,

We have completed 70% development on RabbitMQ.
After development based on RabbitMQ, we will develop using Qpid-proton.

Thank you!

closed 2019-04-13 by ketiiot

RavilN, 2019-11-21:
Hi,
I don't see any resolution note on this issue, could you please clarify, was it really implemented (in this case there would be link to accepted pull request I guess), or decision was made to not implement it at all?
In the latest case I think would be better to keep the issue open.