open62541/open62541 #2596: Using OPC UA server on ARM-Cortex M3 or M4

samAvr, 2019-03-22:
hello,

i just start to learn about OPC UA and want to test a small server example on my blue pill STM32 board which run using ARM-Cortex M3 or on the STM32F429-Discovery. My problem is that i don't know how to start and where should i start.
there is a lot of examples here but i couldn't find something for the embedded systems. i just want to read some variables using the server like temprature that i measure using STM32 and change some GPIOs status (ON/OFF).
so can any one help me and let me know where i can start here?

thnx

samAvr, 2019-03-25:
any help?

jpfr, 2019-03-25:
Do you use freertos or another well known operating system?

samAvr, 2019-03-26:
@jpfr i use FreeRTOS

Pro, 2019-03-26:
Please have a look at this thread: #2511 

It includes instructions on how to build our stack for FreeRTOS. Also have a look into our architectures folder:
https://github.com/open62541/open62541/tree/master/arch

closed 2019-03-26 by Pro

samAvr, 2019-03-26:
@Pro thnx a lot