open62541/open62541 #2598: How to publish multiple values? (datetime AND temperature)

theresa95, 2019-03-22:
I am currently trying to publish a datetime value AND a temperature value with the OPC UA implementation called open62541.

For this, I have been using the test example from the open62541 github repository as it is already publishing the current datetime. I have tried to add another DSF(DataSetField) for the temperature value like it was shown with the datetime. But it won't publish it.

The following code snippet shows how I define methods for adding the DataSetFields:
`/**`
` * **DataSetField handling**`
 `* The DataSetField (DSF) is part of the PDS and describes exactly one published`
` * field. */`
`static void`
`addDataSetField(UA_Server *server) {`
 `   /* Add a field to the previous created PublishedDataSet */`
`    UA_NodeId dataSetFieldIdent;`
`    UA_DataSetFieldConfig dataSetFieldConfig;`
`    memset(&dataSetFieldConfig, 0, sizeof(UA_DataSetFieldConfig));`
`    dataSetFieldConfig.dataSetFieldType = UA_PUBSUB_DATASETFIELD_VARIABLE;`
`    dataSetFieldConfig.field.variable.fieldNameAlias = UA_STRING("Server localtime");`
`    dataSetFieldConfig.field.variable.promotedField = UA_FALSE;`

`    dataSetFieldConfig.field.variable.publishParameters.publishedVariable =`
`                  UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_CURRENTTIME);`
`    dataSetFieldConfig.field.variable.publishParameters.attributeId = `
`                  UA_ATTRIBUTEID_VALUE;`
`    UA_Server_addDataSetField(server, publishedDataSetIdent,`
`                          &dataSetFieldConfig, &dataSetFieldIdent);`
`}`

`static void`
`addDataSetFieldTemperature(UA_Server *server) {`
 `   /* Add a field to the previous created PublishedDataSet */`
  `  UA_NodeId dataSetFieldIdent;`
   ` UA_DataSetFieldConfig dataSetFieldConfig;`
    `memset(&dataSetFieldConfig, 0, sizeof(UA_DataSetFieldConfig));`
    `dataSetFieldConfig.dataSetFieldType = UA_PUBSUB_DATASETFIELD_VARIABLE;`
    `dataSetFieldConfig.field.variable.fieldNameAlias = UA_STRING("Temperature");`
    `dataSetFieldConfig.field.variable.promotedField = UA_FALSE;`
    `dataSetFieldConfig.field.variable.publishParameters.publishedVariable =`
     `                                 UA_NODEID_NUMERIC(0, 50);`
    `dataSetFieldConfig.field.variable.publishParameters.attributeId =`
                                     ` UA_ATTRIBUTEID_VALUE;`
    `UA_Server_addDataSetField(server, publishedDataSetIdent, `
                                      `&dataSetFieldConfig,`
                                      `&dataSetFieldIdent);`
`}`

In the main function, I then call both methods to add them to the PDS (PublishedDataSet):
`addPubSubConnection(server, transportProfile, networkAddressUrl);`
`addPublishedDataSet(server);`
`addDataSetFieldTemperature(server);`
`addDataSetField(server);`
`addWriterGroup(server);`
`addDataSetWriter(server);`

The temperature value is a simple integer value, so what am I doing wrong here? Maybe the error is within the subscribed client application?

For subscribing to the values published, I have also used the example on github (the file is called tutorial_pubsub_subscribe.c) and I am trying to parse the values as followed:

`/* Loop over the fields and print well-known content types */`
`for(int i = 0; i < dsm->data.keyFrameData.fieldCount; i++) {`
`    const UA_DataType *currentType = dsm->data.keyFrameData.dataSetFields[i].value.type;`
`    if(currentType == &UA_TYPES[UA_TYPES_BYTE]) {`
`        UA_Byte value = *(UA_Byte *)dsm->data.keyFrameData.dataSetFields[i].value.data;`
`        UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,`
`                    "Message content: [Byte] \tReceived data: %i", value);`
`    } else if (currentType == &UA_TYPES[UA_TYPES_DATETIME]) {`
 `       UA_DateTime value = *(UA_DateTime *)dsm-``>data.keyFrameData.dataSetFields[i].value.data;`
`        UA_DateTimeStruct receivedTime = UA_DateTime_toStruct(value);`
`        UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,`
`                    "Message content: [DateTime] \t"`
`                    "Received date: %02i-%02i-%02i Received time: %02i:%02i:%02i",`
`                    receivedTime.year, receivedTime.month, receivedTime.day,`
`                    receivedTime.hour, receivedTime.min, receivedTime.sec);`
`    } else if(currentType == &UA_TYPES[UA_TYPES_UINT16]) {`
`        UA_UInt16 receivedTemp =`
`            *(UA_UInt16 *)dsm->data.keyFrameData.dataSetFields[i].value.data;`

`        UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,`
`                    "Message content: [UA_UInt16] \t"`
`                    "Received temperature: %i",`
`                    receivedTemp);`
`    }`

johndzilva, 2019-04-17:
@theresa95 I have couple of questions which could help you to figure out the issue.
1. Are you sure that you have added your temperature variable in the server's address space?
2. Are you sure that it is added in ns=0 i=50? Because it is not. You should not add your node variable under ns=0

theresa95, 2019-04-25:
@johndzilva hey john! First of all, thank you for your reply! As I am very new to open62541 and OPC UA in general, I am a little bit confused with the nodeId issue. I thought with UA_NODEID_NUMERIC(0,50) I am setting the temperature value to 50. 

I would be so grateful if you can explain to me what the ns=0 an i = 50 exactly means? Why should I not add the nodevariable under ns = 0? 
If my temperature variable isn't added to the server's address space, is the timestamp variable not added to the address space either?

I appreciate every help you can possibly give me. Thank you very much in advance!

theresa95, 2019-04-26:
@johndzilva I have created a new issue on this topic. If you want, please have a look at issue #2673 . I would be very grateful if you could help me!

closed 2019-05-12 by theresa95

SKShah36, 2019-05-15:
@theresa95 Will it be possible for you to share the complete repository? I am a novice at this as well. From the discussion above I have been able to figure out publishing multiple values but I still don't know how to parse them. If you've been able to figure out, can you please share the link to your repo. Your help is much appreciated.