open62541/open62541 #2599: server.c is consuming 100% when using UA_Server_run_iterate(server, false) for faster publish to happen at regular intervals

bvinodkumar2008, 2019-03-22:
## Description
It's been found that for faster publish (publishing at regular intervals) for a datachange notification, we had to switch to [UA_Server_run_iterate(server, false)](https://gist.github.com/bvinodkumar2008/49151adddcd0636a062f5c53f78a1b66#file-server-c-L106) and [UA_sleep_ms(5)](https://gist.github.com/bvinodkumar2008/49151adddcd0636a062f5c53f78a1b66#file-server-c-L142) in server.c file.

This has resulted in 100% cpu utilization of 1 core, is there a way to circumvent this without losing the fast publish happening at fixed regular intervals. With UA_Server_run_iterate(server, true), the publish is found to happen at not so regular intervals.

## Background Information / Reproduction Steps
**Run Server**
```
$gcc -std=c99 -I../../include/ -g server.c ../open62541.c -L. -lmbedtls -lmbedx509 -lmbedcrypto -pthread -o server
$./server /etc/ssl/opcua/opcua_server_certificate.der /etc/ssl/opcua/opcua_server_key.der /etc/ssl/ca/ca_certificate.der &> server.txt &
```
**Run Client**
```
$gcc -std=c99 -I../../include/ -g client.c ../open62541.c -L. -lmbedtls -lmbedx509 -lmbedcrypto -pthread -o client
$./client /etc/ssl/opcua/opcua_client_certificate.der /etc/ssl/opcua/opcua_client_key.der /etc/ssl/ca/ca_certificate.der &> client.txt &
```

Note: I'm not able to attach the .der certificates here. The opcua_server_certificate.der and opcua_client_certificate.der are signed by ca_certificate.der. The certs are available at https://github.com/bvinodkumar2008/opcua_certs

Created a gist for server.c, client.c and their respective logs at https://gist.github.com/bvinodkumar2008/49151adddcd0636a062f5c53f78a1b66

## Checklist

Please provide the following information:

 - [https://github.com/open62541/open62541/commit/46979d3b87ecc8c13cac66c0019fd31024b55b6a] open62541 Version (release number or git tag):
 - [No ] Other OPC UA SDKs used (client or server): 
 - [Ubuntu 16.04] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-03-22:
UA_Server_run_iterate will return the number of milliseconds left until the next scheduled callback.
You can use this information to return to the next UA_Server_run_iterate exactly when the publish is due.

bvinodkumar2008, 2019-03-25:
Thanks @jpfr , will check and get back to you. By the way, is there any reference on how this is done anywhere in the repo here like in unit tests etc?

jpfr, 2019-03-25:
https://github.com/open62541/open62541/blob/d070a2ff51a62a2413e3676cc1069827d8bd5464/src/server/ua_server.c#L507
https://github.com/open62541/open62541/blob/d070a2ff51a62a2413e3676cc1069827d8bd5464/tests/server/check_server.c#L55

bvinodkumar2008, 2019-04-03:
Thanks @jpfr , it worked. I followed the logic mentioned at https://github.com/open62541/open62541/blob/master/examples/server_mainloop.c#L45

closed 2019-04-03 by bvinodkumar2008