open62541/open62541 #2600: Reading from opc-server and writing in database

abdellah96, 2019-03-22:
Hello,
I managed write a client that read the values of all the the industrial tags from server using subscription and monitoring items.  I want to write in parallel those values in a no-sql timeseries database but i can't since the program is interrupted when calling the function UA_Client_run_iterate(client, 100000).  
Please if someone knows how to do read/write in parrallel, i would appreciate it.

ccvca, 2019-03-22:
Hi,
There are two possible solutions, You could either use a second thread that writes the data in the database, or you could decrease the timeout of UA_Client_run_iterate, e.g. UA_Client_run_iterate(client, 0) and do everything in one thread.

abdellah96, 2019-03-25:
Thank you so much for your response. The function UA_Client_run_iterate(client, 0) runs for few seconds and stops. I manages to solve this problem by passing this function in an infinite loop and running a second thread for writing. So its a combination of the two solutions.

`pthread_create(&t2, NULL, write_data, (void *)&counter);
while(true){
      UA_Client_run_iterate(client, 0);
    }

    pthread_join(t2, NULL);`

closed 2019-03-25 by abdellah96