open62541/open62541 #2601: Add rust implementation to the list of implementations

shrmrf, 2019-03-22:
## Description
[locka99/opcua](https://github.com/locka99/opcua) is a Rust OPC UA implementation. It might be a good addition to the list here: https://github.com/open62541/open62541/wiki/List-of-Open-Source-OPC-UA-Implementations

## Background Information / Reproduction Steps
n/a
## Checklist
n/a
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-03-23:
Cool!
Can you add it to the wiki?

shrmrf, 2019-03-23:
@jpfr I just added the implementation to the list. 

Thanks!!

closed 2019-03-23 by shrmrf