open62541/open62541 #2608: Question about calling Method defined in ObjectType which writes to Object's Propertyerty

rudoq32, 2019-03-27:
## Description
I created information model as

ns[0] = UA_Server_addNamespace(server, "https://opcfoundation.org/UA/");
ns[1] = UA_Server_addNamespace(server, "https://opcfoundation.org/UA/DI/");
ns[2] = UA_Server_addNamespace(server, "http://yourorganisation.org/Opc.NucleoF767ZI.NodeSet/");

In Opc.NucleoF767ZI.NodeSet I defined NUCLEOF767ZIDeviceType with:
Object|MethodSet|BaseObjectType|Mandatory
Object|ParametersSet|BaseObjectType|Mandatory

MethodSet contains ToggleLD2 as Method with Mandatory as ModellingRule.

ParameterSet contains LD2 as Variable PropertyType Mandatory.

I created an instance NUCLEOF767ZI_01. To generate source file I use open62541 nodeset_compiler. It creates two separate nodes for NUCLEOF767ZIDeviceType ToggleLD2 and NUCLEOF767ZI_01 ToggleLD2. So I need to add callbacks for both.

### Question
It is possible to have only one node ToggleLD2 method (defined in NUCLEOF767ZIDeviceType) and this method will toggle LD2 for NUCLEOF767ZI_01 (instance of NUCLEOF767ZIDeviceType)? Basically if I have ten of NUCLEOF767ZI_0i, that all ToggleLD2 are defined at  NUCLEOF767ZIDeviceType? 

I am quite sure that I am having gaps in concept of _Modeling_ and _References_. If someone could get me a hint of this, would be also helpful.
## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v0.3.0
 - [x] nodeset_compiler, documentation for Method call
 - [x] Operating system: Windows Server 2016
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

rudoq32, 2019-03-28:
>### OPC UA Part 3
>#### 6.4.4.5.2 Mandatory
>An InstanceDeclaration marked with the ModellingRule Mandatory fulfils exactly the semantic
>defined for the NamingRule Mandatory. That means that for each existing BrowsePath on the
>instance a similar Node shall exist, **but it is not defined whether a new Node is created or an
>existing Node is referenced.**

How can I defined that the BrowsePath is referring to the existing node?

jpfr, 2019-05-14:
The same node can appear in several places in the hierarchy of nodes.
(Only cyclic reference chains are prevented).

If you instantiate an object type, the mandatory children are automatically created.

If you want fine-grained control, you can use the _begin and _finish versions of AddNodes.
That way you can add your own children (add references to existing nodes) before calling _finish.
So the manually added children are taken into account.

See here: https://open62541.org/doc/current/server.html#node-addition-and-deletion

teccheck, 2025-03-18:
@jpfr's answer looks good to me and there were no further comments, so I'll close this issue

closed 2025-03-18 by teccheck