open62541/open62541 #2609: Running PubSub samples on different computers does not work?

theresa95, 2019-03-28:
I am currently experimenting with the publish and subscribe example applications. 
If I execute both samples on my windows desktop, it works like a charm: the publish-application publishes timestamps and the subscribe-application prints out the current timestamps. Also, I can see the transmitting udp packages in wireshark.

Now, I want to run the publish application on the raspberry pi and the subscribe application on my windows desktop (as before). But know, I do not see any udp packages in wireshark (windows) and there are no timestamps printed by the subscribe example also! 

I tried the same by running the publish example in a virtual box (with linux as operating system) and the subscribe example on windows desktop and I have the same problem as well: I can NOT see UDP packages transmitted in wireshark on the windows desktop, BUT I can see them if I start Wireshark in my Linux virtual box. So they are published, but it seems like the packages do not make it to the windows desktop. So what do I have to do to publish the UDP packages to my windows desktop, and not only on my raspberry pi/linux virtual box??

jpfr, 2019-03-30:
Do not open duplicate issues.
You can add more comments below to an issue.

jpfr, 2019-03-30:
Please try to set the network interface name together with the IP address.
Closing in favor of #2612.

closed 2019-03-30 by jpfr