open62541/open62541 #2612: Wireshark receives UDP multicast packages, but subscribe sample does not?

theresa95, 2019-03-29:
## Description
I am currently trying to publish timestamps on the raspberry pi and subscribe to them on my windows based PC. For this, I connected BOTH the Raspberry Pi and the Windows PC to a Switch. So, if I start the publish sample on the Raspberry Pi, I can already see the incoming UDP multicast packages in Wireshark on my Windows PC. So theoretically, the subscribe application also should be able to receive the packages and print them out to the console.
Why is that? If I run both the publish and subscribe sample on the windows PC, it works... 

I may have one clue what the problem could be: If I run both publish/subscribe samples on my windows pc, they both print out in the console, that they are listening opc.tcp://mypc:4801/
But when I run the publish sample on the raspi it says, it is listening on opc.tcp://raspberrypi:4840/ but the subscribe sample is still listening on opc.tcp://mypc:4801/ .

Any suggestions? Any advice is highly appreciated!!

## Additional Informations
In the following image, you can see the IP address of the raspberry pi: 
[image]

Below you can see the transmitted UDP multicast packages in Wireshark. They do indeed came from the raspberry pi because the IP addresses match:
[image]

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.3-rc4
 - [ ] Other OPC UA SDKs used (client or server):  No other OPC UA SDKs are used!
 - [x] Operating system: Windows 10 Pro (for subscribe-application) and linux distribution (raspberry pi for publish-application)
 - [x] Wireshark network dump attached
 - [ ] Self-contained code example attached: The code is the same as in the tutorials, nothing has been changed!
 - [x] Critical issue: for me it is critical :)

jpfr, 2019-03-30:
Are you sure this is on version 0.3-rc4?

There must be a line similar to this one:
https://github.com/open62541/open62541/blob/4209f74e2176f75f67b4f80962909c69cdb54126/examples/pubsub/pubsub_subscribe_standalone.c#L118

Which IP adress does the subscriber use?
A multicast IP or the local IP address?
Did you specify the network interface in the UA_NetworkAddressUrlDataType?
That can be important if you have several network interfaces that multicast capable.
For example one LAN and one WIFI.

theresa95, 2019-04-01:
> Are you sure this is on version 0.3-rc4?
> 
> There must be a line similar to this one:
> [open62541/examples/pubsub/pubsub_subscribe_standalone.c](https://github.com/open62541/open62541/blob/4209f74e2176f75f67b4f80962909c69cdb54126/examples/pubsub/pubsub_subscribe_standalone.c#L118)
> 
> Line 118 in [4209f74](/open62541/open62541/commit/4209f74e2176f75f67b4f80962909c69cdb54126)
> 
>  UA_NetworkAddressUrlDataType networkAddressUrl = 
> Which IP adress does the subscriber use?
> A multicast IP or the local IP address?
> Did you specify the network interface in the UA_NetworkAddressUrlDataType?
> That can be important if you have several network interfaces that multicast capable.
> For example one LAN and one WIFI.

hello @jpfr , 
first of all, thank you for your quick answer! 
I have cloned the masterbranch on the 20th of March, so I guess it is version 0.3. Where can I look it up which rc I have cloned?

In the tutorial_pubsub_subscribe.c file, there is indeed such a line of code (in the main function) and it looks like this:
`UA_NetworkAddressUrlDataType networkAddressUrl = {UA_STRING_NULL, UA_STRING("opc.udp://224.0.0.22:4840/")};`

As I have mentioned in the question above, I have not changed the pubsub-tutorial-samples at all. So 
the subscriber uses the multicast address "224.0.0.22:4840".
How can I specify the network interface in the UA_NetworkAddressUrlDataType?? I would very appreciate it, if you can give me an example on how to specify the network interface. 

Thank you so much in advance!

theresa95, 2019-04-02:
hey again @jpfr 
I tried out your suggestion and specified a network interface.
In order to find the correct network interface name, I typed in "ipconfig /all" and looked it up. 
I am pretty sure it is the "Ethernet-Adapter Ethernet 3". And I can kind of confirm this by looking it up on wireshark (as the "Ethernet 3" interface is the only one on which I receive the UDP multicast packages):

Therefore, I changed your suggested line to:
`UA_NetworkAddressUrlDataType networkAddressUrl = {UA_STRING("Ethernet 3"), UA_STRING("opc.udp://224.0.0.22:4840")};`
I tried it both with the pubsub_subscribe_standalone and with the tutorial_pubsub_subscribe.c.
But in both examples, I get the following output:

Pro, 2019-04-03:
The error "Interface selection failed" already gives you a starting hint. You should first try to fix this, before continuing any further tests.

It is printed here:
https://github.com/open62541/open62541/blob/master/plugins/ua_pubsub_udp.c#L249

If you check, where `networkInterface` is used yields the call to `UA_inet_pton`.
The documentation to `int inet_pton(int af, const char *src, void *dst);` says:

> src points to a character string containing an IPv4 network address in dotted-decimal format, "ddd.ddd.ddd.ddd" 

So you should not use the network interface name, but the ip address of the network interface.

theresa95, 2019-04-03:
> The error "Interface selection failed" already gives you a starting hint. You should first try to fix this, before continuing any further tests.
>
> It is printed here: [...]

Hello @Pro , first of all, thank you so much for your answer!
I have tried tried it with the IP address of my network interface and the warning is gone. But the subscriber sample still does not print any timestamps to the console. 
What could be the problem here? I see the UDP packages coming in on Wireshark.

Pro, 2019-04-03:
The 169.254 ip address does look a bit fishy. IP addresses in that range normally mean that there's an issue with your DHCP server.
Try to set static IP addresses on the RasPI and on the windows PC and make sure that the multicast IP settings are correct. i.e. check the routing table of your OS.

theresa95, 2019-04-03:
> The 169.254 ip address does look a bit fishy. IP addresses in that range normally mean that there's an issue with your DHCP server.
> Try to set static IP addresses on the RasPI and on the windows PC and make sure that the multicast IP settings are correct. i.e. check the routing table of your OS.

Should I set IP addresses which begin with 10.0. xx.xx??

theresa95, 2019-04-03:
> The 169.254 ip address does look a bit fishy. IP addresses in that range normally mean that there's an issue with your DHCP server.
> Try to set static IP addresses on the RasPI and on the windows PC and make sure that the multicast IP settings are correct. i.e. check the routing table of your OS.

Hey @Pro  , I have tried to connect the raspberry pi and the windows pc with a router via LAN and specified the IP address ( which is 192.168.0.105 for Ethernet 3 now) for the network interface.  I have done the same within the publish sample on the raspberry pi (which is 192.168.0.106 for eth0). There are no warnings, but I still cannot receive any UDP packages in my subscribe sample (which runs on windows). I still see the UDP packages only on wireshark (which runs on the windows pc). 

Do you have a clue what could be the problem here?

Pro, 2019-04-03:
Can you try to verify if your network setup works?
e.g. use another program to test if multicast messages are successfully received:

First test with default settings, and then with the same multicast group as OPC Pub/Sub is using.

https://github.com/yantisj/multicast-test

https://pymotw.com/2/socket/multicast.html

If that isnt working, then you first need to make sure your devices can successfully receive multicast traffic. E.g. disalbe firewalls and stuff. We can not help you with that.

jpfr, 2019-04-06:
Try to disable all network interfaces but the correct one.
For multicast, Windows likes to take the first interface that says “I’m capable”. And then picks the wrong one...

closed 2019-05-29 by theresa95

audupi, 2021-04-08:
> Hey @Pro , I have tried to connect the raspberry pi and the windows pc with a router via LAN and specified the IP address ( which is 192.168.0.105 for Ethernet 3 now) for the network interface. I [...]

> Hey @Pro , I have tried to connect the raspberry pi and the windows pc with a router via LAN and specified the IP address ( which is 192.168.0.105 for Ethernet 3 now) for the network interface. I [...]

Can you please explain how did you solve this?