open62541/open62541 #2615: When a node is deleted, its children are not deleted.

UsachTim, 2019-04-01:
## Description
Attempting to recreate a tree of nodes with the same numeric identifiers after deletion results in an error:

`AddNodes: Node could not add the new node to the nodestore with error code BadNodeIdExists`

The process creates the parent node correctly. Child nodes are not re-created.

## Background Information / Reproduction Steps
To delete the function is called:

`UA_Server_deleteNode(UA_Server *server, const UA_NodeId nodeId,  UA_Boolean DeleteReferences);`

The debugging showed that the list of remote nodes inside the function **static void removeChildren (....)** returned by the function **void Operation_Browse (...)** has size zero.

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):0.3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Linux OpenSuse 15.0
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-04-05:
This should have a unit test.
On master and on 0.3.

MartinaFelberFEM, 2019-11-13:
Hello!

I think I stumbled upon the same issue. 
I need to add and delete nodes during runtime. 

To delete the nodes I use `UA_Server_deleteNode` function.

I think the function comments (documentation in header file server.h) are a bit misleading:

```c
/* Deletes a node and optionally all references leading to the node. */
UA_StatusCode UA_EXPORT UA_THREADSAFE
UA_Server_deleteNode(UA_Server *server, const UA_NodeId nodeId,
                     UA_Boolean deleteReferences); 
```

Because the function does not only delete the specified node, but recursively deletes all child nodes too, but only if the reference type of the child nodes is of subtype **UA_NS0ID_AGGREGATES**. 

UA_Server_deleteNode -> deleteNodeOperation -> recursiveDeleteNode
https://github.com/open62541/open62541/blob/280012ee016e2f42ab9b1386174a8c12e6b29821/src/server/ua_services_nodemanagement.c#L1543-L1552

If I create a folder node with reference type ORGANIZES then the child nodes are not deleted. 
Because the browse result does not have any nodes. 
Because Organizes reference isn't a subtype of Aggregates.
[image]

If I create a folder node (or any other object node) with reference type HAS_COMPONENT then all children are automatically deleted. 

Every BrowseRequest within ua_services_nodemanagement.c uses referenceTypeId UA_NS0ID_AGGREGATES. 

- findChildByBrowsename
- copyAllChildren
- recursiveCallConstructors
- recursiveDeconstructNode

I am wondering if there's a specific reason for this. 
Or if it would be better/safer to use UA_NS0ID_NONHIERARCHICALREFERENCES ?

Many thanks for your support!

Best regards
Martina

o6-nbeier, 2026-10-20:
Thank you for taking the time to open this issue and describe it so clearly.

The problem has since been fixed on the maintained branches (see 875d33a94c), so we are closing it as resolved.

closed 2026-10-20 by o6-nbeier