open62541/open62541 #2617: Q : Is there a way to close a session from the server programmatically?

juangburgos, 2019-04-02:
## Description

I am experimenting with the Access Control API on the server. I have implemented my custom versions of `activateSession` and `closeSession` where I receive the `const UA_NodeId * sessionId` argument. 

I was wondering; given the session's **nodeId**, is it possible to force an specific session shutdown from the server? The idea would be to kick a single user out whle preserving all other running sessions.

Thank you,

Pro, 2019-04-03:
I guess the simplest approach would be to just remove the session from the session manager.

I.e. use the method:
```c
UA_StatusCode
UA_SessionManager_removeSession(UA_SessionManager *sm, const UA_NodeId *token) {
```
And then also remove the session from the secure channel by forcefully closing it:
```
void
UA_SecureChannel_close(UA_SecureChannel *channel) {
```

This should force the client to reconnect and re-authenticate.
If you only need to invalidate the session, but keep the secure channel, we would need to add an additional method which allows to remove the session from a channel.

Can you try that out and give us feedback if that works?

juangburgos, 2019-04-03:
Hi, I don't know the souce code of the library, but by supeficially going over it I came up with this:

```c
// header
UA_StatusCode UA_EXPORT
UA_Server_closeSession(UA_Server *server, const UA_NodeId *sessionId);

// source
UA_StatusCode
UA_Server_closeSession(UA_Server *server, const UA_NodeId *sessionId)
{
	UA_Session *session = UA_SessionManager_getSessionById(&server->sessionManager, sessionId);
	UA_SecureChannel *channel = session->header.channel;

	UA_SecureChannel_close(channel);
	UA_StatusCode retval = UA_SessionManager_removeSession(&server->sessionManager, &session->header.authenticationToken);

	return retval;
}
```

And it seems to work in the sense that it does kick out the user, but:

1. I don't know if by forcing the session close with the code above if I am creating some memory leaks or any other secondary effects on the server.

2. After being kicked out, the UA Expert client seems ot be stuck in a Connect Error where it seems impossible to escape, whether if you Ignore it or Abort it.

[image]

```
[2019-04-03 13:08:15.389 (UTC+0200)] warn/securitypolicy        No PKI plugin set. Accepting all certificates
[2019-04-03 13:08:15.390 (UTC+0200)] info/channel       Connection 816 | SecureChannel 11 | Opened SecureChannel
[2019-04-03 13:08:16.661 (UTC+0200)] info/session       Connection 816 | SecureChannel 11 | Session ns=1;g=acfb9eb1-f955-8fd0-ea9e-a90a52e53e83 | ActivateSession: The AccessControl plugin denied the access with the status code BadUserAccessDenied
[2019-04-03 13:08:16.696 (UTC+0200)] warn/session       Connection 816 | SecureChannel 11 | Session ns=1;g=acfb9eb1-f955-8fd0-ea9e-a90a52e53e83 | Calling service 473 on a non-activated session
[2019-04-03 13:08:16.697 (UTC+0200)] info/channel       Connection 816 | SecureChannel 11 | CloseSecureChannel
[2019-04-03 13:08:16.698 (UTC+0200)] info/network       Connection 816 | Closed
[2019-04-03 13:08:21.821 (UTC+0200)] info/network       Connection 820 | New connection over TCP from 127.0.0.1
[2019-04-03 13:08:21.822 (UTC+0200)] info/channel       Creating a new SecureChannel
[2019-04-03 13:08:21.823 (UTC+0200)] warn/securitypolicy        No PKI plugin set. Accepting all certificates
[2019-04-03 13:08:21.824 (UTC+0200)] info/channel       Connection 820 | SecureChannel 12 | Opened SecureChannel
[2019-04-03 13:08:22.796 (UTC+0200)] warn/session       Connection 820 | SecureChannel 12 | Session ns=1;g=bfd4df26-ed26-5ac4-2162-26326e1691d9 | Calling service 473 on a non-activated session
[2019-04-03 13:08:22.833 (UTC+0200)] info/session       Try to use Session with token ns=1;g=55f900a0-657a-6098-4bc4-2c32b56b6646 but is not found
[2019-04-03 13:08:22.835 (UTC+0200)] warn/channel       Connection 820 | SecureChannel 12 | Service request 473 without a valid session
[2019-04-03 13:08:22.836 (UTC+0200)] info/channel       Connection 820 | SecureChannel 12 | CloseSecureChannel
[2019-04-03 13:08:22.837 (UTC+0200)] info/network       Connection 820 | Closed
[2019-04-03 13:08:27.963 (UTC+0200)] info/network       Connection 728 | New connection over TCP from 127.0.0.1
[2019-04-03 13:08:27.964 (UTC+0200)] info/channel       Creating a new SecureChannel
[2019-04-03 13:08:27.965 (UTC+0200)] warn/securitypolicy        No PKI plugin set. Accepting all certificates
[2019-04-03 13:08:27.965 (UTC+0200)] info/channel       Connection 728 | SecureChannel 13 | Opened SecureChannel
[2019-04-03 13:08:29.468 (UTC+0200)] info/session       Connection 728 | SecureChannel 13 | Session ns=1;g=0007ccf1-566e-22b7-75a7-1ae15a3513c1 | ActivateSession: The AccessControl plugin denied the access with the status code BadUserAccessDenied
[2019-04-03 13:08:29.508 (UTC+0200)] warn/session       Connection 728 | SecureChannel 13 | Session ns=1;g=0007ccf1-566e-22b7-75a7-1ae15a3513c1 | Calling service 473 on a non-activated session
[2019-04-03 13:08:29.509 (UTC+0200)] info/channel       Connection 728 | SecureChannel 13 | CloseSecureChannel
[2019-04-03 13:08:29.510 (UTC+0200)] info/network       Connection 728 | Closed
[2019-04-03 13:08:34.639 (UTC+0200)] info/network       Connection 728 | New connection over TCP from 127.0.0.1
[2019-04-03 13:08:34.649 (UTC+0200)] info/channel       Creating a new SecureChannel
[2019-04-03 13:08:34.650 (UTC+0200)] warn/securitypolicy        No PKI plugin set. Accepting all certificates
[2019-04-03 13:08:34.651 (UTC+0200)] info/channel       Connection 728 | SecureChannel 14 | Opened SecureChannel
[2019-04-03 13:08:36.407 (UTC+0200)] info/session       Connection 728 | SecureChannel 14 | Session ns=1;g=39d802bd-81c6-17cc-69b6-3b2378d79dc9 | ActivateSession: The AccessControl plugin denied the access with the status code BadUserAccessDenied
[2019-04-03 13:08:36.453 (UTC+0200)] warn/session       Connection 728 | SecureChannel 14 | Session ns=1;g=39d802bd-81c6-17cc-69b6-3b2378d79dc9 | Calling service 473 on a non-activated session
[2019-04-03 13:08:36.454 (UTC+0200)] info/channel       Connection 728 | SecureChannel 14 | CloseSecureChannel
[2019-04-03 13:08:36.455 (UTC+0200)] info/network       Connection 728 | Closed
[2019-04-03 13:08:41.581 (UTC+0200)] info/network       Connection 616 | New connection over TCP from 127.0.0.1
[2019-04-03 13:08:41.592 (UTC+0200)] info/channel       Creating a new SecureChannel
[2019-04-03 13:08:41.593 (UTC+0200)] warn/securitypolicy        No PKI plugin set. Accepting all certificates
[2019-04-03 13:08:41.594 (UTC+0200)] info/channel       Connection 616 | SecureChannel 15 | Opened SecureChannel
[... cut ...]
[2019-04-03 13:12:17.360 (UTC+0200)] info/network       Connection 728 | Closed
[2019-04-03 13:12:22.485 (UTC+0200)] info/network       Connection 816 | New connection over TCP from 127.0.0.1
[2019-04-03 13:12:22.486 (UTC+0200)] info/channel       Creating a new SecureChannel
[2019-04-03 13:12:22.487 (UTC+0200)] warn/securitypolicy        No PKI plugin set. Accepting all certificates
[2019-04-03 13:12:22.488 (UTC+0200)] info/channel       Connection 816 | SecureChannel 17 | Opened SecureChannel
```

juangburgos, 2019-04-10:
Will close, since the code works. 
Maybe add the proposed method to the source code?
Hope to get feedback on possible secondary effects some day.

closed 2019-04-10 by juangburgos