open62541/open62541 #2622: Amalgamated open62541.h/.c doesn't work on master when running encryption examples

bvinodkumar2008, 2019-04-03:
## Description
Looks like the amalgamated open62541.c/.h generated on latest master fails to compile https://github.com/open62541/open62541/blob/master/examples/encryption/client_encryption.c as it is not able to find few definitions like below:

```
error: too many arguments to function ‘UA_Client_new’
     client = UA_Client_new(UA_ClientConfig_default);
```
Please provide the following information:

 - [217eed4] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

bvinodkumar2008, 2019-04-03:
Hi @jpfr , Is this expected to fail? Is there a workaround for this?

jpfr, 2019-04-04:
Client_new does not take arguments now.
https://github.com/open62541/open62541/blob/4630139dd94f721aaa9345b9cb1240336a73bf8f/include/open62541/client.h#L52

You are using an old version of the examples with a current master.

closed 2019-04-04 by jpfr

bvinodkumar2008, 2019-04-05:
Thanks @jpfr for quick response, let me check and get back to you