open62541/open62541 #2624: Can't build with UA_ENABLE_ENCRYPTION (FreeRTOS LWIP)

embbo, 2019-04-04:
Hi I am trying to build for freetos lwip but i getting this error.

/open62541.c:48293: undefined reference to `mbedtls_x509_crt_init'
 /open62541.c:48294: undefined reference to `mbedtls_x509_crt_parse'
 /open62541.c:48309: undefined reference to `mbedtls_x509_crt_free'
 /open62541.c:48192: undefined reference to `mbedtls_x509_crt_init'
 /open62541.c:48193: undefined reference to `mbedtls_x509_crt_parse'
 /open62541.c:48210: undefined reference to `mbedtls_x509_crt_verify_with_profile'
 /open62541.c:48237: undefined reference to `mbedtls_x509_crt_free'
 /open62541.c:48318: undefined reference to `mbedtls_x509_crt_free'
 /open62541.c:48319: undefined reference to `mbedtls_x509_crl_free'
 /open62541.c:48333: undefined reference to `mbedtls_x509_crt_init'
 /open62541.c:48334: undefined reference to `mbedtls_x509_crl_init'
 /open62541.c:48346: undefined reference to `mbedtls_x509_crt_parse'
 /open62541.c:48351: undefined reference to `mbedtls_x509_crl_parse'
 /open62541.c:49136: undefined reference to `UA_SecurityPolicy_Basic128Rsa15'

juangburgos, 2019-04-04:
Did you link to the [mbedtls](https://github.com/ARMmbed/mbedtls) library? 

Clone the repo, checkout some stable version (e.g. `mbedtls-2.17.0`) then compile the library

```
cd $REPOS_PATH/mbedtls.git
mkdir build; cd build
cmake
```

Then finally recompile again open62541:

```
cd $REPOS_PATH/open62541.git
mkdir build; cd build
cmake -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_ENCRYPTION=ON .. -DMBEDTLS_INCLUDE_DIRS=$REPOS_PATH/mbedtls.git/build/include -DMBEDTLS_LIBRARY=$REPOS_PATH/mbedtls.git/build/library/Debug -DMBEDX509_LIBRARY=$REPOS_PATH/mbedtls.git/build/library/Debug -DMBEDCRYPTO_LIBRARY=$REPOS_PATH/mbedtls.git/build/library/Debug
```

See if that works

embbo, 2019-04-04:
Thank you okay, I inculde manually ..and include in path., but let me try your way and will come back with results

embbo, 2019-04-04:
@juangburgos  till cmake , it works fine and when i try to build in eclipse it gives again same error

juangburgos, 2019-04-04:
Which cmake, first (mbedtls), second (open62541)? You need to adapt your Cmake command to use your genertor. You need to provide more information if you want to get better help. 

Paste your commands with their outputs, use markdown format to make it all readible.

You issue definetively points to how you include and link, I dont think there is something wrong with either library.

jpfr, 2019-04-06:
Seems to be eclipse specific.
Please reopen if we can make improvements on our end.

closed 2019-04-06 by jpfr

embbo, 2019-04-08:
@jpfr , It seems about #defines in msbedtls config file. but i am still working

embbo, 2019-04-18:
@juangburgos  
mbedtls_platform_entropy_poll' undeclared (first use in this function)  
what can be problem .

juangburgos, 2019-04-18:
Undeclared functions means you are not including the mbedtls headers correctly. Maybe take a look at [this link](https://github.com/juangburgos/QUaServer#encryption), it might point you on a better direction.