open62541/open62541 #2626: Browsing variable node causes segfault

tobiaslocker, 2019-04-04:
## Description
I am running the provided `server.cpp` example and i get a segfault in my code here:
```
Identifier Identifier::from_node_id(UA_NodeId const &node_id) {
  switch (node_id.identifierType) {
  case UA_NODEIDTYPE_NUMERIC:
    return Identifier(node_id.identifier.numeric);
  case UA_NODEIDTYPE_STRING:
    return Identifier(node_id.identifier.string);
  case UA_NODEIDTYPE_GUID:
    return Identifier(node_id.identifier.guid);
  case UA_NODEIDTYPE_BYTESTRING:
    return Identifier(node_id.identifier.byteString);
  }
}
```
To avoid this, I have to remove the addition of the variable node in `server.cpp`

```
   40     UA_Server_addVariableNode(server, myIntegerNodeId, parentNodeId,
   41                               parentReferenceNodeId, myIntegerName,
   42                               UA_NODEID_NULL, attr, NULL, NULL);
```
Any idea what could cause this segfault?

 - [ x] open62541 Version v0.3.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-04-04:
Did you already try to use some debugging tools which provide the stacktrace?

Eg gdb, valgrind, clang sanitizer

tobiaslocker, 2019-04-04:
yes i ran gdb to see the backtrace but the segfault happens only on the one variable node as described. The backtrace only points to code which seems to be memory safe. I will reproduce and add the backtrace and more details of the code

tobiaslocker, 2019-04-04:
Here is the backtrace:
```
Program received signal SIGSEGV, Segmentation fault.
0x00005555555606c5 in open62541::NodeId::NodeId(UA_NodeId const&) ()
(gdb) bt
#0  0x00005555555606c5 in open62541::NodeId::NodeId(UA_NodeId const&) ()
#1  0x0000555555561427 in open62541::Node::node_id() const ()
#2  0x000055555557f533 in http_connection::process_post_request()::{lambda(open62541::ReferenceDescription const&)#1}::operator()(open62541::ReferenceDescription const&) const ()
#3  0x00005555555872c0 in http_connection::process_post_request()::{lambda(open62541::ReferenceDescription const&)#1} std::for_each<__gnu_cxx::__normal_iterator<open62541::ReferenceDescription*, std::vector<open62541::ReferenceDescription, std::allocator<open62541::ReferenceDescription> > >, http_connection::process_post_request()::{lambda(open62541::ReferenceDescription const&)#1}>(__gnu_cxx::__normal_iterator<open62541::ReferenceDescription*, std::vector<open62541::ReferenceDescription, std::allocator<open62541::ReferenceDescription> > >, http_connection::process_post_request()::{lambda(open62541::ReferenceDescription const&)#1}, http_connection::process_post_request()::{lambda(open62541::ReferenceDescription const&)#1}) ()
#4  0x000055555557fe7c in http_connection::process_post_request() ()
#5  0x000055555557f00e in http_connection::process_request() ()
#6  0x000055555557ed83 in http_connection::read_request()::{lambda(boost::system::error_code, unsigned long)#1}::operator()(boost::system::error_code, unsigned long) const ()
#7  0x0000555555598aeb in void boost::beast::handler_ptr<boost::beast::http::detail::read_msg_op<boost::asio::basic_stream_socket<boost::asio::ip::tcp>, boost::beast::basic_flat_buffer<std::allocator<char> >, true, boost::beast::http::basic_dynamic_body<boost::beast::basic_multi_buffer<std::allocator<char> > >, std::allocator<char>, http_connection::read_request()::{lambda(boost::system::error_code, unsigned long)#1}>::data, {lambda(boost::system::error_code, unsigned long)#1}>::invoke<boost::system::error_code&, unsigned long&>(boost::system::error_code&, unsigned long&) ()
#8  0x000055555558d9a7 in boost::beast::http::detail::read_msg_op<boost::asio::basic_stream_socket<boost::asio::ip::tcp>, boost::beast::basic_flat_buffer<std::allocator<char> >, true, boost::beast::http::basic_dynamic_body<boost::beast::basic_multi_buffer<std::allocator<char> > >, std::allocator<char>, http_connection::read_request()::{lambda(boost::system::error_code, unsigned long)#1}>::operator()(boost::system::error_code, unsigned long, bool) ()
#9  0x00005555555a0bb3 in boost::beast::http::detail::read_some_op<boost::asio::basic_stream_socket<boost::asio::ip::tcp>, boost::beast::basic_flat_buffer<std::allocator<char> >, true, boost::beast::http::parser<true, boost::beast::http::basic_dynamic_body<boost::beast::basic_multi_buffer<std::allocator<char> > >, std::allocator<char> >, boost::beast::http::detail::read_msg_op<boost::asio::basic_stream_socket<boost::asio::ip::tcp>, boost::beast::basic_flat_buffer<std::allocator<char> >, true, boost::beast::http::basic_dynamic_body<boost::beast::basic_multi_buffer<std::allocator<char> > >, std::allocator<char>, http_connection::read_request()::{lambda(boost::system::error_code, unsigned long)#1}> >::operator()(boost::system::error_code, unsigned long, bool) ()
#10 0x00005555555e7ea1 in boost::asio::detail::binder2<boost::beast::http::detail::read_some_op<boost::asio::basic_stream_socket<boost::asio::ip::tcp>, boost::beast::basic_flat_buffer<std::allocator<char> >, true, boost::beast::http::parser<true, boost::beast::http::basic_dynamic_body<boost::beast::basic_multi_buffer<std::allocator<char> > >, std::allocator<char> >, boost::beast::http::detail::read_msg_op<boost::asio::basic_stream_socket<boost::asio::ip::tcp>, boost::beast::basic_flat_buffer<std::allocator<char> >, true, boost::beast::http::basic_dynamic_body<boost::beast::basic_multi_buffer<std::allocator<char> > >, std::allocator<char>, http_connection::read_request()::{lambda(boost::system::error_code, unsigned long)#1}> >, boost::system::error_code, unsigned long>::operator()() ()
#11 0x00005555555e3ed2 in void boost::asio::asio_handler_invoke<boost::asio::detail::binder2<boost::beast::http::detail::read_some_op<boost::asio::basic_stream_socket<boost::asio::ip::tcp>, boost::beast::basic_flat_buffer<std::allocator<char> >, true, boost::beast::http::parser<true, boost::beast::http::basic_dynamic_body<boost::beast::basic_multi_buffer<std::allocator<char> > >, std::allocator<char> >, boost::beast::http::detail::read_msg_op<boost::asio::basic_stream_socket<boost::asio::ip::tcp>, boost::beast::basic_flat_buffer<std::allocator<char> >, true, boost::beast::http::basic_dynamic_body<boost::beast::basic_multi_buffer<std::allocator<char> > >, std::allocator<char>, http_connection::read_request()::{lambda(boost::system::error_code, unsigned long)#1}> >, boost::system::error_code, unsigned long> >(boost::asio::detail::binder2<boost::beast::http::detail::read_some_op<boost::asio::basic_stream_socket<boost::asio::ip::tcp>, boost::beast::basic_flat_buffer<std::allocator<char> >, true, boost::beast:---Type <return> to continue, or q <return> to quit---

```

Here is the code where signal i noticed:

```
NodeId::NodeId(const UA_NodeId &node_id)
    : m_namespace_index{node_id.namespaceIndex}, m_id{Identifier::from_node_id(
                                                     node_id)} {}
```

which calls `from_node_id` as shown above

tobiaslocker, 2019-04-04:
Valgrind output:

I don't see where in my code there can be any dereferencing of 0x0 so there might be something wrong in the example?
```
==18058==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==18058== 
==18058== 
==18058== Process terminating with default action of signal 11 (SIGSEGV)
==18058==  Access not within mapped region at address 0x0
==18058==    at 0x1146C5: open62541::NodeId::NodeId(UA_NodeId const&) (in /home/me/Projects/open62541-client/build/client)
==18058==    by 0x115426: open62541::Node::node_id() const (in /home/me/Projects/open62541-client/build/client)
==18058==    by 0x133532: http_connection::process_post_request()::{lambda(open62541::ReferenceDescription const&)#1}::operator()(open62541::ReferenceDescription const&) const (in /home/me/Projects/open62541-client/build/client)
==18058==    by 0x13B2BF: http_connection::process_post_request()::{lambda(open62541::ReferenceDescription const&)#1} std::for_each<__gnu_cxx::__normal_iterator<open62541::ReferenceDescription*, std::vector<open62541::ReferenceDescription, std::allocator<open62541::ReferenceDescription> > >, http_connection::process_post_request()::{lambda(open62541::ReferenceDescription const&)#1}>(__gnu_cxx::__normal_iterator<open62541::ReferenceDescription*, std::vector<open62541::ReferenceDescription, std::allocator<open62541::ReferenceDescription> > >, http_connection::process_post_request()::{lambda(open62541::ReferenceDescription const&)#1}, http_connection::process_post_request()::{lambda(open62541::ReferenceDescription const&)#1}) (in /home/me/Projects/open62541-client/build/client)
==18058==    by 0x133E7B: http_connection::process_post_request() (in /home/me/Projects/open62541-client/build/client)
==18058==    by 0x13300D: http_connection::process_request() (in /home/me/Projects/open62541-client/build/client)
==18058==    by 0x132D82: http_connection::read_request()::{lambda(boost::system::error_code, unsigned long)#1}::operator()(boost::system::error_code, unsigned long) const (in /home/me/Projects/open62541-client/build/client)
==18058==    by 0x14CAEA: void boost::beast::handler_ptr<boost::beast::http::detail::read_msg_op<boost::asio::basic_stream_socket<boost::asio::ip::tcp>, boost::beast::basic_flat_buffer<std::allocator<char> >, true, boost::beast::http::basic_dynamic_body<boost::beast::basic_multi_buffer<std::allocator<char> > >, std::allocator<char>, http_connection::read_request()::{lambda(boost::system::error_code, unsigned long)#1}>::data, {lambda(boost::system::error_code, unsigned long)#1}>::invoke<boost::system::error_code&, unsigned long&>(boost::system::error_code&, unsigned long&) (in /home/me/Projects/open62541-client/build/client)
==18058==    by 0x1419A6: boost::beast::http::detail::read_msg_op<boost::asio::basic_stream_socket<boost::asio::ip::tcp>, boost::beast::basic_flat_buffer<std::allocator<char> >, true, boost::beast::http::basic_dynamic_body<boost::beast::basic_multi_buffer<std::allocator<char> > >, std::allocator<char>, http_connection::read_request()::{lambda(boost::system::error_code, unsigned long)#1}>::operator()(boost::system::error_code, unsigned long, bool) (in /home/me/Projects/open62541-client/build/client)
==18058==    by 0x154BB2: boost::beast::http::detail::read_some_op<boost::asio::basic_stream_socket<boost::asio::ip::tcp>, boost::beast::basic_flat_buffer<std::allocator<char> >, true, boost::beast::http::parser<true, boost::beast::http::basic_dynamic_body<boost::beast::basic_multi_buffer<std::allocator<char> > >, std::allocator<char> >, boost::beast::http::detail::read_msg_op<boost::asio::basic_stream_socket<boost::asio::ip::tcp>, boost::beast::basic_flat_buffer<std::allocator<char> >, true, boost::beast::http::basic_dynamic_body<boost::beast::basic_multi_buffer<std::allocator<char> > >, std::allocator<char>, http_connection::read_request()::{lambda(boost::system::error_code, unsigned long)#1}> >::operator()(boost::system::error_code, unsigned long, bool) (in /home/me/Projects/open62541-client/build/client)
==18058==    by 0x19BEA0: boost::asio::detail::binder2<boost::beast::http::detail::read_some_op<boost::asio::basic_stream_socket<boost::asio::ip::tcp>, boost::beast::basic_flat_buffer<std::allocator<char> >, true, boost::beast::http::parser<true, boost::beast::http::basic_dynamic_body<boost::beast::basic_multi_buffer<std::allocator<char> > >, std::allocator<char> >, boost::beast::http::detail::read_msg_op<boost::asio::basic_stream_socket<boost::asio::ip::tcp>, boost::beast::basic_flat_buffer<std::allocator<char> >, true, boost::beast::http::basic_dynamic_body<boost::beast::basic_multi_buffer<std::allocator<char> > >, std::allocator<char>, http_connection::read_request()::{lambda(boost::system::error_code, unsigned long)#1}> >, boost::system::error_code, unsigned long>::operator()() (in /home/me/Projects/open62541-client/build/client)
==18058==    by 0x197ED1: void boost::asio::asio_handler_invoke<boost::asio::detail::binder2<boost::beast::http::detail::read_some_op<boost::asio::basic_stream_socket<boost::asio::ip::tcp>, boost::beast::basic_flat_buffer<std::allocator<char> >, true, boost::beast::http::parser<true, boost::beast::http::basic_dynamic_body<boost::beast::basic_multi_buffer<std::allocator<char> > >, std::allocator<char> >, boost::beast::http::detail::read_msg_op<boost::asio::basic_stream_socket<boost::asio::ip::tcp>, boost::beast::basic_flat_buffer<std::allocator<char> >, true, boost::beast::http::basic_dynamic_body<boost::beast::basic_multi [... cut ...]
==18058==    definitely lost: 0 bytes in 0 blocks
==18058==    indirectly lost: 0 bytes in 0 blocks
==18058==      possibly lost: 0 bytes in 0 blocks
==18058==    still reachable: 78,458 bytes in 548 blocks
==18058==         suppressed: 0 bytes in 0 blocks
==18058== Rerun with --leak-check=full to see details of leaked memory
==18058== 
==18058== For counts of detected and suppressed errors, rerun with: -v
==18058== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault (core dumped)
```

tobiaslocker, 2019-04-04:
Lots of NULL and heap stuff done in server.cpp:
```
   23 int main() {
   24     signal(SIGINT, stopHandler);
   25     signal(SIGTERM, stopHandler);
   26    
   27     UA_ServerConfig *config = UA_ServerConfig_new_default(); 
   28     UA_Server *server = UA_Server_new(config);
   29    
   30     // add a variable node to the adresspace
   31     UA_VariableAttributes attr = UA_VariableAttributes_default;
   32     UA_Int32 myInteger = 42;
   33     UA_Variant_setScalarCopy(&attr.value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
   34     attr.description = UA_LOCALIZEDTEXT_ALLOC("en-US","the answer");
   35     attr.displayName = UA_LOCALIZEDTEXT_ALLOC("en-US","the answer");
   36     UA_NodeId myIntegerNodeId = UA_NODEID_STRING_ALLOC(1, "the.answer");
   37     UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME_ALLOC(1, "the answer");
   38     UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
   39     UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
   40     UA_Server_addVariableNode(server, myIntegerNodeId, parentNodeId,
   41                               parentReferenceNodeId, myIntegerName,
   42                               UA_NODEID_NULL, attr, NULL, NULL);
   43 
   44     /* allocations on the heap need to be freed */
   45     UA_VariableAttributes_deleteMembers(&attr);
   46     UA_NodeId_deleteMembers(&myIntegerNodeId);
   47     UA_QualifiedName_deleteMembers(&myIntegerName);
   48     
   49     UA_StatusCode retval = UA_Server_run(server, &running);
   50     UA_Server_delete(server);
   51     UA_ServerConfig_delete(config);
   52     return retval;
   53 }   
```

mlgiraud, 2019-04-04:
Could you supply a minimal example that reproduces the segfault?

jpfr, 2019-04-05:
> open62541::NodeId::NodeId

Really difficult to say anything about that.
We don't know this C++ wrapper.

tobiaslocker, 2019-04-05:
It's actually a work i have started. Right now it is still private but i would make it available if i make some progress. The error was a mistake by me because i have not yet implemented handling for string type identifiers. Sorry for the misleading and thanks for your help!

closed 2019-04-05 by tobiaslocker

tobiaslocker, 2019-05-10:
For informational purposes: I made the wrapper publicly available at https://github.com/tobiaslocker/open62541-cpp-wrapper