open62541/open62541 #2627: OPC UA WG NodeSet Update Draft - Please check for impacts on this Stack.

opcfoundation-org, 2019-04-05:
Can you take a look at the NodeSet draft in master branch and see if the changes have broken anything for you?
https://github.com/OPCFoundation/UA-Nodeset
 
DisplayName now omitted if same as BrowseName;
DataTypeDefintion.Name a String rather than a QualifiedName;

jpfr, 2019-04-05:
We have to update the nodeset-compiler.
Can we do this transparently to work with nodesets before and after the change?
There are lots of old nodesets in active use out there.

@Pro Did the change resolve this?
https://github.com/OPCFoundation/UA-Nodeset/pull/21

To get on with it, we can just maintain a patch file for the nodeset in our repo.
And apply that automatically in the build system.

opcfoundation-org, 2019-04-05:
The WG has not finalized these changes.

At this point we are looking for feedback on whether the schema change causes problems for existing implementations. If it does we may need to try something else.

opcfoundation-org, 2019-04-05:
I just pushed a fix for the AccessRestrictionType issue.

Pro, 2019-04-05:
I will do some tests and check if and how we need to change our Nodeset Compiler and report back here.

Thanks for the pingback and the fix for AccessRestrictionType

Pro, 2019-04-05:
@opcfoundation-org could you also port the fix for the AccessRestrictionType to the 1.04 branch?

Pro, 2019-04-05:
I just created a PR which adds support for optional display name in #2630 

If a display name is not given, we take the name part of the BrowseName.

This is definitely an improvement since we significantly reduce the generated code size (from 123000 loc to 116000 loc).

@opcfoundation-org this means that we can support the optional DisplayName. Older releases of course may have issues. But since we reference a specific commit for the nodeset, it's not that big of an issue.
We could also backport this change to 0.3

opcfoundation-org, 2019-04-05:
The unnecessary descriptions have also been removed which reduces the size.

One of the options for the DataTypeDefinition is to change the schema to make the name optional and leave it out entirely when it is the same as the BrowseName. This would further reduce the size of the NodeSet. 
 
Would this be a preferable change from your perspective?

Pro, 2019-04-10:
Supported with #2639 in master

closed 2019-04-10 by Pro