open62541/open62541 #2629: Read Data Proprety from node

bnadem, 2019-04-05:
i want to read proprety from node after UA_Client_Service_browse . read like userwritemask, writemask, historizing , usermask and datatype, should i read in every loop in Reference ()UA_Client_Service_browse nodeid this propreties, and wish function to discovery nodeid proprety (userwritemask, writemask, historizing , usermask and datatype)
thanks advice

## Description

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x ] Critical issue

jpfr, 2019-04-06:
Use the ‘read’ service.
And read the docs with the examples.

And no, this is not a critical issue...

closed 2019-04-06 by jpfr

bnadem, 2019-04-08:
with function UA_Client_Service_read ? i becom resoponse dataty UA_DataValue, wish ther are no attribute  for accessLevel  and  userAccessLevel     and Double minimumSamplingInterval;
 and    historizing;