open62541/open62541 #2634: VS2017 64bit doesn't compile v0.3.0 with UA_ENABLE_DISCOVERY_MULTICAST

royalheadwear, 2019-04-07:
## Description

I'm trying to compile the open62541-library using Visual Studio 2017 64bit with the DiscoveryMulticast-Feature enabled. When I run *Build* on the *ALL_BUILD*-project I run into the following error:

`<path-to>\open62541\src\server\ua_services_discovery_multicast.c(341):
 error C2220: warning treated as error - no 'object' file generated `
`<path-to>\open62541\src\server\ua_services_discovery_multicast.c(341): warning C4244: 'function': conversion from 'SOCKET' to 'int', possible loss of data`
`<path-to>\open62541\src\server\ua_services_discovery_multicast.c(631): warning C4244: 'function': conversion from 'SOCKET' to 'int', possible loss of data`

Because of this error the building of the *open62541-object*-project failes and the linker later sais

`9>LINK : fatal error LNK1181: cannot open input file '<path-to>\open62541\build\open62541-object.dir\Debug\ua_services_discovery_multicast.obj'
`

in the end the build process is summarized with

`========== Build All: 8 succeeded, 2 failed, 0 skipped ==========`

## Background Information / Reproduction Steps

I used CMake 3.12.0-rc2 (cmake-gui) for Windows:

1. checked out the branch 0.3 with the submodules
1. Selected \<path-to>/open62541 as "Where is the source code:"
1. Selected \<path-to>/open62541/build as "Where to build the binaries:"
1. clicked *Configure*
1. selected "Visual Studio 15 2017 Win64" to specify the generator for this project
1. enabled the checkbox at "UA_ENABLE_DISCOVERY_MULTICAST"
1. clicked *Generate*
1. opened the Project
1. right-clicked on the *ALL_BUILD*-project in the *Solution Explorer*
1. clicked *Build* in the context menu

## Checklist

 - [x] open62541 Version (release number or git tag): I checked out the [0.3 branch](https://github.com/open62541/open62541/tree/0.3)
 - [x] Other OPC UA SDKs used (client or server): nope
 - [x] Operating system: Windows 10 Pro 64 bit (Version 1803)
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached: not necessary
 - [ ] Self-contained code example attached: not necessary
 - [ ] Critical issue

royalheadwear, 2019-04-10:
I just found a solution to compile the library. The problem actually was that on Windows system the socket-variables seems to have it's own SOCKET datatype, while in non-windows operating systems socket-libraries a normal int variable is used.

The two Warnings in Line 341 and Line 631 occured, because a SOCKET variable was used in method calls that need a int variable at this place.

For me it worked to add a cast-operator with (int).
I couldn't verify if the compiled library will work correctly with this cast-operator.

Pro, 2019-04-30:
Thanks for the report!

Fixed in #2690

PR #2691 (2019-05-01): Stack: Correct cast of SOCKET to int on Windows
Fixes #2634

Same as #2690 for master
commit c8da8302 (2019-05-01): Stack: Correct cast of SOCKET to int on Windows

Fixes #2634
PR #2690 (2019-05-02): Stack: Correct cast of SOCKET to int on Windows
Fixes #2634

#2691 is same fix for master
commit fc61fe93 (2019-05-02): Stack: Correct cast of SOCKET to int on Windows

Fixes #2634
closed 2019-04-30 by Pro