open62541/open62541 #2636: Error: unknown reference type

xlh90, 2019-04-08:
## Description
Hi, I tried to create an OPC UA Server with the current open62541 Version. Therefore, I used two XML-files which is automatically created from the AML-editor (a NodeSet2 file and an xml-file that describes the AML information model). However, when running the code following error occurs:
"Exception: Reference ns=2;s=AutomationMLInterfaceClassLib/AutomationMLBaseInterface/Port.Direction<--[ns=0;i=47]--ns=2;s=AutomationMLInterfaceClassLib/AutomationMLBaseInterface/Port has an unknown reference type"

Does anyone know how I can solve this error?

[20190327153114_SystemModel_LinearAchse_Version1.zip](https://github.com/open62541/open62541/files/3055055/20190327153114_SystemModel_LinearAchse_Version1.zip)

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-04-08:
Please fill out the complete checklist, otherwise we are not able to help you. Especially we need the version you are using.

Building with `UA_NAMESPACE_ZERO=FULL` should fix your issue. If not provide a minimal reproducible code example and reopen the issue.

RBA21, 2019-05-23:
I have the same issue.

Pro, 2019-05-28:
It seems you did not load the nodeset correctly into the nodeset compiler.

I hacked together a demo project here:
https://github.com/Pro/open62541/tree/aml_testing
especially:
https://github.com/Pro/open62541/blob/aml_testing/examples/nodeset/CMakeLists.txt#L110

I do not see your error message. Probably you did not load the XML files in the correct order.

Anyways, #2768 fixes some minor issues with your nodeset.
Still, there is a logical error in your Nodeset:

`RuntimeError: Node ns=2;s=SystemModel_LinearAchse_Version1.aml@SchemaVersion (SchemaVersion) has an invalid type definition. ns=1;i=6004 is not a VariableType node.`

The `SystemModel...xml` file contains:

```xml
  <UAVariable NodeId="ns=2;s=SystemModel_LinearAchse_Version1.aml@SchemaVersion" BrowseName="2:SchemaVersion" ParentNodeId="ns=2;s=SystemModel_LinearAchse_Version1.aml" DataType="String">
    <DisplayName>SchemaVersion</DisplayName>
    <References>
      <Reference ReferenceType="HasTypeDefinition">ns=1;i=6004</Reference>
    </References>
    <Value>
      <String xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">3.0</String>
    </Value>
  </UAVariable>
```

which points to:
AMLBaseTypes.NodeSet2.xml

```xml
	<UAVariable NodeId="ns=1;i=6004" BrowseName="1:AMLSchemaVersion" ParentNodeId="ns=1;i=1002" DataType="String">
		<DisplayName>SchemaVersion</DisplayName>
		<References>
			<Reference ReferenceType="HasTypeDefinition">i=68</Reference>
			<Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=1002</Reference>
			<Reference ReferenceType="HasModellingRule">i=78</Reference>
		</References>
	</UAVariable>
```

This is not allowed. A HasTypeDefinition of a Variable shall always point to a VariableType node.

Excerpt from OPC UA Spec
[image]

You need to fix this first.

xlh90, 2019-06-05:
Thanks a lot for your help. We are currently trying to solve the problem with your descriptions.
I have a question regarding the logical error in our NodeSet: This unallowed reference, which you have mentioned above, is automatically created by the AML-PlugIn. Do we have to fix this logical error in order to use the open62541 implementation or would it just create a "wrong" OPC UA Server? 

Was it possible for you to create a server from the files I have uploaded or did you receive a different error message?

Thanks a lot in advance.

Pro, 2019-07-25:
Sorry for my late reply.

These errors need to be fixed in your NodeSet, or even better by raising an issue in the AML Plugin project.

We have strict type checking to avoid inconsistencies in the server nodeset. Therefore compilation or startup just stops after such an error occurs.

I this case the nodeset compiler already checks for a valid nodeset. In the next step the server initialization would check that too.

Due to this it's not possible to create a "wrong" OPC UA Server out of this nodeset. The only quick-fix you can do is to manually delete the problematic nodes from the nodeset.

On the long run I strongly recommend that you create an issue for the AML Plugin.

closed 2019-07-25 by Pro