open62541/open62541 #2637: Getting BadCertificateUriInvalid error while establishing client connection to server in latest master

bvinodkumar2008, 2019-04-08:
## Description
Getting BadCertificateUriInvalid error while establishing client connection to server in latest master but the same certs is working fine with older version of open62541(https://github.com/open62541/open62541/tree/e2d4b0bf0355dfe8a064889f69307169959059b9)

## Background Information / Reproduction Steps
**Run Server**
```
$ gcc -std=c99 -I. -g server_basic256sha256.c open62541.c -L. -lmbedtls -lmbedx509 -lmbedcrypto -pthread -o server_basic256sha256
$./server_basic256sha256 /etc/ssl/opcua/opcua_server_certificate.der /etc/ssl/opcua/opcua_server_key.der /etc/ssl/ca/ca_certificate.der
```
**Run Client**
```
$gcc -std=c99 -I. -g client_encryption.c open62541.c -L. -lmbedtls -lmbedx509 -lmbedcrypto -pthread -o client_encryption
$ ./client_encryption "opc.tcp://localhost:4840" /etc/ssl/opcua/opcua_client_certificate.der /etc/ssl/opcua/opcua_client_key.der /etc/ssl/ca/ca_certificate.der
```
Note: I'm not able to attach the .der certificates here. The opcua_server_certificate.der and opcua_client_certificate.der are signed by ca_certificate.der. The certs are available at https://github.com/bvinodkumar2008/opcua_certs

Created a gist for server.c, client.c and their respective logs at https://gist.github.com/bvinodkumar2008/49151adddcd0636a062f5c53f78a1b66 where it is working with the older version of open62541(https://github.com/open62541/open62541/tree/e2d4b0bf0355dfe8a064889f69307169959059b9) with the same certificates shared at https://github.com/bvinodkumar2008/opcua_certs. 

When the same certificates are used to run the encryption server and client samples, seeing the below server and client logs.

**server.log**:
```
./server_basic256sha256 /etc/ssl/opcua/opcua_server_certificate.der /etc/ssl/opcua/opcua_server_key.der /etc/ssl/ca/ca_certificate.der
[2019-04-08 22:03:02.473 (UTC+0550)] info/network       TCP network layer listening on opc.tcp://localhost:4840/
[2019-04-08 22:03:53.829 (UTC+0550)] info/network       Connection 5 | New connection over TCP from 127.0.0.1
[2019-04-08 22:03:53.829 (UTC+0550)] info/channel       Creating a new SecureChannel
[2019-04-08 22:03:53.829 (UTC+0550)] warn/securitypolicy        No PKI plugin set. Accepting all certificates
[2019-04-08 22:03:53.829 (UTC+0550)] info/channel       Connection 5 | SecureChannel 1 | Opened SecureChannel
[2019-04-08 22:03:53.830 (UTC+0550)] info/channel       Connection 5 | SecureChannel 1 | CloseSecureChannel
[2019-04-08 22:03:53.830 (UTC+0550)] info/network       Connection 5 | Closed
[2019-04-08 22:03:53.830 (UTC+0550)] info/network       Connection 5 | New connection over TCP from 127.0.0.1
[2019-04-08 22:03:53.854 (UTC+0550)] info/channel       Creating a new SecureChannel
[2019-04-08 22:03:53.894 (UTC+0550)] info/channel       Connection 5 | SecureChannel 2 | Opened SecureChannel
[2019-04-08 22:03:53.943 (UTC+0550)] info/channel       Connection 5 | SecureChannel 2 | CloseSecureChannel
[2019-04-08 22:03:53.943 (UTC+0550)] info/network       Connection 5 | Closed
```

**client.log**:
```
 ./client_encryption "opc.tcp://localhost:4840" /etc/ssl/opcua/opcua_client_certificate.der /etc/
ssl/opcua/opcua_client_key.der /etc/ssl/ca/ca_certificate.der
[2019-04-08 22:03:53.828 (UTC+0550)] info/client        Connecting to endpoint opc.tcp://localhost:4840
[2019-04-08 22:03:53.828 (UTC+0550)] info/client        SecurityPolicy not specified -> use default #None
[2019-04-08 22:03:53.829 (UTC+0550)] warn/securitypolicy        No PKI plugin set. Accepting all certificates
[2019-04-08 22:03:53.829 (UTC+0550)] info/client        TCP connection established
[2019-04-08 22:03:53.829 (UTC+0550)] info/client        Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2019-04-08 22:03:53.829 (UTC+0550)] info/client        Endpoint and UserTokenPolicy unconfigured, perform GetEndpoints
[2019-04-08 22:03:53.830 (UTC+0550)] info/client        Found 3 endpoints
[2019-04-08 22:03:53.830 (UTC+0550)] info/client        Rejecting endpoint 0: security mode doesn't match
[2019-04-08 22:03:53.830 (UTC+0550)] info/client        Rejecting endpoint 1: security mode doesn't match
[2019-04-08 22:03:53.830 (UTC+0550)] info/client        Endpoint 2 has 2 user token policies
[2019-04-08 22:03:53.830 (UTC+0550)] info/client        Selected Endpoint opc.tcp://localhost:4840 with SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2019-04-08 22:03:53.830 (UTC+0550)] info/client        Selected UserTokenPolicy open62541-anonymous-policy with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2019-04-08 22:03:53.830 (UTC+0550)] info/client        Disconnect to switch to a different SecurityPolicy
[2019-04-08 22:03:53.830 (UTC+0550)] info/client        Connecting to endpoint opc.tcp://localhost:4840
[2019-04-08 22:03:53.830 (UTC+0550)] info/client        TCP connection established
[2019-04-08 22:03:53.942 (UTC+0550)] info/client        Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2019-04-08 22:03:53.943 (UTC+0550)] error/userland     Error: BadCertificateUriInvalid
```

## Checklist

Please provide the following information:

 - [master] open62541 Version (release number or git tag):
 - [No ] Other OPC UA SDKs used (client or server): 
 - [Ubuntu 16.04] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-04-08:
Did you set the URI in the ApplicationDescription of client and server?

We probably need better logging for this...

bvinodkumar2008, 2019-04-08:
Hi Julius,
You mean in the certificates? Nope I have not but have the SAN field
populated in the certs. Could you please point me how to set it up? When I
see the diff of tools/cert between the master version and the older
open62541 version, don't see a much difference. Also, looked at the code
where it is thrown, see no difference in the section where this error code
is thrown

Does debug logs be suffice?

Regards,
Vinod

bvinodkumar2008, 2019-04-09:
Hi @jpfr ,

Adding the server and client trace logs (loglevel was set to 100 for open62541):

**server_basic256sha256 log**:
```
vkb1@vkb1-dev:~/opcua_samples$ ./server_basic256sha256 /etc/ssl/opcua/opcua_server_certificate.der /etc/ssl/opcua/opcua_server_key.der /etc/ssl/ca/ca_certificate.der
[2019-04-09 10:10:19.656 (UTC+0550)] debug/session      Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: No value given; Copy the value from the Typ
eDefinition
[2019-04-09 10:10:19.656 (UTC+0550)] debug/session      Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: No value given; Copy the value from the Typ
eDefinition
[2019-04-09 10:10:19.658 (UTC+0550)] debug/session      Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: No value given; Copy the value from the Typ
eDefinition
[2019-04-09 10:10:19.658 (UTC+0550)] debug/session      Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: No value given; Copy the value from the Typ
eDefinition
[2019-04-09 10:10:19.658 (UTC+0550)] debug/session      Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: No value given; Copy the value from the Ty$
eDefinition
[2019-04-09 10:10:19.658 (UTC+0550)] debug/session      Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: No value given; Copy the value from the Ty$
eDefinition
[2019-04-09 10:10:19.658 (UTC+0550)] debug/session      Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: No value given; Copy the value from the Typ
eDefinition
[2019-04-09 10:10:19.658 (UTC+0550)] info/network       TCP network layer listening on opc.tcp://localhost:4840/
[2019-04-09 10:10:25.230 (UTC+0550)] trace/network      Connection 5 | New TCP connection on server socket 3
[2019-04-09 10:10:25.230 (UTC+0550)] info/network       Connection 5 | New connection over TCP from 127.0.0.1
[2019-04-09 10:10:25.230 (UTC+0550)] trace/network      Connection 5 | Activity on the socket
[2019-04-09 10:10:25.230 (UTC+0550)] trace/network      Connection 5 | Received a packet.
[2019-04-09 10:10:25.230 (UTC+0550)] trace/network      Connection 5 | No channel attached to the connection. Process the chunk directly
[2019-04-09 10:10:25.230 (UTC+0550)] trace/network      Connection 5 | Activity on the socket
[2019-04-09 10:10:25.230 (UTC+0550)] trace/network      Connection 5 | Received a packet.
[2019-04-09 10:10:25.230 (UTC+0550)] trace/network      Connection 5 | No channel attached to the connection. Process the chunk directly
[2019-04-09 10:10:25.230 (UTC+0550)] trace/network      Connection 5 | Process OPN message
[2019-04-09 10:10:25.230 (UTC+0550)] info/channel       Creating a new SecureChannel
[2019-04-09 10:10:25.230 (UTC+0550)] warn/securitypolicy        No PKI plugin set. Accepting all certificates
[2019-04-09 10:10:25.230 (UTC+0550)] trace/channel      Connection 5 | SecureChannel 0 | Decrypting chunk
[2019-04-09 10:10:25.230 (UTC+0550)] trace/channel      Connection 5 | SecureChannel 0 | Chunk size before and after decryption: 132, 132
[2019-04-09 10:10:25.230 (UTC+0550)] trace/channel      Connection 5 | SecureChannel 0 | Verifying chunk signature
[2019-04-09 10:10:25.230 (UTC+0550)] trace/channel      Connection 5 | SecureChannel 0 | Decrypted and verified chunk with request id 1 and sequence number 1
[2019-04-09 10:10:25.230 (UTC+0550)] trace/channel      Connection 5 | SecureChannel 0 | Sequence Number processed: 1
[2019-04-09 10:10:25.230 (UTC+0550)] trace/channel      Connection 5 | SecureChannel 0 | Process an OPN on an open channel
[2019-04-09 10:10:25.230 (UTC+0550)] trace/channel      Connection 5 | SecureChannel 1 | Generating new local keys
[2019-04-09 10:10:25.230 (UTC+0550)] trace/channel      Connection 5 | SecureChannel 1 | Generating new remote keys
[2019-04-09 10:10:25.230 (UTC+0550)] info/channel       Connection 5 | SecureChannel 1 | Opened SecureChannel
[2019-04-09 10:10:25.230 (UTC+0550)] trace/network      Connection 5 | Activity on the socket
[2019-04-09 10:10:25.230 (UTC+0550)] trace/network      Connection 5 | Received a packet.
[2019-04-09 10:10:25.230 (UTC+0550)] trace/channel      Connection 5 | SecureChannel 1 | Decrypting chunk
[2019-04-09 10:10:25.230 (UTC+0550)] trace/channel      Connection 5 | SecureChannel 1 | Chunk size before and after decryption: 93, 93
[2019-04-09 10:10:25.230 (UTC+0550)] trace/channel      Connection 5 | SecureChannel 1 | Decrypted and verified chunk with request id 2 and sequence number 2
[2019-04-09 10:10:25.230 (UTC+0550)] trace/channel      Connection 5 | SecureChannel 1 | Sequence Number processed: 2
[2019-04-09 10:10:25.230 (UTC+0550)] trace/channel      Connection 5 | SecureChannel 1 | Process a MSG
[2019-04-09 10:10:25.230 (UTC+0550)] debug/session      Connection 5 | SecureChannel 1 | Session g=00000000-0000-0000-0000-000000000000 | Processing GetEndpointsRequest with endpointUrl "opc.
tcp://localhost:4840"
[2019-04-09 10:10:25.230 (UTC+0550)] trace/network      Connection 5 | Activity on the socket
[2019-04-09 10:10:25.230 (UTC+0550)] trace/network      Connection 5 | Received a packet.
[2019-04-09 10:10:25.230 (UTC+0550)] trace/channel      Connection 5 | SecureChannel 1 | Decrypting chunk
[... cut ...]
1f8pWv8rVWpv0JIJpTsro2JRufe+vWU5ZksyBEaH7AMISMdQk2SnIosmJ7vV4d7V
RIkfAnhG0nQaZmJb83n5i1tKnNN8I/qXB3ez1sGBq3CkUYqZQX+/GtMlr3UxFN0E
M/RoAvFs1nylBClrWg6kzRCu6xyi77HaxlgLA8KIvpOcpKutrGCaGnMHTs7FgVB6
Hutlw8tBVISVWAqx4Pf8PYchZP1gx4ug+YBvP8P/4zV7o0vizIW3G9HQktJs0dw4
PySMV4WA/g4Nlrd9+oESRCrD4whNvPcKy8Kegh6wcrEmfp8dCI5Wa92LIZi1W0xt
X4pjMSgQvhkCeg6yk5PRlvw3Ab0ak3750HOCybRhEQm+AUVqnBLPtIGlY9tzBr0J
BXE86gOgAhRv9dm6v2vR2JOc54lyktSd
-----END CERTIFICATE-----
```

Let me know if you need any further information, i'll be happy to provide you.

Pro, 2019-04-09:
In your server certificate you have:

```
X509v3 Subject Alternative Name: 
                DNS:localhost, DNS:opcua, DNS:10.223.210.50
```

This means that the ApplicationUri of the server needs to be set to one of those three values:
`config->applicationDescription->applicationUri`

It is recommended that you use a better FQDN for your SAN value, e.g. something like `opcua-server1.example.com`. This can be almost any string, but it should be included in the SAN field.

Pro, 2019-04-09:
Created a new issue #2638

bvinodkumar2008, 2019-04-09:
Thanks @Pro , let me quickly verify this and get back to you.

bvinodkumar2008, 2019-04-09:
Hi @Pro , 
I still see the same issue even after setting the one of those DNS names as the applicationUri. Here are the samples for server_basic256sha256 and client_encryption c files (tweaked little in server_basic256sha256 to accept hostname).

**server_basic256sha256**:
```
/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */

#include "open62541.h"

#include <signal.h>
#include <stdlib.h>

#include "common.h"

UA_Boolean running = true;
static void stopHandler(int sig) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
    running = false;
}

int main(int argc, char* argv[]) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    if(argc < 3) {
        UA_LOG_FATAL(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                     "Missing arguments. Arguments are "
                     "<server-certificate.der> <private-key.der> "
                     "[<trustlist1.crl>, ...]");
        return EXIT_FAILURE;
    }

    /* Load certificate and private key */
    UA_ByteString certificate = loadFile(argv[1]);
    UA_ByteString privateKey = loadFile(argv[2]);

    /* Load the trustlist */
    size_t trustListSize = 0;
    if(argc > 3)
        trustListSize = (size_t)argc-3;
    UA_STACKARRAY(UA_ByteString, trustList, trustListSize);
    for(size_t i = 0; i < trustListSize; i++)
        trustList[i] = loadFile(argv[i+3]);

    /* Loading of a revocation list currently unsupported */
    UA_ByteString *revocationList = NULL;
    size_t revocationListSize = 0;

    UA_ServerConfig *config =
        UA_ServerConfig_new_basic256sha256(4840, &certificate, &privateKey,
                                          trustList, trustListSize,
                                          revocationList, revocationListSize);

    **config->applicationDescription.applicationUri = UA_STRING("localhost");
    UA_ServerConfig_set_customHostname(config, UA_STRING("localhost"));**

    UA_ByteString_clear(&certificate);
    UA_ByteString_clear(&privateKey);
    for(size_t i = 0; i < trustListSize; i++)
        UA_ByteString_clear(&trustList[i]);

    if(!config) {
        UA_LOG_FATAL(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                     "Could not create the server config");
        return EXIT_FAILURE;
    }

    UA_Server *server = UA_Server_new(config);
    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;;
}
```

**client_encryption.c**:
```
/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */

#include "open62541.h"

#include <stdlib.h>

#include "common.h"

#define MIN_ARGS 4

int main(int argc, char* argv[]) {
    if(argc < MIN_ARGS) {
        UA_LOG_FATAL(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                     "Arguments are missing. The required arguments are "
                     "<opc.tcp://host:port> "
                     "<client-certificate.der> <client-private-key.der> "
                     "[<trustlist1.crl>, ...]");
        return EXIT_FAILURE;
    }

    const char *endpointUrl = argv[1];

    /* Load certificate and private key */
    UA_ByteString certificate = loadFile(argv[2]);
    UA_ByteString privateKey  = loadFile(argv[3]);

    /* Load the trustList. Load revocationList is not supported now */
    size_t trustListSize = 0;
    if(argc > MIN_ARGS)
        trustListSize = (size_t)argc-MIN_ARGS;
    UA_STACKARRAY(UA_ByteString, trustList, trustListSize);
    for(size_t trustListCount = 0; trustListCount < trustListSize; trustListCount++)
        trustList[trustListCount] = loadFile(argv[trustListCount+4]);

    UA_ByteString *revocationList = NULL;
    size_t revocationListSize = 0;

    UA_Client *client = UA_Client_new();
    UA_ClientConfig *cc = UA_Client_getConfig(client);
    cc->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
    UA_ClientConfig_setDefaultEncryption(cc, certificate, privateKey,
                                         trustList, trustListSize,
                                         revocationList, revocationListSize);

    UA_ByteString_clear(&certificate);
    UA_ByteString_clear(&privateKey);
    for(size_t deleteCount = 0; deleteCount < trustListSize; deleteCount++) {
        UA_ByteString_clear(&trustList[deleteCount]);
    }

    /* Secure client connect */
    cc->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT; /* require encryption */
    UA_StatusCode retval = UA_Client_connect(client, endpointUrl);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Error: %s", UA_StatusCode_name(retval));
        UA_Client_delete(client);
        return EXIT_FAILURE;
    }

    UA_Variant value;
    UA_Variant_init(&value);

    /* NodeId of the variable holding the current time */
    const UA_NodeId nodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_CURRENTTIME);
    retval = UA_Client_readValueAttribute(client, nodeId, &value);

    if(retval == UA_STATUSCODE_GOOD &&
       UA_Variant_hasScalarType(&value, &UA_TYPES[UA_TYPES_DATETIME])) {
        UA_DateTime raw_date  = *(UA_DateTime *) value.data;
        UA_DateTimeStruct dts = UA_DateTime_toStruct(raw_date);
        UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "date is: %u-%u-%u %u:%u:%u.%03u\n",
                    dts.day, dts.month, dts.year, dts.hour, dts.min, dts.sec, dts.milliSec);
    }

    /* Clean up */
    UA_Variant_clear(&value);
    UA_Client_delete(client);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}

```

jpfr, 2019-04-13:
There is now logging via #2648.

You might have to set the URI instead a DNS in the certificate. Similar to how we do it in the generation script in /tools/certs.

bvinodkumar2008, 2019-04-14:
Thanks @jpfr . I will check and revert back by EOD tomorrow.

bvinodkumar2008, 2019-04-15:
It's working @jpfr, thanks for swift response. I'm closing this.

closed 2019-04-15 by bvinodkumar2008