open62541/open62541 #2640: arch/ua_network_tcp.c  UA_ClientConnectionTCP_poll()

jswangjunsheng, 2019-04-09:
arch/ua_network_tcp.c  line 662   in UA_ClientConnectionTCP_poll()   
`#ifdef _OS9000
        /* OS-9 can't use select for checking write sockets.
         * Therefore, we need to use connect until success or failed
         */
        UA_UInt32 timeout_usec = (tcpConnection->timeout - timeSinceStart)
                        * 1000;
        int resultsize = 0;
        do {
            u_int32 time = 0x80000001;
            signal_code sig;

            timeout_usec -= 1000000/256;    // Sleep 1/256 second
            if (timeout_usec < 0) // uint will never less than 0
                break;
`
UA_UInt32 timeout_usec --> UA_DateTime timeout_usec

jpfr, 2019-04-13:
I‘m not sure what you want to say here...
Can you explain or send us a patch?

jswangjunsheng, 2019-04-16:
> I‘m not sure what you want to say here...
> Can you explain or send us a patch?

arch/ua_network_tcp.c line 662
the type of timeout_usec, can not be UA_UInt32
maybe UA_DateTime will be ok.
because line 669 

            timeout_usec -= 1000000/256;    // Sleep 1/256 second
            if (timeout_usec < 0)   // always true, because timeout_usec is  uint32 
                break;

timeout_usec will never less than 0, becasue its type is uint32
so the while will never stop

teccheck, 2025-03-18:
This code path was overhauled as part of the new event loop, so I'll close this

closed 2025-03-18 by teccheck