open62541/open62541 #2642: Pub/Sub stops working after 1:05 minutes

Marioam, 2019-04-11:
Hey all,
I am currently working on porting the open62541 for ESP32 (freertos/LwIP).
I managed to get a Publisher and a Subscriber running on the ESP32 but it always runs exactly 1 minute and 5 seconds. After this time it just stops.
For example the Publisher stops publishing although the task is still running. I tried different publishing intervals (100ms, 1s, 10s) but this has no effect. No matter how many messages are published, it always stops after this time.
I checked the free heap space and there is enough free.
The same code (Publisher and Subscriber) runs perfectly fine on a Raspberry Pi 3B+.
Anyone faces a similar issue?

 - [ ] open62541 Version (release number or git tag): current master-branch
 - [ ] Other OPC UA SDKs used (client or server): No other OPC UA SDKs are used!
 - [ ] Operating system: freeRTOS/LwIP
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-04-11:
Did you already check out #2511?
Also make sure that you are using current master branch.
There were similar issues with subscriptions especially on freertos.

Marioam, 2019-04-11:
@Pro Yes I checked the problem with the server time but this doesn't seems to be the issue here.

jpfr, 2019-04-13:
Can you print a log message in the publish callback.
If the log continues to be printed, it is likely a problem of the network stack.

closed 2025-03-18 by teccheck