open62541/open62541 #2645: Updating/Writing a Variable to an ObjectNode

gedoensmax, 2019-04-11:
## Description
I am trying to build an OPC UA Server which monitors a REST API. The code for the Object Class is generated from an XML file as explained in the documentation from open62541.

## Background Information / Reproduction Steps

I have also tried to use the parent ID and i have also tried to browse for references of both nodes and use these NodeIds in the write dunction but nothing is working. I always get the error "Write Request returned Status code BadNodeClassInvalid" although the retval inside nodeIter() is always 0.

`#define COUNTER_NODE_ID

UA_Server *server;

static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

static UA_StatusCode
nodeIter(UA_NodeId childId, UA_Boolean isInverse, UA_NodeId referenceTypeId, 
         void *handle) {
    UA_NodeId *parent = (UA_NodeId *)handle;
    printf("%d, %d --- %d ---> NodeId %d, %d\n",
           parent->namespaceIndex, parent->identifier.numeric,
           referenceTypeId.identifier.numeric, childId.namespaceIndex,
           childId.identifier.numeric);
                ...
                get_variable_from_rest_api();
                ...
                //Update the OPC-UA node
                UA_Variant value;
                UA_Float newFloat = (UA_Float)current_element>valuedouble;
                UA_Variant_setScalarCopy(&value, &newFloat, 
                                         &UA_TYPES[UA_TYPES_FLOAT]);
                retval |= UA_Server_writeValue(server, childId, value);
                // retval is 0
    return retval;
}

//Function that the server will call each time that the information is updated
void *apiWatcher(void *ptr) {
    UA_NodeId *parent = UA_NodeId_new();
    *parent = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    while (running == true) {
        UA_Server_forEachChildNodeCall(server, UA_NODEID_NUMERIC(0, COUNTER_NODE_ID), 
                                       nodeIter, (void *) parent);
        usleep(500);
    }
    UA_NodeId_delete(parent);
    return ptr;
}`

`int main(int argc, char **argv) {

    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    pthread_t threadSensor;

    UA_ServerConfig *config = UA_ServerConfig_new_default();
    server = UA_Server_new(config);

    ....
    UA_Server_addObjectNode(server, UA_NODEID_NUMERIC(1, 0),
                            UA_NODEID_NUMERIC(0,UA_NS0ID_OBJECTSFOLDER), 
                            UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES), 
                            UA_QUALIFIEDNAME(1, "FMU"), 
                            UA_NODEID_NUMERIC(2, COUNTER_NODE_ID), 
                            object_attr, NULL, &createdNodeId);

        if(pthread_create( &threadSensor, NULL, apiWatcher, server ))
        {
            fprintf(stderr,"Error - pthread_create(): %d\n",ret);
            exit(EXIT_FAILURE);
        }
        retval = UA_Server_run(server, &running);
    }
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return (int) retval;
}`

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ubuntu ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2019-04-12:
Depending on your title "Updating/Writing a Variable to an ObjectNode" and the error "Write Request returned Status code BadNodeClassInvalid" I would assume, that you try to write a value into an ObjectNode, which is impossible. Only VariableNodes have an value attribute in OPC UA! So if you need to write a value into that particular node you need to change it from an ObjectNode to a VariableNode. (Variables could also have child nodes).

gedoensmax, 2019-04-12:
Oh sorry i forgot to attach the XML example i updated the issue accordingly.

ccvca, 2019-04-13:
I've noticed the following things in your code:

- In _apiWatcher_, you are iterating over UA_NODEID_NUMERIC(**0**, COUNTER_NODE_ID), while in _main_ an Object with NodeId UA_NODEID_NUMERIC(**2**, COUNTER_NODE_ID) is created.
- Your XML-File has no namespace definition. (e.g. as defined here: https://github.com/OPCFoundation/UA-Nodeset/blob/master/CNC/Opc.Ua.CNC.NodeSet.xml#L33 ,as this is the first (and only) definition in this file, all NodeIds refer to this as ns 1 , e.g. **ns=1**;i=1000 ), so you are trying to put everything in Namespace 0, which should not be modified, as it is defined by the OPC Foundation. It looks like the generator put your nodes automaticaly in ns=2 in your server.

gedoensmax, 2019-04-14:
I checked different Namespaces and it seems that everything is in Namespace 0.
0, 85 --- 47 ---> NodeId 0, 1003
0, 85 --- 47 ---> NodeId 0, 1001
0, 85 --- 47 ---> NodeId 0, 1002
0, 85 --- 47 ---> NodeId 0, 1005
0, 85 --- 47 ---> NodeId 0, 1004
0, 85 --- 40 ---> NodeId 0, 58
This is my output from the print command in nodeIter().
But i will try redoing the xml with a different namespace.

ccvca, 2019-04-14:
Oh, I screw up the function signature of UA_Server_addObjectNode. 
These lines add an Object of type UA_NODEID_NUMERIC(2, COUNTER_NODE_ID), not to namespace 2 as I thought before.
> UA_Server_addObjectNode(server, UA_NODEID_NUMERIC(1, 0),
>                            UA_NODEID_NUMERIC(0,UA_NS0ID_OBJECTSFOLDER),
>                            UA_NODEID_NUMERIC(0, [...]

Can you try to get the exact node, which produces this Error:
"Write Request returned Status code BadNodeClassInvalid"?

I just realized, that you pass (and output) a parent-NodeId to NodeIter, which points to ns=0;i=UA_NS0ID_OBJECTSFOLDER, while you are iteration over UA_NODEID_NUMERIC(0, COUNTER_NODE_ID) in
>UA_Server_forEachChildNodeCall(server, UA_NODEID_NUMERIC(0, COUNTER_NODE_ID), 
>                                       nodeIter, (void *) parent);

The second parameter is the parent:
https://github.com/open62541/open62541/blob/cca3ead344186f9c0aac0e177263c48c52e49c9c/src/server/ua_server.c#L93-L95

gedoensmax, 2019-04-15:
```
[2019-04-15 09:50:15.621 (UTC+0200)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Node type not found in nodestore
0, 85 --- 46 ---> NodeId 0, 1006
[2019-04-15 09:50:15.624 (UTC+0200)] info/network	TCP network layer listening on opc.tcp://macbook-mmueller.local:4840/
0, 85 --- 47 ---> NodeId 0, 1003
0, 85 --- 47 ---> NodeId 0, 1001
0, 85 --- 47 ---> NodeId 0, 1002
0, 85 --- 47 ---> NodeId 0, 1005
0, 85 --- 47 ---> NodeId 0, 1004
0, 85 --- 40 ---> NodeId 0, 58
[2019-04-15 09:50:15.703 (UTC+0200)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | WriteRequest returned status code BadNodeClassInvalid
[2019-04-15 09:50:15.703 (UTC+0200)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | WriteRequest returned status code BadNodeClassInvalid
[2019-04-15 09:50:15.703 (UTC+0200)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | WriteRequest returned status code BadNodeClassInvalid
[2019-04-15 09:50:15.703 (UTC+0200)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | WriteRequest returned status code BadNodeClassInvalid
[2019-04-15 09:50:15.703 (UTC+0200)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | WriteRequest returned status code BadNodeClassInvalid
[2019-04-15 09:50:15.703 (UTC+0200)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | WriteRequest returned status code BadNodeClassInvalid
```

This is the output i get from the print statement in nodeIter(). So i would say the nodeIter works as it should and just my write request fails. And it fails at every write. If i set a breakpoint it the childId also has exactly the value i want. (UA_NODEID_NUMERIC(0,100x))

ccvca, 2019-04-16:
Ok, I do not have any more idead, why it fails.
Do you know the source of this error:
> Node type not found in nodestore

gedoensmax, 2019-04-23:
I changed my XML to this simplified version and got rid of the error you mentioned. But i still get the error from before, that i cannot write to it.
```
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
           xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd"
           xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd"
           xmlns:s1="http://yourorganisation.org/example_nodeset/"
           xmlns:xsd="http://www.w3.org/2001/XMLSchema">
  <NamespaceUris>
    <Uri>http://yourorganisation.org/example_nodeset/</Uri>
  </NamespaceUris>
  <Aliases>
    <Alias Alias="Float">i=10</Alias>
    <Alias Alias="HasModellingRule">i=37</Alias>
    <Alias Alias="HasTypeDefinition">i=40</Alias>
    <Alias Alias="HasSubtype">i=45</Alias>
    <Alias Alias="HasProperty">i=46</Alias>
    <Alias Alias="HasComponent">i=47</Alias>
    <Alias Alias="Date">i=293</Alias>
  </Aliases>
  <UAObjectType BrowseName="1:StandardBuilding" NodeId="ns=1;i=1001">
    <DisplayName>StandardBuilding</DisplayName>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=58</Reference>
      <Reference ReferenceType="HasComponent">ns=1;i=2001</Reference>
    </References>
  </UAObjectType>
  <UAVariable BrowseName="1:slack.P" DataType="Float" NodeId="ns=1;i=2001" ParentNodeId="ns=1;i=1001">
    <DisplayName>slack.P</DisplayName>
    <References>
      <Reference IsForward="false" ReferenceType="HasComponent">ns=1;i=1001</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
      <Reference ReferenceType="HasModellingRule">i=78</Reference>
    </References>
    <Value>
      <uax:Float>8.0</uax:Float>
    </Value>
  </UAVariable>
</UANodeSet>
```

gedoensmax, 2019-04-24:
I fixed it with the above XML and iterating over the Objectfolder by calling:
`UA_Server_forEachChildNodeCall(server, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER), objectNodeIter, (void*)parent);`

Afterwards I iterate again over all Nodes in my used namespace:
```
static UA_StatusCode
objectNodeIter(UA_NodeId childId, UA_Boolean isInverse, UA_NodeId referenceTypeId, void *handle) {
    if(isInverse)
        return UA_STATUSCODE_GOOD;
    UA_NodeId *parent = (UA_NodeId *)handle;
//    printf("%d, %d --- %d ---> NodeId %d, %d\n",
//           parent->namespaceIndex, parent->identifier.numeric,
//           referenceTypeId.identifier.numeric, childId.namespaceIndex,
//           childId.identifier.numeric);
    if (referenceTypeId.identifier.numeric == 35 && childId.namespaceIndex == 1)
        return UA_Server_forEachChildNodeCall(server, childId, nodeIter, (void*)parent);
}
```

Correct me if there are any major faults. At least it works.

closed 2019-04-24 by gedoensmax