open62541/open62541 #2647: How can multi-dimensional array be added to variable value callback?

santoshpanda1989, 2019-04-12:
## Description 
How can multi-dimensional array be added to variable value callback? I want to add a matrix to a variable when it is subscribe through a UA client. For e.g. a 5X2 array. I can do it for 2D array but bit confused how to proceed for  multi-dimensional array.

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Master
 - [ ] Other OPC UA SDKs used (client or server): UAExpert
 - [ ] Operating system: Ubuntu
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-04-12:
Every Variant has an ArrayDimensions field.
Set the core data to a 10-element array.
And the ArrayDimensions to {5,2}.

closed 2019-04-21 by jpfr

santoshpanda1989, 2019-04-24:
I have creates a sample array and tried to assign the dimensions for the Variant. But, I am getting BadTypeMismatch error. PFB the server screenshot. 
[image]
Also, find below the the nodeset.xml snippet.
**<UAVariable DataType="Double" ParentNodeId="ns=1;i=5008" ValueRank="2" NodeId="ns=1;i=6010" ArrayDimensions="0,0" BrowseName="1:ProcessValue" AccessLevel="3">
        <DisplayName>ProcessValue</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=5008</Reference>
        </References>
    </UAVariable>**

Find the code snippet below.
UA_Variant v;
    UA_Double d[9] = {1.0, 2.0, 3.0, 4.0, 5.0, 6.0, 7.0, 8.0, 9.0};

    /* Set array dimensions */
    v.arrayDimensions = (UA_UInt32 *)UA_Array_new(9, &UA_TYPES[UA_TYPES_UINT32]);
    v.arrayDimensionsSize = 2;
    v.arrayDimensions[0] = 3;
    v.arrayDimensions[1] = 3;
    UA_Variant_setArrayCopy(&v, d, 9, &UA_TYPES[UA_TYPES_DOUBLE]);
    UA_NodeId currentNodeId = UA_NODEID_NUMERIC(3, 6010);
    UA_Server_writeValue(server, currentNodeId, v);

I would really appreciate any help regarding this.

Thanks,
Santosh