open62541/open62541 #2650: Single file distribution not compiling

darxridge, 2019-04-15:
The very basic tutorial example fails to compile with the single file distribution:

Git-Revision: v0.3.0 and according tutorial

OS: Debian stretch, amd64

Single file distribution + myServer.c from Tutorial

$ gcc -std=c99 -DUA_ARCHITECTURE_POSIX open62541.c myServer.c -o server
> myServer.c:1:41: fatal error: open62541/plugin/log_stdout.h: No such file or directory
>  #include <open62541/plugin/log_stdout.h> 
> compilation terminated.

jpfr, 2019-04-15:
> open62541/plugin/log_stdout.h

This file does not exist with the single-file distribution.
Just include open62541.h and nothing else.

darxridge, 2019-04-15:
Thank you for the quick reply. I managed to compile the example with the following includes:
>#include "open62541.h"
>#include <signal.h>
>#include <stdlib.h>

closed 2019-04-15 by Pro

Aliazzzz, 2019-12-08:
Same issue, newer version;
The very basic tutorial example fails to compile with the single file distribution:
Git-Revision: v1.0 and according tutorial
OS: Debian stretch, amd64
Single file distribution + myServer.c from Tutorial

Why hasn't the documentation been updated?
Found all kinds of undocumented issue's when following the documentation on the letter.