open62541/open62541 #2652: Assert in ua_securechannel sendSymmetricChunk

MampfHamster, 2019-04-15:
## Description

Hello,

I have an Issue that results in a "false" Statement in 

ua_secureChannel.c

    static UA_StatusCode sendSymmetricChunk(UA_MessageContext *messageContext){
    ...
    UA_assert(total_length <= connection->config.sendBufferSize) 
    ...
    }

I build a UA_Client that reads a custom struct datatype which contains a large ByteArray. There were some Bugs in that Client that lead to random crashes, these Bugs are my own Problem and not part of the Issue i try to discribe. While my Client crashed i realized that it also chrashed the Server because of the before mentiont UA_assert. I think the Problem is in:

ua_types_encoding_binary.c 

    static status
    encodeBinaryStruct(const void *src, const UA_DataType *type, Ctx *ctx) {
    /* Check the recursion limit */
    if(ctx->depth > UA_ENCODING_MAX_RECURSION)
        return UA_STATUSCODE_BADENCODINGERROR;
    ctx->depth++;

    uintptr_t ptr = (uintptr_t)src;
    status ret = UA_STATUSCODE_GOOD;
    u8 membersSize = type->membersSize;
    const UA_DataType *typelists[2] = { UA_TYPES, &type[-type->typeIndex] };

    /* Loop over members */
    for(size_t i = 0; i < membersSize; ++i) {
        const UA_DataTypeMember *m = &type->members[i];
        const UA_DataType *mt = &typelists[!m->namespaceZero][m->memberTypeIndex];
        ptr += m->padding;

        /* Array. Buffer-exchange is done inside Array_encodeBinary if required. */
        if(m->isArray) {
            const size_t length = *((const size_t*)ptr);
            ptr += sizeof(size_t);
            ret = Array_encodeBinary(*(void *UA_RESTRICT const *)ptr, length, mt, ctx);
            ptr += sizeof(void*);
            continue;
        }

        /* Scalar */
        ret = encodeWithExchangeBuffer((const void*)ptr, mt, ctx);
        ptr += mt->memSize;
    }

    ctx->depth--;
    return ret;
}

The Server steped through the for-loop even tough the return Value of encodeWithExchangeBuffer() was already bad. I added

    for(size_t i = 0; i < membersSize; ++i) {
    ...
        ret = encodeWithExchangeBuffer((const void*)ptr, mt, ctx);
        ptr += mt->memSize;

		if (ret != UA_STATUSCODE_GOOD) {
			break;
		}
			
    }  		
  

at the bottom of the for-loop which seems to be working for me.
Maybe you can check if this leads to other problems somewhere else?
Or if not add the change to the stack.

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 65bdb77f
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 7 & Ubuntu 18.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2022-11-14:
Hey there,

were you able to reproduce with one of the more recent stable releases 1.2 or 1.3?

MampfHamster, 2022-11-15:
Hi,

I totally forgot that this was still open.
I never encountered the problem with newer builds of the stack, so I would consider it resolved

Mit freundlichen Grüßen / Best Regards

Jens Rademacher
Software

BST GmbH
Remusweg 1
33729 Bielefeld, Germany

        [cid:bst_70c299e1-fa1d-4bdb-af7c-a861f6c6c59e.png] <https://bst.elexis.group/>

Phone:

+49 521 400 70 537
                <https://www.youtube.com/channel/UCb_Te7XIX1rcbdlbbJkcK8w> [cid:bst_youtube_66e703f7-392b-4c96-afaf-de3038112dd0.png] <https://www.youtube.com/c/BSTInternationalGmbHBielefeld>       [cid:Insta_a13ccf6f-240f-4ef3-bb4b-96dafb0290e3.jpg] <https://www.instagram.com/wearebstgroup/>

Fax:    +49 521 400 70 999
Web:    https://bst.elexis.group/
Amtsgericht / District Court: Bielefeld, HRB Nr. 30830
Geschäftsführung / Management board: Alexander Heck (Vors.), Ingo Koesfeld
Sitz der Gesellschaft / Location: Bielefeld, Deutschland
Vertrauliche E-Mail von / Confidential e-mail from BST GmbH
Angaben zur gesetzlichen Informationspflicht nach Artikel 13/14 DSGVO finden Sie HIER<https://www.bst.elexis.group/fileadmin/user_upload/bst/11_Documents/2022-02-09_BST_Informationspflicht_DSGVO_13_14_final_d.pdf>.
Information on the legal duty according to article 13/14 GDPR can be found HERE<https://www.bst.elexis.group/fileadmin/user_upload/bst/11_Documents/2022-02-09_BST_Informationspflicht_DSGVO_13_14_final_en.pdf>.

Please consider the environment before printing this email.

MampfHamster, 2022-11-15:
Hi,
I never encountered that problem on newer builds so i would consider this resolved

closed 2022-11-15 by MampfHamster

jpfr, 2022-11-16:
@MampfHamster 👍