open62541/open62541 #2654: Deleting instance of custom type deletes methods of all other instances

juangburgos, 2019-04-16:
## Description

I think is related to #1730 and #1812. Since the issue is aknowledged, I am wondering if there is any workaround available? I attach code example.

### Before calling `deleteInstance1`

All instances have the method.

[image]

### After calling `deleteInstance1`

Remaining instances have no method

[image]

Thanks

## Background Information / Reproduction Steps

```c
#include <signal.h>
#include "open62541.h"

static UA_StatusCode
methodCallback(UA_Server *server,
                         const UA_NodeId *sessionId, void *sessionHandle,
                         const UA_NodeId *methodId, void *methodContext,
                         const UA_NodeId *objectId, void *objectContext,
                         size_t inputSize, const UA_Variant *input,
                         size_t outputSize, UA_Variant *output) {
    return UA_STATUSCODE_GOOD;
}

static UA_NodeId myObjectTypeNodeId;
static void addMyObjectType(UA_Server *server) 
{
	// create variable type attributes
	UA_ObjectTypeAttributes otAttr = UA_ObjectTypeAttributes_default;
	otAttr.displayName             = UA_LOCALIZEDTEXT("", "MyObjectType");
	// add new object type
	UA_QualifiedName browseName;
	browseName.namespaceIndex = 1;
	browseName.name = UA_STRING_ALLOC("MyObjectType");
	UA_StatusCode st = UA_Server_addObjectTypeNode(
		server,
		UA_NODEID_NULL,                                // requested nodeId
		UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE), // parent (variable type)
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),     // parent relation with child
		browseName,
		otAttr,
		(void*)"MyObjectType",                         // context 
		&myObjectTypeNodeId                            // new variable type id
	);
	assert(st == UA_STATUSCODE_GOOD);
	// add variable to object type
	UA_VariableAttributes vAttr = UA_VariableAttributes_default;
	vAttr.displayName = UA_LOCALIZEDTEXT("", "MyVariable");
	UA_NodeId childNodeId;
	st = UA_Server_addVariableNode(
		server,
		UA_NODEID_NULL,                                      // requested nodeId
		myObjectTypeNodeId,                                  // parent
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),         // parent relation with child
		browseName,
		UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), // type id
		vAttr,
		(void *)"MyVariableOtherType",                       // context
		&childNodeId                                         // output nodeId to make mandatory
	);
	assert(st == UA_STATUSCODE_GOOD);
	// make child mandatory
	st = UA_Server_addReference(server,
		                        childNodeId,
		                        UA_NODEID_NUMERIC(0, UA_NS0ID_HASMODELLINGRULE),
		                        UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_MODELLINGRULE_MANDATORY), 
		                        true);
	assert(st == UA_STATUSCODE_GOOD);
	// add method to object
	UA_NodeId methodNodeId;
	UA_MethodAttributes helloAttr = UA_MethodAttributes_default;
	helloAttr.description = UA_LOCALIZEDTEXT("", "Does nothing but exist. Or does it even exist?");
	helloAttr.displayName = UA_LOCALIZEDTEXT("", "emptyMethod");
	helloAttr.executable = true;
	helloAttr.userExecutable = true;
	st = UA_Server_addMethodNode(server, UA_NODEID_STRING(1, "emptyMethod"),
		myObjectTypeNodeId,
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASORDEREDCOMPONENT),
		UA_QUALIFIEDNAME(1, "emptyMethod"),
		helloAttr, &methodCallback,
		0, NULL, 0, NULL, NULL, &methodNodeId);
	assert(st == UA_STATUSCODE_GOOD);
	// make method mandatory
	st = UA_Server_addReference(server,
		methodNodeId,
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASMODELLINGRULE),
		UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_MODELLINGRULE_MANDATORY),
		true);
	assert(st == UA_STATUSCODE_GOOD);
}

static UA_NodeId addMyObjectInstance(const UA_NodeId parentNodeId, UA_Server * server, char * browseName)
{
	UA_NodeId myObjectInstanceNodeId;
	UA_ObjectAttributes oAttr = UA_ObjectAttributes_default;
	oAttr.displayName = UA_LOCALIZEDTEXT("", browseName);
	// add variable
	UA_QualifiedName qualBrowseName;
	qualBrowseName.namespaceIndex = 1;
	qualBrowseName.name = UA_STRING(browseName);
	auto st = UA_Server_addObjectNode(server,
		UA_NODEID_NULL,                              // requested nodeId
		parentNodeId,                                // parent
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT), // parent relation with child
		qualBrowseName,
		myObjectTypeNodeId,
		oAttr,
		(void*)browseName,                           // context
		&myObjectInstanceNodeId);                    // new variable instance id
	assert(st == UA_STATUSCODE_GOOD);
	return myObjectInstanceNodeId;
}

static UA_NodeId gInstance1;
static UA_StatusCode
deleteMethodCallback(UA_Server *server,
	const UA_NodeId *sessionId, void *sessionHandle,
	const UA_NodeId *methodId, void *methodContext,
	const UA_NodeId *objectId, void *objectContext,
	size_t inputSize, const UA_Variant *input,
	size_t outputSize, UA_Variant *output) {
	// Delete instance1
	UA_StatusCode st = UA_Server_deleteNode(server, gInstance1, false);
	return st;
}

static void
addDeleteMethod(UA_Server *server) {

	UA_MethodAttributes helloAttr = UA_MethodAttributes_default;
	helloAttr.description = UA_LOCALIZEDTEXT("", "Deletes Instance1");
	helloAttr.displayName = UA_LOCALIZEDTEXT("", "deleteInstance1");
	helloAttr.executable = true;
	helloAttr.userExecutable = true;
	UA_Server_addMethodNode(server, UA_NODEID_STRING(1, "deleteInstance1"),
		UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASORDEREDCOMPONENT),
		UA_QUALIFIEDNAME(1, "deleteInstance1"),
		helloAttr, &deleteMethodCallback,
		0, NULL, 0, NULL, NULL, NULL);
}

UA_Boolean running = true;
static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_ServerConfig *config = UA_ServerConfig_new_default();
    UA_Server *server = UA_Server_new(config);

	addMyObjectType(server);
	gInstance1 = addMyObjectInstance(UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER), server, "Instance1");
	addMyObjectInstance(UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER), server, "Instance2");
	addMyObjectInstance(UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER), server, "Instance3");
	addDeleteMethod(server);

    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return (int)retval;
}
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [x] Other OPC UA SDKs used (client or server): UA Expert
 - [x] Operating system: Windows
 - [x] Self-contained code example attached

jpfr, 2019-04-16:
That's a bug.
We have this code in `ua_services_nodemanagement.c`.
That should prevent the deletion of types that have instances.

```c
    if(UA_Node_hasSubTypeOrInstances(node)) {
        UA_LOG_INFO_SESSION(&server->config.logger, session,
                            "Delete Nodes: Cannot delete a type node "
                            "with active instances or subtypes");
        UA_Nodestore_release(server, node);
        *result = UA_STATUSCODE_BADINTERNALERROR;
        return;
    }
```

The background is that methods for a type instance are only referencing into the type but are not copied...

juangburgos, 2019-04-16:
Yes, but I am not deleting a *type*, I am just deleting an **instance**. 

I understand now that methods are shared by the *type* and by all the *instances* of the type. I also notice that the method node contains `ComponentOf` references to the *type* and all *instances*. 

So maybe an idea is to modify the code that deletes an instances's children and make a condition in the lines of:

```
if child node class is method, and it has other ComponentOf references, 
then do not delete it
```

jpfr, 2019-04-19:
A fix is proposed in #2657.

juangburgos, 2019-04-20:
@jpfr , I tested the fix on my setup and it behaves correctly. Thank you for the fix, hope it gets merged soon. Will close when if merges.

commit 1c081e08 (2019-04-23): NodeManagement: Don't delete nodes that are in several hierarchies (fix #2654)
closed 2019-04-23 by jpfr