open62541/open62541 #2658: arch/ua_network_tcp.c   UA_ClientConnectionTCP_poll  timeSinceStart bug

jswangjunsheng, 2019-04-20:
arch/ua_network_tcp.c  line 655

// the unit of timeSinceStart should be ms  so we cannot (UA_DateTime_nowMonotonic() - connStart) * UA_DATETIME_MSEC
UA_UInt32 timeSinceStart =
                        (UA_UInt32) ((UA_Double) (UA_DateTime_nowMonotonic() - connStart)
                                        ***** UA_DATETIME_MSEC);

// may be we should  (UA_Double) (UA_DateTime_nowMonotonic() - connStart) / UA_DATETIME_MSEC
UA_UInt32 timeSinceStart =
                        (UA_UInt32) ((UA_Double) (UA_DateTime_nowMonotonic() - connStart)
                                        **/** UA_DATETIME_MSEC);

jswangjunsheng, 2019-04-22:
and another question.
does minheap have more efficiency than zip-tree？
https://github.com/libevent/libevent/blob/master/minheap-internal.h

jpfr, 2019-04-27:
The ziptree is different. It is a sorted tree.
Minheap does not order the entries.

jpfr, 2019-04-27:
Fixed on master in 3ac9d00bfd2be1ee68f79988a73887400956cfd5.

closed 2019-04-27 by jpfr