open62541/open62541 #2659: Error while building with MS Visual Studio Enterprise 2017

jadhavshital, 2019-04-22:
I have built the code using CMake 3.13.2 and these files were set BUILD_SHARED_LIBS, UA_BUILD_EXAMPLES. When I opened the solution in MS Visual studio and tried to build the code, I got the following errors.

    1)  path\open62541\include\open62541/server.h(16): fatal error C1083: Cannot open include file: 'open62541/nodeids.h': No such file or directory
    2)  path\open62541\src\client\ua_client_connect.c(12): fatal error C1083: Cannot open include file: 'open62541/transport_generated.h': No such file or directory
    3)  path\open62541\src\client\ua_client.c(20): fatal error C1083: Cannot open include file: 'open62541/types_generated_encoding_binary.h': No such file or directory
    4)  path\open62541\src\server\ua_server_binary.c(18): fatal error C1083: Cannot open include file: 'open62541/transport_generated.h': No such file or directory
    5)  path\open62541\src\server\ua_server_ns0.c(13): fatal error C1083: Cannot open include file: 'open62541/namespace0_generated.h': No such file or directory
    6)  path\include\open62541/types.h(21): fatal error C1083: Cannot open include file: 'open62541/statuscodes.h': No such file or directory

So what can be the possible solution?

jpfr, 2019-04-23:
So you build the dll and then copy the project with the header files?

The missing files are created at the first build.
The are created in the build dir and need to be copied to be used with the dll.

jadhavshital, 2019-04-26:
Thanks for your feedback.
I have resolved this issue by setting UA_ENABLE_AMALGAMATION flag in CMake.
And then I have built the solution using Visual Studio.

Thanks again.

closed 2019-04-26 by jadhavshital