open62541/open62541 #2663: Custom nodeset compilation errors

cponcedeleon, 2019-04-23:
I'm trying to use the nodeset_compiler tool to generate the C and header files for my nodeset but I'm getting errors:

```
>python ./nodeset_compiler.py --types-array=UA_TYPES --existing ./Opc.Ua.NodeSet2_1_04.xml --xml opas_signal.xml opas_signal
INFO:__main__:Preprocessing (existing) ./Opc.Ua.NodeSet2_1_04.xml
INFO:__main__:Preprocessing opas_signal.xml
Traceback (most recent call last):
  File "./nodeset_compiler.py", line 196, in <module>
    ns.allocateVariables()
  File "C:\...\open62541\tools\nodeset_compiler\nodeset.py", line 306, in allocateVariables
    n.allocateValue(self)
  File "C:\...\open62541\tools\nodeset_compiler\nodes.py", line 297, in allocateValue
    self.value.parseXMLEncoding(self.xmlValueDef, dataTypeNode, self, nodeset.namespaceMapping[self.modelUri])
  File "C:\...\open62541\tools\nodeset_compiler\datatypes.py", line 161, in parseXMLEncoding
    val = self.__parseXMLSingleValue(el, parentDataTypeNode, parent, namespaceMapping=namespaceMapping)
  File "C:\...\open62541\tools\nodeset_compiler\datatypes.py", line 294, in __parseXMLSingleValue
    extobj.value.append(extobj.__parseXMLSingleValue(ebodypart, parentDataTypeNode, parent, namespaceMapping=namespaceMapping, alias=None, encodingPart=e))
  File "C:\...\open62541\tools\nodeset_compiler\datatypes.py", line 236, in __parseXMLSingleValue
    alias=alias, encodingPart=enc[1], valueRank=enc[2] if len(enc)>2 else None)
  File "C:\...\open62541\tools\nodeset_compiler\datatypes.py", line 206, in __parseXMLSingleValue
    val.parseXML(el, namespaceMapping=namespaceMapping)
  File "C:\...\open62541\tools\nodeset_compiler\datatypes.py", line 341, in parseXML
    self.value = int(unicode(xmlvalue.firstChild.data))
ValueError: invalid literal for int() with base 10: '\n                  '
```

I'm using this nodeset:

```<?xml version="1.0" encoding="utf-8"?>
<UANodeSet 
	xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" 
	xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" 
	xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" 
	xmlns:s1="http://opengroup.org/OPAF/CF/" 
	xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://opengroup.org/OPAF/CF/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="Boolean">i=1</Alias>
        <Alias Alias="LocalizedText">i=21</Alias>
        <Alias Alias="Number">i=26</Alias>
        <Alias Alias="UInteger">i=28</Alias>
        <Alias Alias="HasModellingRule">i=37</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
        <Alias Alias="HasSubtype">i=45</Alias>
        <Alias Alias="HasProperty">i=46</Alias>
        <Alias Alias="HasComponent">i=47</Alias>
        <Alias Alias="Range">i=884</Alias>
    </Aliases>
    <UAObjectType IsAbstract="true" NodeId="ns=1;i=1002" BrowseName="1:OPAS_NodeType">
        <DisplayName>OPAS_NodeType</DisplayName>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=58</Reference>
        </References>
    </UAObjectType>
    <UAObjectType NodeId="ns=1;i=1003" BrowseName="1:OPAS_IOType">
        <DisplayName>OPAS_IOType</DisplayName>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">ns=1;i=1002</Reference>
            <Reference ReferenceType="HasComponent">ns=1;i=5001</Reference>
        </References>
    </UAObjectType>
    <UAObject ParentNodeId="ns=1;i=1003" NodeId="ns=1;i=5001" BrowseName="1:OPAS_Signals">
        <DisplayName>OPAS_Signals</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">ns=1;i=1005</Reference>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1003</Reference>
        </References>
[... 119 lines cut ...]
        <References>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1005</Reference>
            <Reference ReferenceType="HasProperty">ns=1;i=6028</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=2376</Reference>
            <Reference ReferenceType="HasModellingRule">i=11508</Reference>
        </References>
    </UAVariable>
    <UAVariable DataType="LocalizedText" ParentNodeId="ns=1;i=6027" ValueRank="1" NodeId="ns=1;i=6028" BrowseName="EnumStrings">
        <DisplayName>EnumStrings</DisplayName>
        <References>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=6027</Reference>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
        </References>
    </UAVariable>
    <UAObject SymbolicName="OPAS_SignalsType" ParentNodeId="ns=1;i=1005" NodeId="ns=1;i=5002" BrowseName="1:&lt;OPAS_SignalsType>">
        <DisplayName>&lt;OPAS_SignalsType></DisplayName>
        <References>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1005</Reference>
            <Reference ReferenceType="HasTypeDefinition">ns=1;i=1005</Reference>
            <Reference ReferenceType="HasModellingRule">i=11508</Reference>
        </References>
    </UAObject>
</UANodeSet>
```

cpipero, 2019-04-25:
Looking at a part of the nodeset file 
```...
 <UAVariable DataType="Range" ParentNodeId="ns=1;i=6017" NodeId="ns=1;i=6018" BrowseName="EURange">
        <DisplayName>EURange</DisplayName>
        <References>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=6017</Reference>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
        </References>
    </UAVariable> ```

There might be an issue with **DataType="Range"** as that structure does not appear to be defined in the SDK. 
I will see if others have had problems with variable that may use a range like `AnalogDataItemType`

cpipero, 2019-04-25:
I found that we were using the amalgamation code from the website. The `Range` structure has been included now. 
I just rebuilt the current master (0.4) and it has it. 
will test and advise but this might be resolved now.

Pro, 2019-04-29:
@cponcedeleon unfortunately you did not fill out our issue template, therefore we can not really help you.

I just tested your nodeset on current master and it is building fine. If you issue still exists on current master branch please reopen this issue and post more details on how to reproduce.

closed 2019-04-29 by Pro