open62541/open62541 #2664: Amalgamation in visual studio 2019

klasik16, 2019-04-24:
Hello community I just compiled project with cmake with Amalgamation only enables.
Created empty project in VS2019 attached open62541.c and open62541.h files.

created Main.c file and included open62541.h

Compiling getting me to error:
```
Severity	Code	Description	Project	File	Line	Suppression State
Error (active)	E0020	identifier "UA_Argument" is undefined	Project1	C:\Users\ScadaAdmin\Desktop\open62541.c	40059
```	
Function code is:
```
/* OutputArguments - ns=0;i=11494 */

static UA_StatusCode function_namespace0_generated_108_begin(UA_Server *server, UA_UInt16* ns) {
UA_StatusCode retVal = UA_STATUSCODE_GOOD;
UA_VariableAttributes attr = UA_VariableAttributes_default;
attr.minimumSamplingInterval = 0.000000;
attr.userAccessLevel = 1;
attr.accessLevel = 1;
attr.valueRank = 1;
attr.arrayDimensionsSize = 1;
UA_UInt32 arrayDimensions[1];
arrayDimensions[0] = 0;
attr.arrayDimensions = &arrayDimensions[0];
attr.dataType = UA_NODEID_NUMERIC(ns[0], 296);
UA_Argument variablenode_ns_0_i_11494_variant_DataContents[2];

UA_init(&variablenode_ns_0_i_11494_variant_DataContents[0], &UA_TYPES[UA_TYPES_ARGUMENT]);
variablenode_ns_0_i_11494_variant_DataContents[0].name = UA_STRING("ServerHandles");
variablenode_ns_0_i_11494_variant_DataContents[0].dataType = UA_NODEID_NUMERIC(ns[0], 7);
variablenode_ns_0_i_11494_variant_DataContents[0].valueRank = (UA_Int32) 1;
variablenode_ns_0_i_11494_variant_DataContents[0].description = UA_LOCALIZEDTEXT("", "");

UA_init(&variablenode_ns_0_i_11494_variant_DataContents[1], &UA_TYPES[UA_TYPES_ARGUMENT]);
variablenode_ns_0_i_11494_variant_DataContents[1].name = UA_STRING("ClientHandles");
variablenode_ns_0_i_11494_variant_DataContents[1].dataType = UA_NODEID_NUMERIC(ns[0], 7);
variablenode_ns_0_i_11494_variant_DataContents[1].valueRank = (UA_Int32) 1;
variablenode_ns_0_i_11494_variant_DataContents[1].description = UA_LOCALIZEDTEXT("", "");
UA_Variant_setArray(&attr.value, &variablenode_ns_0_i_11494_variant_DataContents, (UA_Int32) 2, &UA_TYPES[UA_TYPES_ARGUMENT]);
attr.displayName = UA_LOCALIZEDTEXT("", "OutputArguments");
retVal |= UA_Server_addNode_begin(server, UA_NODECLASS_VARIABLE,
UA_NODEID_NUMERIC(ns[0], 11494),
UA_NODEID_NUMERIC(ns[0], 11492),
UA_NODEID_NUMERIC(ns[0], 46),
UA_QUALIFIEDNAME(ns[0], "OutputArguments"),
UA_NODEID_NUMERIC(ns[0], 68),
(const UA_NodeAttributes*)&attr, &UA_TYPES[UA_TYPES_VARIABLEATTRIBUTES],NULL, NULL);

return retVal;
}

```
How I can use it in my projects? any tips?

klasik16, 2019-04-26:
For compiling project in VS2019 we need to use defoult settings in cmake and check Amalgamation option
Allso in vs2019 Project>Properties>Linker>Input>Additional Dependencies add ws2_32.lib;wsock32.lib;

Pro, 2019-04-30:
Seems like you already fixed the issue. I just tested and could not reproduce.

Please use the issue template next time. If you still have the issue, please reopen.

closed 2019-04-30 by Pro