open62541/open62541 #2665: Add multi dimension array to variable value callback

santoshpanda1989, 2019-04-24:
## Description

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Master
 - [ ] Other OPC UA SDKs used (client or server): UAExpert
 - [ ] Operating system: Windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue
@jpfr Sorry for creating another issue as I was unavailable for 2 weeks. I did as you mentioned in the earlier issue. PFB the details.
I have creates a sample array and tried to assign the dimensions for the Variant. But, I am getting BadTypeMismatch error. PFB the server screenshot. 
[image]
Also, find below the the nodeset.xml snippet.

[image]

Find the code snippet below.

    UA_Variant v;
    UA_Double d[9] = {1.0, 2.0, 3.0, 4.0, 5.0, 6.0, 7.0, 8.0, 9.0};
    v.arrayDimensions = (UA_UInt32 *)UA_Array_new(9, &UA_TYPES[UA_TYPES_UINT32]);
    v.arrayDimensionsSize = 2;
    v.arrayDimensions[0] = 3;
    v.arrayDimensions[1] = 3;
    UA_Variant_setArrayCopy(&v, d, 9, &UA_TYPES[UA_TYPES_DOUBLE]);
    UA_NodeId currentNodeId = UA_NODEID_NUMERIC(3, 6010);
    UA_Server_writeValue(server, currentNodeId, v);

I would really appreciate any help regarding this.

Thanks,
Santosh

jpfr, 2019-04-25:
UA_Variant_setArrayCopy is overriding the arraydimensions you have set before.

santoshpanda1989, 2019-04-26:
Thanks @jpfr. It works now.

closed 2019-04-26 by santoshpanda1989